espressif/arduino-esp32 #6295: ESP32 using SD card with HSPI works at full speed under 1.0.6, fails or runs very slowly under 2.0.2

avillacis, 2022-02-16:
### Board

ESP32 Dev Module

### Device Description

YUBOX Node

### Hardware Configuration

#define SD_SCK  GPIO_NUM_14
#define SD_MISO GPIO_NUM_12
#define SD_MOSI GPIO_NUM_13
#define SD_SS   GPIO_NUM_15

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Fedora 35 x86_64

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When using the sketch provided below using version 2.0.2 of Arduino-ESP32, the attempted I/O to the SD card is faulty or very slow, and riddled with error messages. Under 1.0.6, the same sketch runs correctly on the same board and the same SD card.

Under a different board using ESP32-S2 the issue is even worse - the card completely fails to mount. However, this report is ESP32 only because only ESP32 is testable under both 1.0.6 and 2.0.2.

The sketch initializes an SPI object in HSPI mode, with the pins shown as follows:
#define SD_SCK  GPIO_NUM_14
#define SD_MISO GPIO_NUM_12
#define SD_MOSI GPIO_NUM_13
#define SD_SS   GPIO_NUM_15

Other than that, and the conditional defines for ESP32-S2, the sketch is the same as the official example for testing SD card I/O in SPI mode. Under both 1.0.6 and 2.0.2 the sketch was compiled in full verbose mode.

### Sketch

```cpp
/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       -
 *    D3       SS
 *    CMD      MOSI
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      SCK
 *    VSS      GND
 *    D0       MISO
 *    D1       -
 */
[... 766 lines cut ...]
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 4467 ms
1048576 bytes written for 8039 ms
Total space: 486MB
Used space: 1MB
OK
```

### Other Steps to Reproduce

I have searched existing bug reports. I have found and applied https://github.com/espressif/arduino-esp32/pull/6162 manually on top of 2.0.2. My tests for 2.0.2 use this attempted fix, without much success. Therefore, I think the pull will *not* fix the issue, at least for me.

The comment at https://github.com/espressif/arduino-esp32/issues/6189#issuecomment-1034399765 appears to be the same issue as mine, but there is no verbose information to check whether this is so.

https://github.com/espressif/arduino-esp32/issues/5900 might be relevant to this bug, but lacks much information other than an attempt to use HSPI.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

avillacis, 2022-02-16:
If I revert https://github.com/espressif/arduino-esp32/pull/5988 under Arduino-ESP32 2.0.2, the behavior is apparently back to being tolerable, even with https://github.com/espressif/arduino-esp32/pull/6162 still applied:

```
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Trying mount of SD card at freq 4000000...
[  2512][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2512][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  2514][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2618][W][sd_diskio.cpp:186] sdCommand(): token error [59] 0x5
SD Card Type: SDSC
SD Card Size: 486MB
Listing directory: /
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
Creating Dir: /mydir
Dir created
Listing directory: /
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  DIR : mydir
Removing Dir: /mydir
Dir removed
Listing directory: /
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 4479 ms
1048576 bytes written for 7873 ms
Total space: 486MB
Used space: 1MB
OK
Trying mount of SD card at freq 4000000...
SD Card Type: SDSC
SD Card Size: 486MB
Listing directory: /
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
Creating Dir: /mydir
Dir created
Listing directory: /
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  DIR : mydir
Removing Dir: /mydir
Dir removed
Listing directory: /
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 4479 ms
1048576 bytes written for 8067 ms
Total space: 486MB
Used space: 1MB
OK

```

P-R-O-C-H-Y, 2022-02-17:
Please try master branch of Arduino-Esp32. Its already fixed. Fix will be included in release 2.0.3. Thanks

avillacis, 2022-02-17:
> Please try master branch of Arduino-Esp32. Its already fixed. Fix will be included in release 2.0.3. Thanks

Could you please tell me which particular commit(s) or merged pull request(s) are supposed to fix this? I explicitly remarked that I manually applied https://github.com/espressif/arduino-esp32/pull/6162 and it did **not** fix the bug for me. I also commented that manually reverting https://github.com/espressif/arduino-esp32/pull/5988 on top of 2.0.2 does fix the bug for me.

avillacis, 2022-02-17:
> Could you please tell me which particular commit(s) or merged pull request(s) are supposed to fix this? I explicitly remarked that I manually applied #6162 and it did **not** fix the bug for me. I [...]

Answering myself, I have just noticed https://github.com/espressif/arduino-esp32/pull/6103 , so I am going to test this and https://github.com/espressif/arduino-esp32/pull/6162 together with https://github.com/espressif/arduino-esp32/pull/5988 .

avillacis, 2022-02-17:
> Answering myself, I have just noticed #6103 , so I am going to test this and #6162 together with #5988 .

Confirmed that the referenced pulls fix my bug. Closing.

closed 2022-02-17 by avillacis