espressif/arduino-esp32 #6297: Architecture or board not supported.

Ivan-Dosev, 2022-02-17:
### Board

ESP32 Dev Mode

### Device Description

plain module

### Hardware Configuration

DHT 11 sensor and LED lamp

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

  from /Users/ivandimitrov/Desktop/Firebase-ESP32/Firebase-ESP32.ino:2:
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^~~~~
exit status 1
Error compiling for board ESP32 Dev Module.
Board at /dev/cu.usbserial-0001 is not available

### Sketch

```cpp
#include <WiFi.h>
#include <FirebaseESP32.h>
#include <ArduinoJson.h>
#include <HTTPClient.h>
#include "DHT.h"

#define DHTTYPE DHT11
// DHT Sensor
uint8_t DHTPin = 4; 
               
// Initialize DHT sensor.
DHT dht(DHTPin, DHTTYPE);                

float Temperature;
float Humidity;
int  intValue;
int  intReadBlock;
int  block;

int random1 = 0;
int random2 = 0;
int random3 = 0;
int random4 = 0;

const int output2  = 2;
const int output25 = 25;
const int output26 = 26;
const int output27 = 27;
const int output33 = 33;

// Replace with your network credentials
 char* ssid = "FIT-IoT-HUERTH";
//const char* password = "48575443B5BF570D";

const char* firebase_host = "https://make-it-so-77244-default-rtdb.firebaseio.com/";
const char* firebase_auth = "Czuz57U0imGCbTdpP98h4o2EwNJQJlDkdMFVNicj";

FirebaseData firebasedata;
FirebaseData fbdo;

void setup() {
  
  pinMode(output2,  OUTPUT);
  pinMode(output25, OUTPUT);
  pinMode(output26, OUTPUT);
  pinMode(output27, OUTPUT);
  pinMode(output33, OUTPUT);
  block = 1;
  intReadBlock = 0;
  
  // Set outputs to LOW
  digitalWrite(output2,  LOW);
  digitalWrite(output25, HIGH);
  digitalWrite(output26, HIGH);
  digitalWrite(output27, HIGH);
  digitalWrite(output33, HIGH);

  dht.begin();  
  Serial.begin(9600);
  
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid);
//  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Firebase.begin(firebase_host, firebase_auth);
  Firebase.reconnectWiFi(true);
  Firebase.setReadTimeout(firebasedata, 1000 * 60);
  Firebase.setwriteSizeLimit(firebasedata , "tiny");
  delay(500);
  Firebase.RTDB.setInt(&fbdo, "/test/ReadBlock", 0);
  Firebase.RTDB.setInt(&fbdo, "/test/relay-0", 0);

}

void loop() {
  
  if (intReadBlock) {

                     delay(30000);
                     readDataFromContract(); 
                     delay(-1);
    
  }else{
       readData();
       delay(1000);
      
       Temperature = dht.readTemperature(); 
       Humidity    = dht.readHumidity();
       delay(100);

       Firebase.RTDB.setFloat(&fbdo, "/DHT11/Temperature",  Temperature);
       Firebase.RTDB.setFloat(&fbdo, "/DHT11/Humidity", Humidity);
       delay(200);
       
       Relaycode();
       delay(1000);
  }

  

}

void readData() {
      if (Firebase.RTDB.getInt(&fbdo, "/test/relay-0")) {
      if (fbdo.dataType() == "int") {
        intValue = fbdo.intData();
     //   Serial.println(intValue);
         digitalWrite(output2,  intValue );
         block = intValue;
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }

      if (Firebase.RTDB.getInt(&fbdo, "/test/ReadBlock")) {
      if (fbdo.dataType() == "int") {
        intReadBlock = fbdo.intData();
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
}

void Relaycode(){

if ( !block ){

    delay(1000);
    random1 =  random(2);
    random2 =  random(2);
    random3 =  random(2);
    random4 =  random(2);
    

    if (random1 == 1) {
          digitalWrite(output33, HIGH);
          Firebase.RTDB.setInt(&fbdo, "/test/relay-1", 1);            
    }else{
          digitalWrite(output33, LOW);
           Firebase.RTDB.setInt(&fbdo, "/test/relay-1", 0);           
    }

    if (random2 == 1) {
          digitalWrite(output27, HIGH);
           Firebase.RTDB.setInt(&fbdo, "/test/relay-2", 1);           
    }else{
          digitalWrite(output27, LOW);
             Firebase.RTDB.setInt(&fbdo, "/test/relay-2", 0);            
    }

    if (random3 == 1) {
          digitalWrite(output26, HIGH);
             Firebase.RTDB.setInt(&fbdo, "/test/relay-3", 1);       
    }else{
          digitalWrite(output26, LOW);
             Firebase.RTDB.setInt(&fbdo, "/test/relay-3", 0);            
    }

    if (random4 == 1) {
          digitalWrite(output25, HIGH);
           Firebase.RTDB.setInt(&fbdo, "/test/relay-4", 1);          
    }else{
          digitalWrite(output25, LOW);
           Firebase.RTDB.setInt(&fbdo, "/test/relay-4", 0);               
    }
  }
}

void readDataFromContract(){

  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("readDataFromContract.....");
    String redaData = "{\"jsonrpc\":\"2.0\",\"method\":\"eth_call\",\"params\":[{\"to\":\"0xA4380435a13153C41658acC729B369Ddb84f5c11\",\"data\":\"0x57de26a4\"},\"latest\"],\"id\":1}";
    Serial.println(redaData);
    StaticJsonBuffer<1000> JSONbufferRead;
    String gethResultRead = callGeth(redaData);  //Get the response from Geth JSONRPC
    JsonObject& gethJSONRPCRead = JSONbufferRead.parseObject(gethResultRead);
    String lightOnString = gethJSONRPCRead["result"];
    lightOnString.remove(66,64);
    
    
    //parsing & converting Geth JSON-RPC hex results is not easy
    long int lightOn = strtol(lightOnString.c_str(), NULL, 16);
    Serial.print("The Code is: = ");
    Serial.println(lightOn);
    Serial.println( lightOnString);

                       digitalWrite(output2,  HIGH);
                       digitalWrite(output25, HIGH);
                       digitalWrite(output26, HIGH);
                       digitalWrite(output27, HIGH);
                       digitalWrite(output33, HIGH);
      } else {
 
    Serial.print(".");

  }
}

String callGeth(String inputJSON) {

  HTTPClient http;  
                                          
  http.begin("https://rinkeby.infura.io/v3/a146daf63d93490995823f0910f50118");  
          
    http.addHeader("Content-Type", "application/json");  
 
    int httpCode = http.POST(inputJSON);  
    String JSONResult = http.getString();  
    
    http.end();
    return JSONResult;
}
```

### Debug Message

```plain
#error Architecture or board not supported.
  ^~~~~
exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Ivan-Dosev, 2022-02-17:
Any ideas how I can fix that?

Jason2866, 2022-02-17:
The lib is insfalled? https://github.com/mobizt/Firebase-ESP32

VojtechBartoska, 2022-04-11:
Do you still need help?

Ivan-Dosev, 2022-04-12:
Yes please

me-no-dev, 2022-04-13:
I would guess that one of the libs is trying to use the general Arduino SD library, which comes with the IDE and does not support esp32. There is nothing we can do about that though. That lib needs to start using our SD library instead.

VojtechBartoska, 2022-05-03:
Closing as we cannot help on this one.

closed 2022-05-03 by VojtechBartoska