espressif/arduino-esp32 #6302: Make Serial.onReceive() accept std::function

gonzabrusco, 2022-02-18:
### Related area

UART

### Hardware specification

ESP32 DevKit

### Is your feature request related to a problem?

Hello. It would be useful if you could make onReceive() on HardwareSerial accept std:function so I can point it to a class method with std:bind.  You would need to include <functional>.

This request could be extended to other callbacks in the framework (like Wifi.onEvent() ). 

If you don't want to include <functional> by default, could you at least provide us with a MACRO that enables or disables this?

Thanks

### Describe the solution you'd like

Make callbacks use functional

### Describe alternatives you've considered

Global function. But if I'm programming in pure C++ and I need to access privates variables of a class from the callback, it gets weird. I would prefer to be able to call a class method directly to avoid exposing data outside the class.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-02-19:
@gonzabrusco 

What about this C++ approach?

``` cpp
class Receiver {
  public:
  static void gotUARTdata(void);
};

void Receiver::gotUARTdata(void) {
  Serial.println("Got Data on UART...");
}

void setup() {
  Serial.begin(115200);
  Serial.onReceive(Receiver::gotUARTdata);
}

void loop() {
  // Nothing here as this time.
}
```

SuGlider, 2022-02-19:
I also think that Arduino doen't support Starndard C++
Thus Functional won't work at all...

Could you provide an example with ```std::function``` that compiles in ESP Arduino?

SuGlider, 2022-02-19:
@gonzabrusco 

>Global function. But if I'm programming in pure C++ and I need to access privates variables of a class from the callback, it gets weird. I would prefer to be able to call a class method directly to avoid exposing data outside the class.

Try this approach:

``` cpp
class Receiver {
  int a;
  public:
  static void gotUARTdata(void);
  void set(int i);
  int get();
};

Receiver r;

void Receiver::gotUARTdata(void) {
  Serial.println("Got Data on UART...");
  Serial.print("Private Receiver::a = ");
  Serial.println(r.get());
  r.set(r.get() + 1);
}

void Receiver::set(int i) {
  a = i;
}

int Receiver::get(void) {
  return a;
}

void setup() {
  r.set(10);   // stating point
  Serial.begin(115200);
  Serial.onReceive(Receiver::gotUARTdata);
}

void loop() {
  // Nothing here as this time.
}
```

## Output --> just hit Enter on Serial Monitor

```
Got Data on UART...
Private Receiver::a = 10
Got Data on UART...
Private Receiver::a = 11
Got Data on UART...
Private Receiver::a = 12
Got Data on UART...
Private Receiver::a = 13
Got Data on UART...
Private Receiver::a = 14
Got Data on UART...
Private Receiver::a = 15
Got Data on UART...
Private Receiver::a = 16
Got Data on UART...
Private Receiver::a = 17
Got Data on UART...
Private Receiver::a = 18
Got Data on UART...
Private Receiver::a = 19
```

atanisoft, 2022-02-19:
@SuGlider Arduino-ESP32 does in fact support C++11 (or later) which supports std::function. I use it quite extensively in the OpenMRNLite library (latest published version only compiles on 1.0.6, 2.0.x support is complete but not published yet)

SuGlider, 2022-02-19:
Nice @atanisoft !
I'll take a look.

gonzabrusco, 2022-02-19:
Yes, I see the trick you did there. By declaring as static inside the class the function behaves as a global function for all the objects of that class. I see no real difference between using your method and a global function with global objects (with methods to access class variables). The advantage of functional is that you can bind a non-static class method and be able to access class variables directly without the need of having global objects or specific getters or setters.

Functional works in Arduino and I use it in multiple libraries inside my project. Other examples of libraries already using functional:
https://github.com/emelianov/modbus-esp8266
https://github.com/256dpi/arduino-mqtt

SuGlider, 2022-02-19:
I agree that a static function in a class is just a global function for all objects. 
But this is the best way to make it work with the C HAL and IDF layers used in Arduino-ESP, not having to refactor everything there.

Using sdt::function has a big problem... make it work with esp32-hal.uart.c
It's a C layer of code which is used to perform everything around UART.

HardwareSerial.cpp i just a CPP adaptation layer to esp32-hal-uart.c

As far as I know, it's not possible to convert std:function to a C function pointer.
@atanisoft @gonzabrusco  Any idea to get it done?

SuGlider, 2022-02-19:
I think that the way to make onReceive(std::function) work will be by moving all the processing of events to the HardwareSerial.cpp part of code... 

But it will take some time to be done. I think it is a nice improvenment, but it's low priority at this time.
@gonzabrusco, please feel free to provide a PR for this feature, if you feel like helping on it.

gonzabrusco, 2022-02-21:
Thanks @SuGlider

If I have some time I will try to propose a solution to this. Thanks
Gonzalo

closed 2022-02-21 by gonzabrusco

hreintke, 2022-02-23:
@SuGlider : A not complex solution is (and largely used) is to add a parameter to the onReceive function.
Like `HardwareSerial.onReceive(function, parameter);` The onReceive Callback then includes that parameter
 `uart->onreceive(parameter)` 

The application looks like this : 

```
#include "Arduino.h"

class SerialHandler
{
	SerialHandler(HardwareSerial hs)
	{
		hs.onReceive(staticReceiveData,this);
	}
	static void staticReceiveData(void* onReceiveParam)
	{
		((SerialHandler*)onReceiveParam)->onReceiveData();
	}
	void onReceiveData()
	{
		hs.printf("Received data\n");
	}
};

void setup()
{
}

void loop()
{
}
```

I can make a PR for you if like the idea.