espressif/arduino-esp32 #6304: LoadProhibited exception on Hotspot Shutdown

DreiDe, 2022-02-18:
### Board

ESP32

### Device Description

ESPduino

### Hardware Configuration

No attached devices

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

ESP crashes when Debug Level is set higher than 1 and the access point it is connected to, is forcefully stopped.

**Assumptions about the cause:**
In [WiFiGeneric.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp) `event->event_info.disconnected.reason` turns `0` which causes [reason2str](https://github.com/espressif/arduino-esp32/blob/50e9772ecf3f302df03f9853f040796343d43334/libraries/WiFi/src/WiFiGeneric.cpp#L834) to access `system_event_reasons` at an index of `-1`. In the current implementation of the function it is not possible to access index 0 at all.

The issue seems similar to https://github.com/espressif/arduino-esp32/issues/2334

### Sketch

```cpp
#include "WiFi.h"
 
const char* ssid = "yourNetworkName";
const char* password =  "yourNetworkPass";
 
void setup() {
 
  Serial.begin(115200);
 
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println("Connected to the WiFi network");
 
}
 
void loop() {}
```

### Debug Message

```plain
Backtrace: 0x400014fd:0x3ffbb1b0 0x4010be59:0x3ffbb1c0 0x401133d2:0x3ffbb4d0 0x4011340e:0x3ffbb560 0x400ee2aa:0x3ffbb5a0 0x400dff59:0x3ffbb600 0x400e01ac:0x3ffbb6f0 0x40089e82:0x3ffbb720
  #0  0x400014fd:0x3ffbb1b0 in ?? ??:0
  #1  0x4010be59:0x3ffbb1c0 in _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529
  #2  0x401133d2:0x3ffbb4d0 in _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
  #3  0x4011340e:0x3ffbb560 in vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41 
  #4  0x400ee2aa:0x3ffbb5a0 in log_printf at C:\Users\Simon\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-uart.c:533
  #5  0x400dff59:0x3ffbb600 in WiFiGenericClass::_eventCallback(void*, system_event_t*, wifi_prov_event_t*) at C:\Users\Simon\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiGeneric.cpp:297
  #6  0x400e01ac:0x3ffbb6f0 in _network_event_task(void*) at C:\Users\Simon\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiGeneric.cpp:297
  #7  0x40089e82:0x3ffbb720 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

### Other Steps to Reproduce

Use Connectify Hotspot to easily enable and disable the AP.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, sorry for late reply. Are you able to retest this with Arduino core ESP32 version 2.0.3-rc1? Thanks!

VojtechBartoska, 2022-05-04:
I'm closing the issue as expired due to no answer.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska