espressif/arduino-esp32 #6305: abort() was called at PC 0x401be9b3 on core 1

sebyldino, 2022-02-18:
### Related area

I have this error randomly

### Hardware specification

ESP32 WROOM 32

### Is your feature request related to a problem?

Hello,

I am making a BLE sensor to trigger events in Homey vith beacon. It works, I just sometimes get this error.
I dont know why, and ESP32 reboot and works again.

Thanks.

**Error:**

abort() was called at PC 0x401be85b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f3bc:0x3ffcc330 0x4008f639:0x3ffcc350 0x401be85b:0x3ffcc370 0x401be8a2:0x3ffcc390 0x401bdf77:0x3ffcc3b0 0x401be2b2:0x3ffcc3d0 0x401be015:0x3ffcc3f0 0x400d625f:0x3ffcc410 0x400d7398:0x3ffcc450 0x400d73e0:0x3ffcc480 0x400d0ead:0x3ffcc4a0 0x400d8434:0x3ffcc4c0 0x400906ae:0x3ffcc4e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Done

**Code:**

`#include "BLEDevice.h"

/**************** Homey *************/
#include <WiFi.h>
#include <WiFiClient.h>
#include <Homey.h>

unsigned long previousMillis = 0;
const unsigned long interval = 1000; //Intervale entre chaque mesure

void wifi() {
  if (WiFi.status() != WL_CONNECTED) {
    WiFi.begin("SSID, "PASSWORD"); 
    uint8_t timeout = 30;
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("connecting...");
      if (timeout<1) break;
    }
    if (WiFi.status() == WL_CONNECTED) {
      //Print IP address
      Serial.print("WiFi Connected! The IP adress is: ");
      Serial.print(WiFi.localIP());
      Serial.println("");
    }
  }
}
/**************************************/

static BLEAddress *pServerAddress;
BLEScan* pBLEScan;
BLEClient*  pClient;
bool deviceFound = false;
bool BotonOff = false;
String knownAddresses[] = {"6c:79:b8:b7:4e:0e", "d6:9a:43:5c:dc:b9"};
String AddressTileRaven = "d6:9a:43:5c:dc:b9";
String AddressHM10 = "6c:79:b8:b7:4e:0e";
int ValeurRSSI = -60;
int rssi;
//unsigned long entry;
static void notifyCallback(
BLERemoteCharacteristic* pBLERemoteCharacteristic,
uint8_t* pData,
size_t length,
bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
}
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice Device){
    //Serial.print("BLE Advertised Device found: ");
    //Serial.println(Device.toString().c_str());
    pServerAddress = new BLEAddress(Device.getAddress()); 
    bool known = false;
    bool Master = false;
    for (int i = 0; i < (sizeof(knownAddresses) / sizeof(knownAddresses[0])); i++) {
      if (strcmp(pServerAddress->toString().c_str(), knownAddresses[i].c_str()) == 0) 
      known = true;
    }
    if (known) {
      Serial.print("Device found: ");
      Serial.println(Device.getRSSI());
      rssi = Device.getRSSI();
      updatesensor();                   // FONCTION D'ENVOIE DE LA VALEUR VERS HOMEY
      if (Device.getRSSI() > ValeurRSSI) {
        deviceFound = true;
      }
      else {
        deviceFound = false;
      }
      Device.getScan()->stop();
      delay(100);
      }
    }
};

/*static void my_gap_event_handler(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t* param) {

  Serial.print("RSSI status");Serial.println(param->read_rssi_cmpl.status);
  Serial.print("RSSI ");Serial.println(param->read_rssi_cmpl.rssi);
  Serial.print("Address ");Serial.println(BLEAddress(param->read_rssi_cmpl.remote_addr).toString().c_str());
}
*/

void setup() {
/*************** Homey ****************/
  wifi();
  Homey.begin("ESP32 Rose");  // Remplacer le nom par celui qui apparaitra dans Homey
  Homey.setClass("sensor");  // Catégorie voir: https://apps-sdk-v3.developer.homey.app/tutorial-device-classes.html
  Homey.addCapability("measure_noise");      // measure_rain est la seule capacité en mm, voir: https://apps-sdk-v3.developer.homey.app/tutorial-device-capabilities.html 
/**************************************/

  
 // BLEDevice::setCustomGapHandler(my_gap_event_handler); 
  Serial.begin(115200);
  BLEDevice::init("");
  pClient  = BLEDevice::createClient();
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  Serial.println("Done");
}

void loop() { 
  wifi();
  Homey.loop();
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis > interval) {
  previousMillis = currentMillis;
  Bluetooth();
  }
}

/***************************************************/
/*                    FONCTIONS                    */
/***************************************************/

/******************** Homey ************************/
void updatesensor() {
  // int distance = distanceSensor.measureDistanceCm()*10; //Convert centimeters to millimeters
  // Serial.print(distance);
  // Serial.print(" mm ");
  // Serial.println("");
  Homey.setCapabilityValue("measure_noise", (rssi));
}
/***************************************************/

void Bluetooth() {
  Serial.println();
  Serial.println("BLE Scan Restarted...");
  deviceFound = false;
  BLEScanResults scanResults = pBLEScan->start(5);
  if (deviceFound) {
    Serial.println("ON");
    if (strcmp(pServerAddress->toString().c_str(), AddressTileRaven.c_str()) == 0 && (deviceFound == true)) {
      Serial.println("Tile Raven"); 
    }
    else if (strcmp(pServerAddress->toString().c_str(), AddressHM10.c_str()) == 0 && (deviceFound == true)) {
      Serial.println("HM-10"); 
    }
  }
  else{
    Serial.println("OFF");
  }
}`

### Describe the solution you'd like

Can you tell me what is wrong?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-02-19:
https://github.com/espressif/arduino-esp32#decoding-exceptions

sebyldino, 2022-02-20:
Hello, 
Thank you. Exception Decoder get this. What must i do with it?

`

Decoding stack results
0x4008f3bc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008f639: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x401be85b: __cxxabiv1::__cxa_call_unexpected(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_personality.cc line 742
0x401be8a2: std::get_terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 92
0x401bdf77: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 82
0x401be2b2: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x401be015: operator delete[](void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_opv.cc line 32
0x400d625f: WiFiUDP::parsePacket() at /Users/sebastienmartin/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi/src/WiFiUdp.cpp line 206
0x400d7398: HomeyClass::handleTcp() at /Users/sebastienmartin/Documents/Arduino/libraries/Homeyduino-1.0.2/src/Homey.cpp line 670
0x400d73e0: HomeyClass::loop() at /Users/sebastienmartin/Documents/Arduino/libraries/Homeyduino-1.0.2/src/Homey.cpp line 297
0x400d0ead: Bluetooth() at /Users/sebastienmartin/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_tree.h line 858
0x400d8434: loopTask(void*) at /Users/sebastienmartin/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp line 20
0x400906ae: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on version 2.0.3-rc1 of Arduino core ESP32?

VojtechBartoska, 2022-05-18:
@sebyldino Do you still face this problem on v2.0.4?

VojtechBartoska, 2022-06-14:
Closing as expired, if needed you can reopen the issue. Thanks.

closed 2022-06-14 by VojtechBartoska