espressif/arduino-esp32 #6307: FSBrowser does not work on LittleFS

fred9999, 2022-02-19:
### Board

ESP32 Dev Module 4MB

### Device Description

no other hardware attached

### Hardware Configuration

no other hardware attached

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Win10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

FSBrowser.ino Works great on SPIFFS but does not work on LittleFS.

Could you please make it work on LittleFS.

Thank you in advance.

### Sketch

```cpp
/*
  FSWebServer - Example WebServer with FS backend for esp8266/esp32
  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the WebServer library for Arduino environment.
  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.
  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.
  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  upload the contents of the data folder with MkSPIFFS Tool ("ESP32 Sketch Data Upload" in Tools menu in Arduino IDE)
  or you can upload the contents of a folder if you CD in that folder and run the following command:
  for file in `ls -A1`; do curl -F "file=@$PWD/$file" esp32fs.local/edit; done
  access the sample web page at http://esp32fs.local
  edit the page by going to http://esp32fs.local/edit
*/
#include <WiFi.h>
#include <WiFiClient.h>
[... 261 lines cut ...]
    json += ", \"gpio\":" + String((uint32_t)(0));
    json += "}";
    server.send(200, "text/json", json);
    json = String();
  });
  server.begin();
  DBG_OUTPUT_PORT.println("HTTP server started");

}

void loop(void) {
  server.handleClient();
  delay(2);//allow the cpu to switch to other tasks
}
```

### Debug Message

```plain
not working on LittleFS
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pipi61, 2022-02-19:
You compile to SPIFFS...
Try in source change all SPIFFS text to LittleFS

fred9999, 2022-02-21:
Thank you for the quick reply.

I changed SPIFFS to LittleFS in FSBrowser.ino file. It compiles, uploads and connects to my WiFi OK, but does not work propely:

works: 
-upload
-delete

does not work: 
-download (downloads file but without any contents)
-create

Tested on Firefox and Google Chrome

pipi61, 2022-02-21:
some function call not equal in SPIFFS and LITTLEFS
read this: https://github.com/lorol/LITTLEFS

fred9999, 2022-02-21:
Do not really understand what you mean, too complicated for me :-(

Can we assume that ESP32 official libraries are not ready for LittleFS yet and for beginners it's better to use SPIFFS still? 

But thanks for your reply anyway

pipi61, 2022-02-21:
Try this.
Adapted from ESP8266 (lib version 2022.02.04)  to ESP32
Two extra user buttons without function.
Added filesystem format from browser (in index.html)
web files move from LitteFS to program memory, don't need upload from data folder
[FSBrowser32from8266.zip](https://github.com/espressif/arduino-esp32/files/8111630/FSBrowser32from8266.zip)

fred9999, 2022-02-21:
Brilliant! Absolutely fantastic!

I really appreciate your help. Do not know and do not understand how you did it, but your solution works like a charm.

Keep up the good work

fred9999, 2022-04-11:
Yes this issue can be closed. It works like a charm. 
Thank you and keep up the good work!

VojtechBartoska, 2022-04-12:
@fred9999 Thanks for fast reply and good luck with your projects!

closed 2022-04-12 by VojtechBartoska