espressif/arduino-esp32 #6308: When compiling the code including the BluetoothSerial.h file, it prompts: undefined reference

winseven2018, 2022-02-19:
### Board

ESP32C3

### Device Description

ESP32C3kit

### Hardware Configuration

An I2C Device

### Version

v2.0.2

### IDE Name

Arduino

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I found that when compiling the code including the BluetoothSerial.h file, the console reported a lot of undefined references

### Sketch

```cpp
#include <BluetoothSerial.h>

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

#define BT_DISCOVER_TIME	10000

static bool btScanAsync = true;
static bool btScanSync = true;

void btAdvertisedDeviceFound(BTAdvertisedDevice* pDevice) {
	Serial.printf("Found a device asynchronously: %s\n", pDevice->toString().c_str());
}

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");

  if (btScanAsync) {
    Serial.print("Starting discoverAsync...");
    if (SerialBT.discoverAsync(btAdvertisedDeviceFound)) {
      Serial.println("Findings will be reported in \"btAdvertisedDeviceFound\"");
      delay(10000);
[... 49 lines cut ...]
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:953: undefined reference to `esp_bt_gap_set_scan_mode'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:956: undefined reference to `esp_bt_gap_start_discovery'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:961: undefined reference to `esp_bt_gap_cancel_discovery'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `BluetoothSerial::discoverAsyncStop()':
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:962: undefined reference to `esp_bt_gap_cancel_discovery'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `std::enable_if<std::__and_<std::__not_<std::__is_tuple_like<void (*)(std::_Any_data const&, BTAdvertisedDevice*&&)> >, std::is_move_constructible<void (*)(std::_Any_data const&, BTAdvertisedDevice*&&)>, std::is_move_assignable<void (*)(std::_Any_data const&, BTAdvertisedDevice*&&)> >::value, void>::type std::swap<void (*)(std::_Any_data const&, BTAdvertisedDevice*&&)>(void (*&)(std::_Any_data const&, BTAdvertisedDevice*&&), void (*&)(std::_Any_data const&, BTAdvertisedDevice*&&))':
d:\tools\arduino\hardware\espressif\arduino-esp32\tools\riscv32-esp-elf\riscv32-esp-elf\include\c++\8.4.0\bits/move.h:193: undefined reference to `esp_bt_gap_set_scan_mode'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `BluetoothSerial::discoverAsync(std::function<void (BTAdvertisedDevice*)>, int)':
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:981: undefined reference to `esp_bt_gap_start_discovery'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `.L0 ':
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:114: undefined reference to `esp_bt_gap_resolve_eir_data'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `get_name_from_eir':
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:118: undefined reference to `esp_bt_gap_resolve_eir_data'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `.L0 ':
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:413: undefined reference to `esp_bt_gap_cancel_discovery'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `esp_bt_gap_cb':
D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:415: undefined reference to `esp_spp_start_discovery'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:477: undefined reference to `esp_bt_gap_pin_reply'
d:/tools/arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: D:\Tools\Arduino\hardware\espressif\arduino-esp32\libraries\BluetoothSerial\src/BluetoothSerial.cpp:485: undefined reference to `esp_bt_gap_ssp_confirm_reply'
collect2.exe: error: ld returned 1 exit status
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-03-03:
Bluetooth Serial or SPP (Serial Port Profile)  is part of BT 2.1 and uses RFCOMM to connect two peer devices.
It will only work for the ESP32. 

ESP32-C3 has only BLE support. There is no SPP in BLE. 
Therefore, BluetoothSerial will never work for the C3.

SuGlider, 2022-03-20:
Closing it, given that there is nothing to be done, as already explained.

closed 2022-03-20 by SuGlider

lbernstone, 2022-03-20:
@SuGlider Maybe put a note on the readme for BluetoothSerial that bluetooth classic is only supported on ESP32

QGB, 2022-04-28:
GATT (Generic Attribute profile is a BLE profile, it defines specifications for two BLE devices to communicate through Service and Characteristic, the two parties of GATT communication are Client/Server relationship, Peripheral is GATT Server, Central is the GATT Client, all the communications, both are initiated by Client, and receive the response from Server.

summerfind, 2023-03-15:
the example of: SerialToSerialBT from Arduino IDE, passed compiling for ESP32 Dev Module, failed for ESP32 S3 Dev module, why?

SuGlider, 2023-03-16:
> the example of: SerialToSerialBT from Arduino IDE, passed compiling for ESP32 Dev Module, failed for ESP32 S3 Dev module, why?

SerialBT uses Bluetooth Classic protocol based on BR/EDR, which is only supported by the old ESP32.
ESP32-S3 and ESP32-C3 only support Bluetooth Low Energy (BLE), therefore these are incompatible with SerialBT.

For more details about comparing BT and BLE:
https://www.mathworks.com/help/bluetooth/gs/comparison-of-bluetooth-bredr-and-bluetooth-le.html

PR #6455 (2022-03-28): Warns about SPP only available for ESP32
## Summary
This PR helps users that try to use BT SPP (BluetoothSerial), warning that it only works for the ESP32. 

## Impact
None

## Related links
Fix #6308 
Fix #6422