espressif/arduino-esp32 #6309: After I upload the sample code: SimpleBleDevice, the corresponding bluetooth name cannot be found on the Android phone

winseven2018, 2022-02-19:
### Board

ESP32C3

### Device Description

ESP32C3kit

### Hardware Configuration

An I2C Device

### Version

v2.0.2

### IDE Name

Arduino

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

After I uploaded the sample code SimpleBleDevice, no error was reported during the upload process, but neither the Android device nor the Android app nRF Connect could find the corresponding Bluetooth device name

[image]

### Sketch

```cpp
#include "SimpleBLE.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

SimpleBLE ble;

void onButton(){
    String out = "BLE32 name: ";
    out += String(millis() / 1000);
    Serial.println(out);
    ble.begin(out);
}

void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    pinMode(0, INPUT_PULLUP);
    Serial.print("ESP32 SDK: ");
    Serial.println(ESP.getSdkVersion());
    ble.begin("ESP32 SimpleBLE");
    Serial.println("Press the button to change the device's name");
}

void loop() {
    static uint8_t lastPinState = 1;
    uint8_t pinState = digitalRead(0);
    if(!pinState && lastPinState){
        onButton();
    }
    lastPinState = pinState;
    while(Serial.available()) Serial.write(Serial.read());
}
```

### Debug Message

```plain
No bug reported in Arduino
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, sorry for late reply. Are you able to retest this with Arduino core ESP32 version 2.0.3-rc1? Thanks!

winseven2018, 2022-04-14:
Thank you for your help, after I replaced the example code it ran successfully and found the bluetooth device.

closed 2022-04-14 by VojtechBartoska