espressif/arduino-esp32 #6314: panic_abort when using HTTPUpdateServer

Kasperdelasopa, 2022-02-20:
### Board

ESP32-S2-Solo

### Device Description

ESP32-S2-Solo chip used on a custom board

### Hardware Configuration

static const uint8_t SDA = 42;
static const uint8_t SCL = 2;

### Version

latest master

### IDE Name

Platformio

### Operating System

Win 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

default

### Description

When using the HTTPUpdateServer, i get an panic_abort 

### Sketch

```cpp
HTTPUpdateServer httpUpdater;
WebServer httpServer(8266);

void init(){
    httpUpdater.setup(&httpServer);
    httpServer.begin();
}

loop(){
    httpServer.handleClient();
}
```

### Debug Message

```plain
0x40026db2: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4002c751: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 121
0x400319e5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x40025fa2: esp_flash_erase_region at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/esp_flash_api.c line 541
0x4009b68b: esp_partition_erase_range at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 540
0x40093ae1: EspClass::partitionEraseRange(esp_partition_t const*, unsigned int, unsigned int) at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp line 392
0x40110bca: UpdateClass::_writeBuffer() at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp line 206
0x40110dcd: UpdateClass::write(unsigned char*, unsigned int) at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp line 314
0x4008f901: HTTPUpdateServer::setup(WebServer*, String const&, String const&, String const&)::{lambda()#3}::operator()() const at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/HTTPUpdateServer/src/HTTPUpdateServer.h line 120
0x4008f985: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\manue\.platformio\packages\toolchain-xtensa32s2\xtensa-esp32s2-elf\include\c++\8.4.0\bits/std_function.h line 151
0x4010a6db: std::function ::operator()() const at c:\users\manue\.platformio\packages\toolchain-xtensa32s2\xtensa-esp32s2-elf\include\c++\8.4.0\bits/std_function.h line 260
0x4010a71a: FunctionRequestHandler::upload(WebServer&, String, HTTPUpload&) at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/detail/RequestHandlersImpl.h line 53
0x4010ae12: WebServer::_uploadWriteByte(unsigned char) at c:\users\manue\.platformio\packages\toolchain-xtensa32s2\xtensa-esp32s2-elf\include\c++\8.4.0\bits/unique_ptr.h line 345
0x4010ba89: WebServer::_parseForm(WiFiClient&, String, unsigned int) at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/Parsing.cpp line 445
0x4010c87a: WebServer::_parseRequest(WiFiClient&) at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/Parsing.cpp line 207
0x4010a9f5: WebServer::handleClient() at C:/Users/manue/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp line 312
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2022-02-21:
Hello @Kasperdelasopa 

are you running the [WebUpdater example](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino) or your own code? If you run your own code, did you try the WebUpdater example?

I can run the WebUpdater example on an [ESP32-S2-Saola-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html) board just fine - no crash.

That said, I do get a crash if WiFi has **not** been initialized / connected. Could that be your issue?

Thanks
Felix

Kasperdelasopa, 2022-02-21:
Hello,

The [WebUpdater example](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino) doesn't work for me as well.

Maybe my problems coming from the Platformio configs im using to build:

```
platform = espressif32
platform_packages = 
	  framework-arduinoespressif32@file://C:/Users/manue/Documents/PlatformIO/framework-arduinoespressif32
	  tool-esptoolpy@ ~1.30100.210531
	  tool-mkspiffs@ ~2.230.0
	  toolchain-esp32s2ulp@ ~1.22851.191205
	  toolchain-xtensa32s2@file://C:/Users/manue/Documents/PlatformIO/xtensa-esp32s2-elf

framework = arduino
board = esp32dev
build_flags = -D  CONFIG_FREERTOS_UNICORE -D CONFIG_ESP32_DEFAULT_CPU_FREQ_80
board_build.mcu = esp32s2
board_build.partitions = huge_app.csv
board_build.variant = esp32s2
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.arduino.ldscript = esp32s2_out.ld
```

I found these configs on an other forum, which helped me to use the ESP32-S2-Solo at all.

Kasperdelasopa, 2022-02-21:
I'm an idiot could have seen the issue by myself... sorry

it was this:

`board_build.partitions = huge_app.csv`

Thx all!

closed 2022-02-21 by Kasperdelasopa

felmue, 2022-02-22:
Hello @Kasperdelasopa 

glad to hear you've figured it out.

Thanks
Felix

hientv1999, 2023-06-07:
> I'm an idiot could have seen the issue by myself... sorry
> 
> it was this:
> 
> `board_build.partitions = huge_app.csv`
> 
> Thx all!

Can you explain more about what is wrong with this? Today I have the panic abort on core 0 and I have no idea how to debug it