espressif/arduino-esp32 #6319: esp32s3 compile error when enable BLE

flybeyond, 2022-02-21:
### Board

ESP32S3 Dev Module

### Device Description

Arduino: 1.8.15 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None"

c:/users/xxx/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: libraries\BLE\BLEAdvertising.cpp.o:(.literal._ZN14BLEAdvertising5startEv+0x4): undefined reference to `esp_ble_gap_config_adv_data'

c:/users/xxx/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: libraries\BLE\BLEAdvertising.cpp.o:(.literal._ZN14BLEAdvertising5startEv+0x8): undefined reference to `esp_ble_gap_start_advertising'

c:/users/liu zhen/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: libraries\BLE\BLEAdvertising.cpp.o: in function `BLEAdvertising::start()':

C:\Users\xxx\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEAdvertising.cpp:229: undefined reference to `esp_ble_gap_config_adv_data'

c:/users/xxx/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Liu Zhen\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEAdvertising.cpp:241: undefined reference to `esp_ble_gap_start_advertising'

c:/users/xxx/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\xxx\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEAdvertising.cpp:214: undefined reference to `esp_ble_gap_config_adv_data'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ESP32S3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Hardware Configuration

ESP32S3 Dev Module, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None

### Version

other

### IDE Name

Arduino

### Operating System

N/A

### Flash frequency

40Hz

### PSRAM enabled

yes

### Upload speed

921600

### Description

compile error

### Sketch

```cpp
void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);

  BLEDevice::init("Pseudolite Receiver");
  Serial.println("BLE Name: Pseudolite Receiver");
  Serial.print("BLE Addr: ");
  Serial.println(BLEDevice::getAddress().toString().c_str());
  BLEDevice::setPower(ESP_PWR_LVL_N12);

  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(SERVICE_UUID);

  pRdCharacteristic = pService->createCharacteristic(
                        CHARACTERISTIC_UUID_RD,
                        BLECharacteristic::PROPERTY_READ
                      );
  pRdCharacteristic->setValue("");

  pWrCharacteristic = pService->createCharacteristic(
                        CHARACTERISTIC_UUID_WR,
                        BLECharacteristic::PROPERTY_WRITE
                      );
  pWrCharacteristic->setCallbacks(new MyWrCharacteristicCallbacks());

  pNtCharacteristic = pService->createCharacteristic(
                        CHARACTERISTIC_UUID_NT,
                        BLECharacteristic::PROPERTY_NOTIFY
                      );
  pNtCharacteristic->addDescriptor(new BLE2902());

  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);    // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
}
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

flybeyond, 2022-02-22:
After debuging, the issue locates at libbt.a , esp_gap_ble_api.c. some symbols missed such as  esp_ble_gap_config_adv_data, esp_ble_gap_start_advertising. Can some one help to rebuild the archive file to fix the issue?

me-no-dev, 2022-02-23:
fixed :)

closed 2022-02-23 by me-no-dev

noqnio, 2023-03-19:
hi, how has this been fixed ? I am trying to compile the "pipeline_bt_source" example from esp-adf for ESP32-S3, and I am getting the same errors

BotSpace, 2023-03-30:
Also have this issue! I am using the most current version of the library (2.0.7) but the SerialToSerialBT example fails to compile and give the same error as this.

 I am uploading to ESP32-S3FN8 with the "Adafruit Feather ESP32-S3 No PSRAM" board over USB. All default tools settings.
 
 Any help would be appreciated.

Leadrive, 2023-04-28:
@BotSpace checked BLE 5.0 and BLE 4.2