espressif/arduino-esp32 #6333: Latest update has broken i2c support

Smashcat, 2022-02-23:
### Board

ESP32 custom board.

### Device Description

Custom board with ESP32 VROOM castellated module, although the problem likely affects all ESP32 variants.

### Hardware Configuration

Various devices, including i2C

### Version

latest master

### IDE Name

Arduino

### Operating System

Windows

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Since the latest update (that Arduino pulls in automatically) i2C no longer works on the module.

### Sketch

```cpp
/* ---------------------------------------------------------------- /
// Arduino I2C Scanner
// Re-writed by Arbi Abdul Jabbaar
// Using Arduino IDE 1.8.7
// Using GY-87 module for the target
// Tested on 10 September 2019
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
/ ---------------------------------------------------------------- */

#include <Wire.h> //include Wire.h library

void setup()
{
  Wire.begin(); // Wire communication begin
  Wire.setClock(100000);
  Serial.begin(115200); // The baudrate of Serial monitor is set in 9600
  while (!Serial); // Waiting for Serial Monitor
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address; //variable for error and I2C address
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for (address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.print(address, HEX);
      Serial.println("  !");
      nDevices++;
    }
    else if (error == 4)
    {
      Serial.print("Unknown error at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000); // wait 5 seconds for the next I2C scan
}
```

### Debug Message

```plain
"No devices found"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Smashcat, 2022-02-23:
I pulled an older version of the core from an old laptop that had not auto-updated, and the code all compiles and works fine again. Seems like this latest update broke something in the i2C library.

me-no-dev, 2022-02-23:
I am using the following function to scan I2C and do not have any issues finding devices. It's not any different than yours:
```cpp
void scanI2C(){
  uint8_t devices, error, address;
  Serial.println("Scanning for I2C devices...");
  devices = 0;
  for(address = 0x01; address < 0x7f; address++){
    Wire.setClock(100000);
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      devices++;
      Serial.printf("I2C device found at address 0x%02X\n", address);
    } else if(error == 4){
      Serial.printf("Error at address 0x%02X\n", address);
    } else if(error == 5){
      Serial.printf("Timeout at address 0x%02X\n", address);
    } else if(error != 2){
      Serial.printf("Unknown error %u at address 0x%02X\n", error, address);
    }
  }
  Serial.printf("%u I2C devices found\n", devices);
}
```

me-no-dev, 2022-02-23:
Which version is that "older" core? We have recently switched to use the ESP-IDF driver for I2C

Smashcat, 2022-02-23:
I'm not sure which version was on my old laptop. I believe it was from around November last year. 

The hardware I'm loading the code on has not changed in 6 months. It was working fine up until I tried to load the exact same firmware onto some more boards, and the code could no longer see any i2C devices (e.g. PCF8563 clock chip). As soon as I recompiled with the older core, it worked again as usual.

Jason2866, 2022-02-24:
I2C works for project Tasmota. Currently using a build which is a few commits newer than the released v2.0.2. Tasmota is installed in many thousands devices where a lot I2C sensors are connected. If there would be a general problem with I2C we would have many issues opened in Tasmota github. There is *no* issue open with I2C.

Smashcat, 2022-02-24:
Ok thanks, I'll remain on the older version for now and test the next release of the core to see if it's resolved. Is there any way to prevent Arduino from auto-updating?

big-penguin, 2022-03-04:
> I2C works for project Tasmota. Currently using a build which is a few commits newer than the released v2.0.2. Tasmota is installed in many thousands devices where a lot I2C sensors are connected. If there would be a general problem with I2C we would have many issues opened in Tasmota github. There is _no_ issue open with I2C. The new I2C solution is more restrictive for correct implementation. Probably the PCF8563 library has quirks

You mention the new I2C implementation being "more restrictive", is there any documentation on what changed? I'm having the same problem with some I2C code I wrote to interface with a servo driver (not using Wire.h).

VojtechBartoska, 2022-03-04:
> You mention the new I2C implementation being "more restrictive", is there any documentation on what changed? I'm having the same problem with some I2C code I wrote to interface with a servo driver [...]

You can find newly written documentation about I2C here: https://docs.espressif.com/projects/arduino-esp32/en/latest/api/i2c.html

Smashcat, 2022-03-04:
Is there any documentation on what has changed between versions that is breaking compatibility? As Arduino auto-updates to this version, it means I'm currently having to keep copying the older core across so I can run my code without all of the i2c devices failing to work.

Jason2866, 2022-03-04:
The relevant PR https://github.com/espressif/arduino-esp32/pull/5683
In general use of i2c has not changend. Many librarys do have wrong implementations.
For example https://github.com/espressif/arduino-esp32/issues/6373#issuecomment-1059196867

big-penguin, 2022-03-05:
Thank you for your answer. 

All I can find relating to this issue is "your Arduino library is written wrong use a different one". However, I'm not using an Arduino library at all for my I2C, only ESP-IDF's native I2C functions, and am having the same issue. As far as I can tell, everything I have written matches up with the ESP-IDF documentation and provided I2C examples. The code works as expected on an older version of esp32-arduino (1.0.4) but fails on the latest version. Any pointers on what I (and all these library devs apparently) are doing wrong would be appreciated.
[i2c_snippet.txt](https://github.com/espressif/arduino-esp32/files/8190060/i2c_snippet.txt)

tsctrl, 2022-03-05:
upgrading from 4.3.2 to 4.4 causing the i2c not working. There is details logged, but does not indicates error during initialization. Somehow the init pass with out error but communication to i2c failed. i will try again and provide the logs.

me-no-dev, 2022-03-05:
what devices are you guys using?

Jason2866, 2022-03-05:
We use the ds3231 too. Using our own (small) customized [driver](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_33_ds3231.ino) it works well with esp8266, esp32, esp32c3, esp32s2

tsctrl, 2022-03-06:
Hi @me-no-dev, 4.4 arduino as component

Wire.begin(SDA, SCL, 100000);

[    31][E][esp32-hal-i2c-slave.c:224] i2cSlaveInit(): invalid pins sda=22, scl=-96
[    31][E][Wire.cpp:167] begin(): Slave Init ERROR

me-no-dev, 2022-03-06:
try `Wire.begin((int)SDA, (int)SCL, 100000);`

Smashcat, 2022-03-14:
Is there any update on this being fixed? Also tested on an HDC1080 sensor, and it doesn't work with the latest core. Seems like it's completely broken the Arduino i2C wire library.

VojtechBartoska, 2022-03-23:
@Smashcat Did you solve your problem?

Smashcat, 2022-03-23:
@VojtechBartoska Not really. I think the developers have just decided it's not their fault that things broke after they made changes, and everyone else has just been using it wrong so they won't fix it. I'm just keeping a copy of the last working release for now, so I can replace any future updates that break other things. This isn't the first time they've broken the Arduino libs with an update, which is why I guessed what had happened when previously working code suddenly stopped working. I think the previous time they broke it, the build process failed completely, and so people had to link to the development branch while they fixed it...

VojtechBartoska, 2022-03-24:
@Smashcat It's actually not exactly as this. I2C was refactored too use high level ESP-IDF API. this brings a lot of benefits and it's more solid solution for Arduino core also to keep it up to date. Unfortunately some third-party libraries which are not maintained by us will need some changes to work with this refactored API and we are not able to cover this issue. We wrote an extensive documentation which was also mentioned above and we will try to help you with your problem as much as possible. 

So if you can provide more info what exactly is not working, what you have tried and so on, it will help to move this issue forward.

Smashcat, 2022-03-24:
> @Smashcat It's actually not exactly as this. I2C was refactored too use high level ESP-IDF API. this brings a lot of benefits and it's more solid solution for Arduino core also to keep it up to [...]

I think the issue is that the Arduino IDE automatically updates the ESP core when it checks at startup. There is no indication this has happened to the user, and so people will just notice that things are suddenly broken. It's not obvious why they're broken, or that they should search through the documentation here to find the cause. The issue is that it breaks the standard Ardunio i2C library - at least in version 1.8.16 that I have here (from sept 2021 I think). Maybe it would have been better to keep the old version as an option - "i2c-compat" or something so people could migrate, and also should some compile time errors, or something to indicate there's a problem with how the new library is being used if possible? Anyway, I am just staying with the old version - it seems to work fine with i2c, so not sure why I'd need to change over to a new version. Thanks.

Avenitos, 2022-04-06:
> try `Wire.begin((int)SDA, (int)SCL, 100000);`

this work!

VojtechBartoska, 2022-04-11:
Can I consider this solved?

VojtechBartoska, 2022-04-22:
I'm closing this issue as solved.

If you still face some issues, please reopen it.

Thanks!

closed 2022-04-22 by VojtechBartoska