espressif/arduino-esp32 #6334: Provisioning never finds any networks anymore

jwktje, 2022-02-23:
### Board

ESP32 Dev Module

### Device Description

It is on a custom board but I can also replicate the issue with other breakouts with a WROOM module.

### Hardware Configuration

A button on pin 18
A buzzer on pin 32
A Neopixel on 16

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

MacOS 12.1

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Whenever I try provisioning via esp_prov.py or the SoftAp iOS app, I can never get any networks from the scan.
When I manually type out the SSID and password, the provisioning works fine.

But whatever I try, the scan always results in 0 APs

### Sketch

```cpp
#include "WiFiProv.h"
#include "WiFi.h"
void SysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        Serial.print("\nConnected IP address : ");
        Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
        break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        Serial.println("\nDisconnected. Connecting to the AP again... ");
        break;
    case ARDUINO_EVENT_PROV_START:
        Serial.println("\nProvisioning started\nGive Credentials of your access point using \" Android app \"");
        break;
    case ARDUINO_EVENT_PROV_CRED_RECV: { 
        Serial.println("\nReceived Wi-Fi credentials");
        Serial.print("\tSSID : ");
        Serial.println((const char *) sys_event->event_info.prov_cred_recv.ssid);
        Serial.print("\tPassword : ");
        Serial.println((char const *) sys_event->event_info.prov_cred_recv.password);
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL: { 
        Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
        if(sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR) 
            Serial.println("\nWi-Fi AP password incorrect");
        else
            Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");        
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
        Serial.println("\nProvisioning Successful");
        break;
    case ARDUINO_EVENT_PROV_END:
        Serial.println("\nProvisioning Ends");
        break;
    default:
        break;
    }
}

void setup() {
  Serial.begin(115200);
  WiFi.onEvent(SysProvEvent);
  WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, "abcd1234", "Prov_123");
}

void loop() {
}
```

### Debug Message

```plain
Relevant CLI output

jwktje@JanWillemsMBP2 esp_prov % python3 esp_prov.py --transport SoftAp --sec_ver 1 --pop wakeup
Connecting to 192.168.4.1:80

==== Starting Session ====
==== Session Established ====

==== Scanning Wi-Fi APs ====
++++ Scan process executed in 2.559149742126465 sec
++++ Scan results : 0

++++ Scan finished in 2.5976369380950928 sec
No APs found!
jwktje@JanWillemsMBP2 esp_prov % python3 esp_prov.py --transport SoftAp --sec_ver 1 --pop wakeup
Connecting to 192.168.4.1:80

==== Starting Session ====
Server responded with error code 404
Failed to establish session. Ensure that security scheme and proof of possession are correct
---- Error in establishing session ----
jwktje@JanWillemsMBP2 esp_prov % python3 esp_prov.py --transport SoftAp --sec_ver 1 --pop wakeup
Connecting to 192.168.4.1:80

==== Starting Session ====
==== Session Established ====

==== Scanning Wi-Fi APs ====
++++ Scan process executed in 2.5476889610290527 sec
++++ Scan results : 0

++++ Scan finished in 2.5605411529541016 sec
No APs found!
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jwktje, 2022-02-23:
I feel like I might have found a solution already. I tried experimenting more with the Flash options. 
It looks like the important part is that you set:
```
Arduino Runs On: Core 1
Events Run On: Core 0
```

Maybe it was a lucky accident so I'll try and test some more. But it would be great if someone can confirm that this setting is indeed needed to make provisioning work. I would love to understand why. And maybe it would be something that could be added to the documentation to clarify this for others with the same issue as me.

VojtechBartoska, 2022-04-11:
Hello @jwktje, can we consider this as solved?

jwktje, 2022-04-11:
Absolutely

closed 2022-04-11 by jwktje