espressif/arduino-esp32 #6345: LITTLEFS 10 times slower than SPIFFS on ESP32

fred9999, 2022-02-24:
### Board

ESP32 Dev Module 4MB

### Device Description

The ESP32 board info:
Board: "ESP32 Dev Module"
Upload Speed: "921600"
CPU Frequency: "240MHz (WiFi/BT"
Flash Frequency: "80MHz"
Flash Mode: "QIO"
Flash Size: "4MB (32Mb)"
Partition Scheme: "No OTA (2MB APP 2MB SPIFFS)"
Core Debug Level: "None"
PSRAM: "Disabled"
Arduino Runs On: Core1
Events Run On: Core1
Port: "COM9"

### Hardware Configuration

no external devices

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Win10 64bit

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi,

I initially posted it on forum.arduino.cc but got no replies there. Maybe that was the wrong place to post it.

I have a project in mind where I have to write sensor data into file in ESP32 flash memory (yes, I am aware of 10000 writes limit). As I understood SPIFFS is deprecated and LittleFS would be the way to go.

I decided to do a little speed test first. I uploaded the Arduino IDE ESP32 LITTLEFS_test.ino sketch onto the regular ESP32 Dev Module and added a couple of lines of code into loop to measure the append time of text "qwerty" to the file data.txt on ESP32 file system. I uploaded the file via ESP32 Sketch Data Upload tool with LittleFS, that seems to format the flash for using LittleFS.

To my great surprise the append function run time was 20-30ms in the beginning and when the data.txt file grew to a few hundred bytes it became 50-70ms. And there is very odd behavior - every 15th time the append operation takes 250-350ms. That does not depend on the append function call period. Extra long append time occurs every 15th time would the call period be 1, 2 or 3 seconds.

When I run the same script modified to use SPIFFS the append function run time is steady 5ms.

I tested the sketch on 2 different boards with the same results. I also uninstalled ESP32 boards from Arduino IDE and reinstalled. No difference.

Thank you in advance to get the LittleFS working.

### Sketch

```cpp
#include "FS.h"
#include "LittleFS.h"

#define FORMAT_LITTLEFS_IF_FAILED true

unsigned long lastAppendMillis = 0; // append function call period, ms
unsigned long beginAppendMillis = 0; // append function start time, ms
unsigned long appendMillis = 0; // append function end time, ms

void setup(){
    Serial.begin(115200);

    if(!LittleFS.begin(FORMAT_LITTLEFS_IF_FAILED)){
        Serial.println("LittleFS Mount Failed");
        return;
    }

    Serial.println("File system info:"); 
    Serial.print("  Total space:      ");
    Serial.print(LittleFS.totalBytes());
    Serial.println("byte");
 
    Serial.print("  Total space used: ");
    Serial.print(LittleFS.usedBytes());
    Serial.println("byte"); 
    Serial.println();
 
    listDir(LittleFS, "/", 3);
}

void loop(){

	if (millis() - lastAppendMillis > 1000) { 
		lastAppendMillis = millis();

		Serial.println("###########################");
		Serial.println("Append qwerty");

		beginAppendMillis = millis(); 

		appendFile(LittleFS, "/data.txt", "qwerty\r\n");	

		appendMillis = millis() - beginAppendMillis;
		Serial.print("appendMillis = ");
		Serial.print(appendMillis);	
		Serial.println(" ms");
	}
}

// LittleFS functions
void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void appendFile(fs::FS &fs, const char * path, const char * message){ 
    Serial.printf("Appending to file: %s\r\n", path);
    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("- failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("- message appended");
    } else {
        Serial.println("- append failed");
    }
    file.close();
}
```

### Debug Message

```plain
no special debug messages
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-02-24:
@lorol Any ideas?

bxparks, 2022-03-10:
I can confirm that my application sees a 10X-15X **slower** performance of LittleFS on the ESP32 compared to the ESP8266.

I created a minimal reproducible example (see below) which shows roughly:

* 2X slower `File::read()` performance on ESP32 compared to ESP8266
* 6-10X slower `File::seek()` performance on the ESP32 compared to ESP8266

**Minimal Reproducible Example**

```C++
#include <Arduino.h>

#include <LittleFS.h>
#define FILE_SYSTEM LittleFS

//-----------------------------------------------------------------------------

void createFile(const char* fileName, uint32_t fileSize) {
  unsigned long startMillis = millis();

  File f = FILE_SYSTEM.open(fileName, "w");
  for (uint32_t i = 0; i < fileSize; i++) {
    f.write((uint8_t) i);
  }
  f.close();

  unsigned long elapsedMillis = millis() - startMillis;
  Serial.printf("%-10s(%-11s, %6lu): %6lu millis\n",
      "createFile", fileName, (unsigned long) fileSize, elapsedMillis);
}

void readFile(const char* fileName, uint32_t fileSize) {
  unsigned long startMillis = millis();

  File f = FILE_SYSTEM.open(fileName, "r");
  for (uint32_t i = 0; i < fileSize; i++) {
    uint8_t c = f.read();
    if (c != (uint8_t) i) {
      Serial.printf("Unexpected byte at index %lu\n", (unsigned long) i);
    }
  }
  f.close();

  unsigned long elapsedMillis = millis() - startMillis;
  Serial.printf("%-10s(%-11s, %6lu): %6lu millis\n",
      "readFile", fileName, (unsigned long) fileSize, elapsedMillis);
}

void seekFile(
    const char* fileName,
    uint32_t fileSize,
    uint16_t numSeeks,
    uint32_t readSize
) {
  unsigned long startMillis = millis();

  File f = FILE_SYSTEM.open(fileName, "r");
  for (uint16_t i = 0; i < numSeeks; i++) {
    uint32_t offset = random(fileSize);
    f.seek(offset);
    for (uint32_t i = offset; i < offset + readSize && i < fileSize; i++) {
      uint8_t c = f.read();
      if (c != (uint8_t) i) {
        Serial.printf("Unexpected byte at index %lu\n", (unsigned long) i);
      }
    }
  }
  f.close();

  unsigned long elapsedMillis = millis() - startMillis;
[... 111 lines cut ...]
seekFile  (/file1.txt , 1000,   30):    316 millis
seekFile  (/file2.txt , 1000,   30):    318 millis
seekFile  (/file3.txt , 1000,   30):    899 millis
seekFile  (/file4.txt , 1000,   30):   1466 millis
seekFile  (/file5.txt , 1000,   30):   1970 millis
seekFile  (/file6.txt , 1000,   30):   2386 millis
seekFile  (/file0.txt , 1000,  100):    947 millis
seekFile  (/file1.txt , 1000,  100):    972 millis
seekFile  (/file2.txt , 1000,  100):    991 millis
seekFile  (/file3.txt , 1000,  100):   1587 millis
seekFile  (/file4.txt , 1000,  100):   2140 millis
seekFile  (/file5.txt , 1000,  100):   2632 millis
seekFile  (/file6.txt , 1000,  100):   3163 millis
==== Done
```

**Environments**

* ESP8266 Core 3.0.2
    * D1Mini
    * `esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M2M,ip=lm2f,dbg=Disabled,
lvl=None____,wipe=none,baud=921600`
* ESP32 Core 2.0.2
    * ESP32 Dev Kit v1
    * `esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none`

guestisp, 2022-05-06:
I'm experiencing exactly the same, with arduino-esp32 2.0.2 and 2.0.3

As workound, downgrading to arduino-esp32 1.0.6 and adding https://github.com/lorol/LITTLEFS LittleFS library seems to have fixed the performance issue.

bxparks, 2022-05-06:
@VojtechBartoska: Who is the "you" that you are asking? In my case, for various reasons, I don't have the environment and hardware to re-run the example that I uploaded, at least not for a while. But I uploaded a complete reproducible example, so anyone else can.

VojtechBartoska, 2022-05-09:
Adding this to our Roadmap.

guestisp, 2022-05-09:
Thank you so much from me too

zekageri, 2022-09-22:
Almost the same results for me on ESP32-Wrover-E ( 8mb psram 16mb flash )
I'm testing my ESP32s because i got an error last time out of the blue with this message: `assert failed: block_trim_free heap_tlsf.c:377 (block_is_free(block) && "block must be free")`

So i'm testing external ram, internal ram and flash too. Here is my results:

```C
******************** External ram test ********************
Read Speed 8bit ArraySize    1kb  time: 0.0 sec 12.5 mb/sec
Read Speed 8bit ArraySize    2kb  time: 0.0 sec 12.6 mb/sec  
Read Speed 8bit ArraySize    4kb  time: 0.0 sec 12.6 mb/sec  
Read Speed 8bit ArraySize    8kb  time: 0.0 sec 12.6 mb/sec  
Read Speed 8bit ArraySize   16kb  time: 0.1 sec 12.6 mb/sec  
Read Speed 8bit ArraySize   32kb  time: 0.1 sec 12.2 mb/sec  
Read Speed 8bit ArraySize   64kb  time: 0.4 sec 8.6 mb/sec  
Read Speed 8bit ArraySize  128kb  time: 0.7 sec 8.6 mb/sec  
Read Speed 8bit ArraySize  256kb  time: 1.5 sec 8.6 mb/sec  
Read Speed 8bit ArraySize  512kb  time: 0.6 sec 8.6 mb/sec  
Read Speed 8bit ArraySize 1024kb  time: 1.2 sec 8.6 mb/sec  
Read Speed 8bit ArraySize 2048kb  time: 2.3 sec 8.6 mb/sec  
Read Speed 8bit ArraySize 4000kb  time: 4.6 sec 8.5 mb/sec  

Read Speed 16bit ArraySize    1kb  time: 0.0 sec 22.3 mb/sec
Read Speed 16bit ArraySize    2kb  time: 0.0 sec 22.5 mb/sec  
Read Speed 16bit ArraySize    4kb  time: 0.0 sec 22.5 mb/sec
Read Speed 16bit ArraySize    8kb  time: 0.0 sec 22.5 mb/sec  
Read Speed 16bit ArraySize   16kb  time: 0.0 sec 22.5 mb/sec  
Read Speed 16bit ArraySize   32kb  time: 0.1 sec 21.8 mb/sec  
Read Speed 16bit ArraySize   64kb  time: 0.3 sec 12.4 mb/sec  
Read Speed 16bit ArraySize  128kb  time: 0.5 sec 12.3 mb/sec  
Read Speed 16bit ArraySize  256kb  time: 1.0 sec 12.3 mb/sec  
Read Speed 16bit ArraySize  512kb  time: 0.4 sec 12.3 mb/sec  
Read Speed 16bit ArraySize 1024kb  time: 0.8 sec 12.3 mb/sec  
Read Speed 16bit ArraySize 2048kb  time: 1.6 sec 12.3 mb/sec  
Read Speed 16bit ArraySize 4000kb  time: 3.2 sec 12.4 mb/sec  

Read Speed 32bit ArraySize    1kb  time: 0.0 sec 43.7 mb/sec
Read Speed 32bit ArraySize    2kb  time: 0.0 sec 44.6 mb/sec  
Read Speed 32bit ArraySize    4kb  time: 0.0 sec 44.6 mb/sec
Read Speed 32bit ArraySize    8kb  time: 0.0 sec 44.7 mb/sec  
Read Speed 32bit ArraySize   16kb  time: 0.0 sec 44.7 mb/sec  
Read Speed 32bit ArraySize   32kb  time: 0.0 sec 42.1 mb/sec  
Read Speed 32bit ArraySize   64kb  time: 0.2 sec 17.0 mb/sec  
Read Speed 32bit ArraySize  128kb  time: 0.4 sec 16.9 mb/sec  
Read Speed 32bit ArraySize  256kb  time: 0.7 sec 16.9 mb/sec  
Read Speed 32bit ArraySize  512kb  time: 0.3 sec 16.9 mb/sec  
Read Speed 32bit ArraySize 1024kb  time: 0.6 sec 16.9 mb/sec  
Read Speed 32bit ArraySize 2048kb  time: 1.2 sec 16.9 mb/sec  
Read Speed 32bit ArraySize 4000kb  time: 2.3 sec 16.9 mb/sec  
****************** External ram test DONE ******************

******************** Internal ram test ********************
Read Speed 8bit ArraySize    1kb  time: 0.0 sec 12.6 mb/sec  
Read Speed 8bit ArraySize    2kb  time: 0.0 sec 12.6 mb/sec  
Read Speed 8bit ArraySize    4kb  time: 0.0 sec 12.6 mb/sec  
Read Speed 8bit ArraySize    8kb  time: 0.0 sec 12.5 mb/sec  
Read Speed 8bit ArraySize   16kb  time: 0.1 sec 12.6 mb/sec  
Read Speed 8bit ArraySize   32kb  time: 0.1 sec 12.2 mb/sec  
Read Speed 8bit ArraySize   64kb  time: 0.4 sec 8.6 mb/sec  
Read Speed 8bit ArraySize  128kb  time: 0.7 sec 8.6 mb/sec  

Read Speed 16bit ArraySize    1kb  time: 0.0 sec 22.6 mb/sec
Read Speed 16bit ArraySize    2kb  time: 0.0 sec 22.7 mb/sec
Read Speed 16bit ArraySize    4kb  time: 0.0 sec 22.7 mb/sec  
Read Speed 16bit ArraySize    8kb  time: 0.0 sec 22.3 mb/sec  
Read Speed 16bit ArraySize   16kb  time: 0.0 sec 22.5 mb/sec  
Read Speed 16bit ArraySize   32kb  time: 0.1 sec 21.8 mb/sec  
Read Speed 16bit ArraySize   64kb  time: 0.3 sec 12.4 mb/sec  
Read Speed 16bit ArraySize  128kb  time: 0.5 sec 12.3 mb/sec  

Read Speed 32bit ArraySize    1kb  time: 0.0 sec 45.2 mb/sec
[... 116 lines cut ...]
            rm = (int*)ps_malloc(ramsize);
        }else{
            rm = (int*)malloc(ramsize);
        }

        int iters = 10;  // Just enuff to boot the dog
        if (rs[a] < 512) iters = 50;
        double st = getTime();
        for (int b = 0; b < iters; b++)
            for (int c = 0; c < ramsize / 4; c++)
                xx |= rm[c];
        st = getTime() - st;
        vTaskDelay(1);  // Dog it!
        double speed = ((double)(iters * ramsize) / (1024 * 1024)) / (st);
        printf(" time: %2.1f sec %2.1f mb/sec  \n", st, speed);
        free(rm);
    }
    if( isExtRam ){
        printf("****************** External ram test DONE ******************\n\n");
    }else{
        printf("****************** Internal ram test DONE ******************\n\n");
    }
    return xx;
}
```

txf-, 2023-03-13:
Just as un update to another thing that I have noticed, which may be related to this problem.

I'm using [this library](https://github.com/siara-cc/sqlite_micro_logger_arduino) to store log entries in a partition. I have noticed that vs SPIFFS not only do row inserts become slower, but that for the same amount of data on LittleFS, the file size has grown 3x the size when compared to SPIFFS.

guestisp, 2023-06-30:
any update on this ?

zekageri, 2023-06-30:
Waiting

hitecSmartHome, 2024-06-26:
I'm experiencing LittleFS slowdown out of the blue.
IDF 4.4.7
Arduino 2.0.17

Suresh6060, 2024-08-23:
> Just as un update to another thing that I have noticed, which may be related to this problem.
> 
> I'm using [this library](https://github.com/siara-cc/sqlite_micro_logger_arduino) to store log entries in a partition. I have noticed that vs SPIFFS not only do row inserts become slower, but that for the same amount of data, on LittleFS the used space has grown 3x when compared to SPIFFS.

Does it mean using SPIFFS.h is better than LittleFS.h ?

lbernstone, 2024-08-23:
I just did some quick testing on 3.0.4.  LittleFS writes do get a bit slower as the disk gets fragmented (goes from 96ms to 200ms for open/write 4096/close), but is comparable to SPIFFS on a clean FS. SPIFFS does not get slower as it gets more fragmented, but it never says that it is out of space, so when it gets to the limit it dramatically slows down (1500ms for a write), but it will continue behaving as though it is writing, even though the disk is clearly full. 
Based on the other benefits of LittleFS over SPIFFS, I'd say this speed issue will generally be an acceptable tradeoff, so SPIFFS is not a recommended filesystem.

Suresh6060, 2024-08-24:
> I just did some quick testing on 3.0.4. LittleFS writes do get a bit slower as the disk gets fragmented (goes from 96ms to 200ms for open/write 4096/close), but is comparable to SPIFFS on a clean [...]

Thank you.

Parsaabasi, 2025-01-10:
Hello, 

I close this since this report contains the release we no longer support. Please try the new versions and in case the issue persists, feel free to reopen it. 

Thanks

closed 2025-01-10 by Parsaabasi

hitecSmartHome, 2025-01-10:
Thanks. Will reopen it because it still exists on latest arduino

garudaonekh, 2025-04-05:
Seek is terribly slow on ESP32. Any new input on this? THanks

zekageri, 2025-05-13:
We are on `IDF 5.3.2.250210` and LittleFS is still slow A.F

x29a, 2025-07-21:
@hitecSmartHome i assume you didnt create a new issue because this one already exists, or? https://github.com/espressif/arduino-esp32/issues/7337

hitecSmartHome, 2025-07-21:
@x29a because @Parsaabasi said

>> Please try the new versions and in case the issue persists, feel free to reopen it.

x29a, 2025-07-21:
ok, then i misunderstood your [next comment](https://github.com/espressif/arduino-esp32/issues/6345#issuecomment-2582713709) "Thanks. Will reopen it because it still exists on latest arduino".

so issue is solved for you?