espressif/arduino-esp32 #6347: HTTP OTA update feature (not with DEFINES on compile time) (IDFGH-6824)

gonzabrusco, 2022-02-24:


SuGlider, 2022-02-25:
> I tried defining CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP and CONFIG_OTA_ALLOW_HTTP but it fails with error:
> 
> `esp_https_ota: No option for server verification is enabled in esp_http_client config.`
> 
> It seems that when using Arduino core the IDF comes precompiled? (the error appears on UART0 when it shouldn't !!).

Yes, IDF is precompiled for Arduino. 
Arduino has a lot of precompiled libs and header files from IDF.

This is done with a separated project:
https://github.com/espressif/esp32-arduino-lib-builder

Using Arduino Lib Builder, it's possible to create a particularly configured Arduino Core.

SuGlider, 2022-02-25:
> (the error appears on UART0 when it shouldn't !!).

@gonzabrusco 

With Arduino Lib Builder you can also disable all debug messages and also make the boot process to be silent (no UART output).

This can be achieved by configuring it with ```idf.py menuconfig``` options.

In order to enter in the menu, you must uncomment this line:
https://github.com/espressif/esp32-arduino-lib-builder/blob/master/build.sh#L36

gonzabrusco, 2022-02-25:
> Yes, IDF is precompiled for Arduino. Arduino has a lot of precompiled libs and header files from IDF.
>
> This is done with a separated project: https://github.com/espressif/esp32-arduino-lib-builder [...]

Wow very nice. I didn't know that. Thanks for the help my friend. Closing this.

closed 2022-02-25 by gonzabrusco