espressif/arduino-esp32 #6348: ESP32 DEVKITC WROOM 32U cannot connect to home WiFi

audreylim, 2022-02-25:
### Board

ESP Dev Module

### Device Description

ESP32-DEVKITC-32U

### Hardware Configuration

Connected to RGB OLED with SPI interface, PIN 18 is connected

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Cannot connect to home wifi, but can connect to personal hotspot if the phone is close enough.

I have an ESP8266 which can connect to home wifi with no issues.

### Sketch

```cpp
Serial.begin(9600); Serial.println();
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  Serial.println("Connecting to Wi-Fi");

  while (WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print(".");
  }
```

### Debug Message

```plain
.......................................................................
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-25:
enable debug in the board menu and provide the logs that will come up in the console :)

audreylim, 2022-02-25:
Hmm,

```
Connecting to Wi-Fi

....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
...
```

me-no-dev, 2022-02-25:
could you be having some sort of antenna issues with that esp32 module? What is the RSSI reported from WiFi Scan?

audreylim, 2022-02-26:
Is this normal?
```
scan start
scan done
5 networks found
1: SSID (-80)*
```

It might be due to the 32U module I got which apparently doesn't have an antenna. I'll check again when my other ESP32 modules arrive next week.

yongkimleng, 2022-02-27:
your module (assuming its the same module I see on Amazon) has an on-board antenna. You might have a lemon.

VojtechBartoska, 2022-04-11:
I'm closing this one as duplicate. Please track issue #6379

closed 2022-04-11 by VojtechBartoska