espressif/arduino-esp32 #6349: WiFiMulti drops WiFi connection every 5 minutes

gjt211, 2022-02-26:
### Board

ESP32 Dev module

### Device Description

Generic ESP32 Dev module

### Hardware Configuration

No

### Version

latest master

### IDE Name

Arduino IDE 1.18.15

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Using the example code, WiFi disconnects and reconnects every 5 minutes. I have looked in both source and header file for wifimulti and can't identify a location where this timeout is set.
Wrapping
```
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
```
in
```
if (!WiFi.isConnected()){
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}
```
makes the problem dissapear.

### Sketch

```cpp
/*
 *  This sketch trys to Connect to the best AP based on a given list
 *
 */

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

void setup()
{
    Serial.begin(115200);
    delay(10);

    wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");
    wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
    wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");

    Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop()
{
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}
```

### Debug Message

```plain
No errors output
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-04:
any update on this?

VojtechBartoska, 2022-05-18:
Due to no response, I'm closing this issue as expired.

If needed, you can reopen it.

Thanks for understanding.