espressif/arduino-esp32 #6352: ESP32 ExternalWakeUp example won't compile on ESP32C3 Mini

kriswiner, 2022-02-27:
### Board

ESP32 C3 Mini

### Device Description

Custom dev board with all GPIO exposed at the edge.

### Hardware Configuration

Native USB on GPIO 18/19

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Trying to compile ExternalWakeUp example for ESP32 C3 Dev Module and I see:

esp_sleep_enable_ext0_wakeup' was not declared in this scope

### Sketch

```cpp
#define BUTTON_PIN_BITMASK 0x200000000 // 2^33 in hex

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

void setup(){
[... 36 lines cut ...]
### Debug Message

```plain
esp_sleep_enable_ext0_wakeup' was not declared in this scope
```

### Other Steps to Reproduce

Tried version 2.0.0 and still get:

C:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\riscv32-esp-elf-gcc\\gcc8_4_0-esp-2021r1/bin/riscv32-esp-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-dev-2313-gc69f0ec32\"" -DESP_PLATFORM "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/config" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/newlib/platform_include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/freertos/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/freertos/port/riscv/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_hw_support/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_hw_support/include/soc" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_hw_support/include/soc/esp32c3" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3/private_include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/heap/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/log/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/lwip/include/apps" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/lwip/include/apps/sntp" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/lwip/lwip/src/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/lwip/port/esp32/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/lwip/port/esp32/include/arch" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/soc/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/soc/esp32c3" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/soc/esp32c3/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/hal/esp32c3/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/hal/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/hal/platform_port/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_rom/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_rom/esp32c3" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_rom/include/esp32c3" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_common/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_system/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_system/port/soc" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_system/port/include/riscv" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_system/port/public_compat" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/riscv/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/driver/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/driver/esp32c3/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_pm/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_ringbuf/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/efuse/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/efuse/esp32c3/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/vfs/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_wifi/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_event/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_netif/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_eth/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/tcpip_adapter/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_phy/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_phy/esp32c3/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/app_trace/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_timer/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/esp_ipc/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.0/tools/sdk/esp32c3/include/mbedtls/port/include" "-IC:\\Users\\kris\\AppData\\Local\\Arduino1 [... cut ...]
C:\Users\kris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\ESP32\examples\DeepSleep\ExternalWakeUp\ExternalWakeUp.ino:69:3: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
   esp_sleep_enable_ext0_wakeup(BUTTON_PIN_BITMASK,1); //1 = High, 0 = Low
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
   esp_sleep_enable_wifi_wakeup
exit status 1
'esp_sleep_enable_ext0_wakeup' was not declared in this scope

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2022-02-27:
Hello @kriswiner 

I think this is not supported on the C3 - see [here](https://github.com/espressif/arduino-esp32/issues/5194).

Thanks
Felix

kriswiner, 2022-02-27:
Thanks, verified timed deep sleep does work.

closed 2022-02-27 by kriswiner