espressif/arduino-esp32 #6366: BLE Simultaneous Scanning/Multiple Advertisement Not working

biiinoida, 2022-03-02:
### Board

ESP32-C3 M5-Stamp

### Device Description

None

### Hardware Configuration

Default

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

I have tried to integrate Multiple Advertsiment Example and BLE Scanner example to have a 4-simultaneous Advertisements iBeacon Adverstiment and iBeacon Scanner. Simultaneous operation is not working.

if the scanner code is called first then only scanner works, no iBeacon advertisement
if the advertiser code is called first then only the advertiser code works no iBeacon scan is reported.

I have tried different advertisement interval and scan interval/window settings but unable to get it work

### Sketch

```cpp
BLEScan *pBLEScan;
BLEMultiAdvertising advert(4); // max number of advertisement data

static const uint8_t ibeacon_adv_payload[] = {
    0x02, 0x01, 0x04, // 0x06,
    0x1A, 0xFF,       // BeaconLen and MFG FeildType
    0xFF, 0xFF,       // MFGID
    0x02, 0x15,       // BeaconType & len
};
static uint8_t _adv_data[4][30];

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advDev)
    {
        Serial.printf("Beacon At RSSI: %d\n", advDev.getRSSI());
    }
};

void setup_ble_arduino(uint32_t uid)
{
    //
    // setup ble
    //
    BLEDevice::init("");
    BLEDevice::setPower(ESP_PWR_LVL_P9, ESP_BLE_PWR_TYPE_ADV);
    //
    // setup multiple advertisement
    //
    uint8_t addr_legacy[6] = {0xC0, 0xE0, 0x00, 0x00, 0x00, 0x01};
    memcpy(&addr_legacy[2], &uid, sizeof(uid));
    esp_ble_gap_ext_adv_params_t legacy_adv_params = {
        .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_NONCONN,
        .interval_min = 0x190, // 250ms
        .interval_max = 0x190, // 250ms
        .channel_map = ADV_CHNL_ALL,
        .own_addr_type = BLE_ADDR_TYPE_RANDOM,
        .filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
        .primary_phy = ESP_BLE_GAP_PHY_1M,
        .max_skip = 10,
        .secondary_phy = ESP_BLE_GAP_PHY_1M,
        .scan_req_notif = false,
    };
    for (uint8_t u8_handle = 0; u8_handle < 4; u8_handle++) {
        // following api paramaeter is copied @ref esp_gap_ble_api.c
        legacy_adv_params.sid = u8_handle;
        advert.setAdvertisingParams(u8_handle, &legacy_adv_params);
         advert.setInstanceAddress(u8_handle, addr_legacy);
        // this api uses pointer as it is
        memcpy(_adv_data[u8_handle], ibeacon_adv_payload, sizeof(ibeacon_adv_payload));
        advert.setAdvertisingData(u8_handle, sizeof(_adv_data[u8_handle]), &_adv_data[u8_handle][0]);
        advert.setDuration(u8_handle);
        addr_legacy[1]++;
    }
    advert.start(4, 0);
    //
    // setup scanner
    //
    pBLEScan = BLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
    pBLEScan->setActiveScan(false);
    pBLEScan->setInterval(1000);
    pBLEScan->setWindow(800);
    pBLEScan->start(0, nullptr, false);
}
```

### Debug Message

```plain
No Error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

biiinoida, 2022-03-05:
using an extended scan instead of a normal scan resolves the issue, for anybody else use the following code to scan while sending multiple advertisements
`
    pBLEScan->setExtendedScanCallback(new MyBLEExtAdvertisingCallbacks());
    pBLEScan->setExtScanParams();  // use with pre-defined/default values, overloaded function allows to pass parameters
    pBLEScan->startExtScan(90, 1); // scan duration in n * 10ms, period - repeat after n seconds (period >= duration)
`

closed 2022-03-05 by biiinoida

slobdell, 2023-11-26:
hey I'm having other errors getting this working around dependencies. This is a very useful snippet - wondering if you can share the full file you have?