espressif/arduino-esp32 #6372: ESP32 Crashes after every 3rd write on SD on HSPI

marcohald, 2022-03-03:
### Board

ESP32 DEVKITV1 ESP32-WROOM-32

### Device Description

ESP32 DEVKITV1 with SDCard to microsd Adapter attached, directly without breakboard
https://www.aliexpress.com/item/32864722159.html?spm=a2g0o.9042311.0.0.27424c4dmIbcfH
[image]

### Hardware Configuration

[image]

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Ubuntu 20.04.3 LTS

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Every 3rd write to SDcard fails.
I also tried using pin 12 as MISO with the same result, tried switching because of I read that https://www.esp32.com/viewtopic.php?t=19233#p71157

Maybe releate to https://github.com/espressif/arduino-esp32/issues/6189 or https://github.com/espressif/arduino-esp32/issues/6295

### Sketch

```cpp
#include <FS.h>
#include <SD.h>
#include <SPI.h>

char currentLogFileName[40];
File currentLogFile;

//HSPI
#define SD_SCK  14
#define SD_MISO  27
#define SD_MOSI  13
#define SD_CS  15

int cnt = 0;

void setup() {
    Serial.begin(115200);

  SPIClass spi = SPIClass(VSPI);
  spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

  if (!SD.begin(SD_CS,spi,10000000)) {   
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD.cardType();

    if (cardType == CARD_NONE) {
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if (cardType == CARD_MMC) {
        Serial.println("MMC");
    } else if (cardType == CARD_SD) {
        Serial.println("SDSC");
    } else if (cardType == CARD_SDHC) {
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    Serial.printf("SD Card Size: %lluMB\n", SD.cardSize() / (1024 * 1024));
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));

    // create a unique currentLogFile name
    int fn = 0;

    sprintf(currentLogFileName, "/candump-%d.csv", fn++);

  currentLogFile = SD.open(currentLogFileName, FILE_APPEND);

}

void loop() {
  
  //cnt++;
  //Serial.printf("Loop: %d\n", cnt);
  if(!currentLogFile){
    Serial.println("Failed to open file for appending");
    return;
  }
  if(currentLogFile.print("Time Stamp,ID,Extended,Dir,Bus,LEN,D1,D2,D3,D4,D5,D6,D7,D8")){
    Serial.println("Message appended");
  } else {
    Serial.println("Append failed");
  }
  sleep(1);
  //file.close();
}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
[��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
SD Card Type: SDHC
SD Card Size: 29818MB
Total space: 3689MB
Used space: 1MB
Message appended
Message appended
[  2551][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2551][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  2551][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[  3056][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  3056][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  3056][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
Message appended
[  6571][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  6571][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  6571][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[  7075][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  7075][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  7075][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
Message appended
[ 10590][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 10590][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 10590][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 11094][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 11094][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 11094][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
Message appended
[ 14609][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 14609][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 14609][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 15113][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 15113][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 15113][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
Message appended
[ 18628][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 18628][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 18628][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 19132][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 19132][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 19132][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
Message appended
[ 22647][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22647][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 22647][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 23151][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23151][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 23151][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
Message appended
[ 26666][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 26666][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 26666][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 27170][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 27170][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 27170][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
Append failed
Message appended
```

### Other Steps to Reproduce

also tried a Module like that https://eckstein-shop.de/TFMicroSDCardMemoryModulArduinoAtmegaKartenadapterCardreader
with vcc connected to 5v

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

marcohald, 2022-03-03:
I just added ``` Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));``` from https://github.com/espressif/arduino-esp32/issues/5701#issuecomment-934136387 to the sketch and now it crashes with another error.
Sketch
```
#include <FS.h>
#include <SD.h>
#include <SPI.h>

char currentLogFileName[40];
File currentLogFile;

//HSPI
#define SD_SCK  14
#define SD_MISO  27
#define SD_MOSI  13
#define SD_CS  15

int cnt = 0;

void setup() {
  Serial.begin(115200);

  SPIClass spi = SPIClass(VSPI);
  spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

  if (!SD.begin(SD_CS, spi, 10000000)) {
    Serial.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE) {
    Serial.println("No SD card attached");
    return;
  }

  Serial.print("SD Card Type: ");
  if (cardType == CARD_MMC) {
    Serial.println("MMC");
  } else if (cardType == CARD_SD) {
    Serial.println("SDSC");
  } else if (cardType == CARD_SDHC) {
    Serial.println("SDHC");
  } else {
    Serial.println("UNKNOWN");
  }

  Serial.printf("SD Card Size: %lluMB\n", SD.cardSize() / (1024 * 1024));
  Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
  Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));

  // create a unique currentLogFile name
  int fn = 0;

  sprintf(currentLogFileName, "/candump-%d.csv", fn++);

  currentLogFile = SD.open(currentLogFileName, FILE_APPEND);
  Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));

}

void loop() {

  //cnt++;
  //Serial.printf("Loop: %d\n", cnt);
  if (!currentLogFile) {
    Serial.println("Failed to open file for appending");
    return;
  }
  if (currentLogFile.print("Time Stamp,ID,Extended,Dir,Bus,LEN,D1,D2,D3,D4,D5,D6,D7,D8")) {
    Serial.println("Message appended");
[... 70 lines cut ...]
0x400d49bc: spiTransferByteNL at /root/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-spi.c line 1093
0x400d3e43: SPIClass::transfer(unsigned char) at /root/Arduino/hardware/espressif/esp32/libraries/SPI/src/SPI.cpp line 170
0x400d2a75: sdWait(unsigned char, int) at /root/Arduino/hardware/espressif/esp32/libraries/SD/src/sd_diskio.cpp line 100
0x400d2b19: sdSelectCard(unsigned char) at /root/Arduino/hardware/espressif/esp32/libraries/SD/src/sd_diskio.cpp line 124
0x400d2f25: sdTransaction(unsigned char, char, unsigned int, unsigned int*) at /root/Arduino/hardware/espressif/esp32/libraries/SD/src/sd_diskio.cpp line 245
0x400d2f52: ff_sd_status(unsigned char) at /root/Arduino/hardware/espressif/esp32/libraries/SD/src/sd_diskio.cpp line 619
0x400e08be: ff_disk_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/diskio/diskio.c line 66
0x400e125e: validate at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/src/ff.c line 3466
0x400e26c1: f_lseek at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/src/ff.c line 4267
0x400e42d7: vfs_fat_lseek at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/vfs/vfs_fat.c line 542
0x400ddc36: esp_vfs_lseek at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 444
0x40086952: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/stdio.c line 86
0x4008532e: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c line 224
0x400853b4: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c line 278
0x400857bb: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fvwrite.c line 189
0x400e79b5: _fwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fwrite.c line 137
0x400e7a15: fwrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fwrite.c line 177
0x400d1a6f: VFSFileImpl::write(unsigned char const*, unsigned int) at /root/Arduino/hardware/espressif/esp32/libraries/FS/src/vfs_api.cpp line 368
0x400d16a7: fs::File::write(unsigned char const*, unsigned int) at /root/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h line 1018
0x400d5c6b: Print::write(char const*) at /root/Arduino/hardware/espressif/esp32/cores/esp32/Print.h line 67
0x400d5d19: Print::print(char const*) at /root/Arduino/hardware/espressif/esp32/cores/esp32/Print.cpp line 89
0x400d13d6: loop() at /root/Arduino/canlogger-just-writes/canlogger-just-writes.ino line 74
0x400d62e1: loopTask(void*) at /root/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 50

```

VojtechBartoska, 2022-03-04:
Hi @marcohald, on what version of Arduino ESP32 core are you running? Do you have 2.0.2 version or you manually set up latest master? I'm asking because @P-R-O-C-H-Y already fixed the issue but it's only avaialble now in latest master you need to checkout manually.

marcohald, 2022-03-04:
Hi @VojtechBartoska  I'm using the latest Master cloned via 

git clone https://github.com/espressif/arduino-esp32.git ./

The Arduino IDE has the version 1.8.19

P-R-O-C-H-Y, 2022-03-04:
Thats really weird behaviour. Can you try that with version 2.0.0 of Arduino-esp32 please?

marcohald, 2022-03-04:
Hi @P-R-O-C-H-Y ,

I tried to switch to version 2.0.0
with the command git checkout tags/2.0.0

HEAD ist jetzt bei 44c11981 Added UM FeatherS2 Neo to boards.txt and added appropriate variants. (#5615)

run 
```
~/Arduino/hardware/espressif/esp32/tools# python get.py 
System: Linux, Bits: 64, Info: Linux-5.13.0-10027-tuxedo-x86_64-with-glibc2.29
Platform: x86_64-pc-linux-gnu
Downloading riscv32-esp-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz ...
Done
Extracting riscv32-esp-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz ...
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz ...
Done
Extracting xtensa-esp32-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz ...
Downloading xtensa-esp32s2-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz ...
Done
Extracting xtensa-esp32s2-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz ...
Tool esptool-3.1.0-linux.tar.gz already downloaded
Extracting esptool-3.1.0-linux.tar.gz ...
Tool x86_64-linux-gnu.mklittlefs-c41e51a.200706.tar.gz already downloaded
Extracting x86_64-linux-gnu.mklittlefs-c41e51a.200706.tar.gz ...
Tool mkspiffs-0.2.3-arduino-esp32-linux64.tar.gz already downloaded
Extracting mkspiffs-0.2.3-arduino-esp32-linux64.tar.gz ...
Renaming mkspiffs-0.2.3-arduino-esp32-linux64 to mkspiffs ...
Platform Tools Installed
```
But now I get this Errors
Sketch is the same as in https://github.com/espressif/arduino-esp32/issues/6372#issuecomment-1058451149
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
[��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
SD Card Type: SDHC
SD Card Size: 29818MB
Total space: 3689MB
Used space: 1MB
ESP-IDF version is: v4.4-dev-2313-gc69f0ec32
Message appended
Message appended
Message appended
Message appended
Message appended
Message appended
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f976d  PS      : 0x00060d30  A0      : 0x800d3f9a  A1      : 0x3ffb2570  
A2      : 0x0000ff00  A3      : 0x3ffc0784  A4      : 0x3ffc077c  A5      : 0xb33fffff  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x8008bed8  A9      : 0x3ffb2560  
A10     : 0x00000003  A11     : 0x00060d23  A12     : 0x00060d20  A13     : 0xb33fffff  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000ff00  LBEG    : 0x40086b55  LEND    : 0x40086b65  LCOUNT  : 0xfffffff1  

Backtrace:0x400f976a:0x3ffb25700x400d3f97:0x3ffb2590 0x400d34b9:0x3ffb25c0 0x400e19ed:0x3ffb25f0 0x400e3df3:0x3ffb2610 0x400e62da:0x3ffb2650 0x400de8c5:0x3ffb2680 0x40086871:0x3ffb26a0 0x40085222:0x3ffb26c0 0x400852ac:0x3ffb26e0 0x400856b7:0x3ffb2710 0x400e9d9d:0x3ffb2740 0x400e9dfd:0x3ffb2780 0x400d1c09:0x3ffb27a0 0x400f951d:0x3ffb27c0 0x400d5db7:0x3ffb27e0 0x400d16d3:0x3ffb2800 0x400d6538:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
[... 41 lines cut ...]
```
PC: 0x400f976d: spiGetClockDiv at /root/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-spi.c line 472
EXCVADDR: 0x0000ff00

Decoding stack results
0x400f976a: spiGetClockDiv at /root/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-spi.c line 468
0x400d3f97: SPIClass::beginTransaction(SPISettings) at /root/Arduino/hardware/espressif/esp32/libraries/SPI/src/SPI.cpp line 142
0x400d34b9: ff_sd_write(unsigned char, unsigned char const*, unsigned int, unsigned int) at /root/Arduino/hardware/espressif/esp32/libraries/SPI/src/SPI.h line 34
0x400e19ed: ff_disk_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/diskio/diskio.c line 74
0x400e3df3: f_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/src/ff.c line 3891
0x400e62da: vfs_fat_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/fatfs/vfs/vfs_fat.c line 377
0x400de8c5: esp_vfs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 443
0x40086871: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/stdio.c line 94
0x40085222: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c line 224
0x400852ac: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c line 278
0x400856b7: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fvwrite.c line 189
0x400e9d9d: _fwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fwrite.c line 137
0x400e9dfd: fwrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fwrite.c line 177
0x400d1c09: VFSFileImpl::write(unsigned char const*, unsigned int) at /root/Arduino/hardware/espressif/esp32/libraries/FS/src/vfs_api.cpp line 330
0x400f951d: fs::File::write(unsigned char const*, unsigned int) at /root/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h line 1018
0x400d5db7: Print::print(char const*) at /root/Arduino/hardware/espressif/esp32/cores/esp32/Print.h line 67
0x400d16d3: loop() at /root/Arduino/canlogger-just-writes/canlogger-just-writes.ino line 74
0x400d6538: loopTask(void*) at /root/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 46

```

closed 2022-03-04 by P-R-O-C-H-Y

marcohald, 2022-03-05:
worked fine thank you

TNeutron, 2022-07-04:
> Thank you @marcohald. I can confirm that your sketch have this problem, but it have a simple solution :)
> 
> Please place `SPIClass spi = SPIClass(VSPI);` before setup() function and the problem is gone.
> 
> So your sketch will be this:
> 
> ```
> #include <FS.h>
> #include <SD.h>
> #include <SPI.h>
> 
> char currentLogFileName[40];
> File currentLogFile;
> 
> //HSPI
> #define SD_SCK  14
> #define SD_MISO  27
> #define SD_MOSI  13
> #define SD_CS  15
> 
> int cnt = 0;
> 
> SPIClass spi = SPIClass(VSPI);
> 
> void setup() {
>   Serial.begin(115200);
> 
>   spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
> 
>   if (!SD.begin(SD_CS,spi,10000000)) {   
>         Serial.println("Card Mount Failed");
>         return;
>     }
>     uint8_t cardType = SD.cardType();
> 
>     if (cardType == CARD_NONE) {
>         Serial.println("No SD card attached");
>         return;
>     }
> 
>     Serial.print("SD Card Type: ");
>     if (cardType == CARD_MMC) {
>         Serial.println("MMC");
>     } else if (cardType == CARD_SD) {
>         Serial.println("SDSC");
>     } else if (cardType == CARD_SDHC) {
>         Serial.println("SDHC");
>     } else {
>         Serial.println("UNKNOWN");
>     }
> 
>     Serial.printf("SD Card Size: %lluMB\n", SD.cardSize() / (1024 * 1024));
>     Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
>     Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
> 
>     // create a unique currentLogFile name
>     int fn = 0;
> 
>     sprintf(currentLogFileName, "/candump-%d.csv", fn++);
> 
>     currentLogFile = SD.open(currentLogFileName, FILE_APPEND);
> }
> 
> void loop() {
>   
>   //cnt++;
>   //Serial.printf("Loop: %d\n", cnt);
>   if(!currentLogFile){
>     Serial.println("Failed to open file for appending");
>     return;
>   }
>   if(currentLogFile.print("Time Stamp,ID,Extended,Dir,Bus,LEN,D1,D2,D3,D4,D5,D6,D7,D8")){
>     Serial.println("Message appended");
>   } else {
>     Serial.println("Append failed");
>   }
>   sleep(1);
>   //file.close();
> }
> ```
> 
> For me this solved your problem, let me know if that worked for you. Tested on master branch.

I was facing this issue, I Completely missed this line it was causing all the issues. THANK YOU!

AllanOricil, 2024-03-11:
could someone help me here. I dont know why, but I can only read from the SD Card. When I try to append or write, I get this error

````bash
Waiting for time to be set...
Waiting for time to be set...
Waiting for time to be set...
Waiting for time to be set...
Mon, Mar 11 2024 19:18:33
SD Card Type: SDHC
SD Card Size: 121942MB
Total space: 121912MB
Used space: 3MB
fetching /keys.txt
decoding secrets
...

Message arrived [esp32-totp-write-new-secret] Hello, MQTT!
Appending to file: /keys.txt
[ 35220][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 35220][E][sd_diskio.cpp:624] ff_sd_status(): Check status failed
[ 35720][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 36220][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 36720][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 37220][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 37220][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/keys.txt) failed
````