espressif/arduino-esp32 #6373: MPU6050_raw.ino works in 1.6.0, 2.0.0 , but fail to initialize I2C interface with v2.0.1 , 2.0.2

rkuo2000, 2022-03-04:
### Board

NodeMCU-32S

### Device Description

ESP32 (NodeMCU-32S)

### Hardware Configuration

GPIO 18 & 19 I2C connected to MPU6050

### Version

v2.0.2

### IDE Name

ArduinoIDE

### Operating System

Ubuntu20.04 & Win11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

it hang at Initialing I2C Interace...
Both examples MPU6050_DMP.ino & MPU6050_raw.ino  failed.

### Sketch

```cpp
// initialize device
    Serial.println(F("Initializing I2C devices..."));
    mpu.initialize();
```

### Debug Message

```plain
No error message, just hang at Initializing I2C devices...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-04:
MPU6050 works in Project Tasmota. We use our own driver for the MPU6050.
So the library you use is not compatible.

rkuo2000, 2022-03-04:
I use [MPU6050 library](https://github.com/electroniccats/mpu6050) for DMP, which can provide stable pitch-roll-yaw,
and quaternions.

Jason2866, 2022-03-04:
I would open a issue there.

me-no-dev, 2022-03-04:
that library's code is all broken. Calling `beginTransaction`, followed by `requestFrom` and `endTransaction` is NOT a valid code! `beginTransaction` and `endTransaction` are to be used ONLY when writing. The fact that is works on other MCUs with more basic I2C peripherals and not RTOS environments, does not make the code valid.

rkuo2000, 2022-03-04:
BTW, my robot design use MPU6050 DMP for AHRS and ESP32 v2.02 for microROS, I really like to see them work together,
let me review MPU6050 DMP I2C coding, and see if I can fix it to make them compatible.

VojtechBartoska, 2022-04-11:
I think we are not able to help here, the problem is in mentioned Library.

closed 2022-04-11 by VojtechBartoska