espressif/arduino-esp32 #6374: [BUG] ESP32C3 BLE shuts CDC dowb (and kills USB until next power cycle)

0x0fe, 2022-03-04:
### Board

ESP32-C3 dev module

### Device Description

ESP32C3 Dev module

### Hardware Configuration

no

### Version

latest master

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

CDC

### Description

As soon as the BLE ini initialized, the CDC port is shut, it cannot be revived by cdc reset (aka opening and closing the serial monitor), and to revive USB the esp32c3 has to be power cycled and booted with GPIO9 to force bootlader then reprogrammed with aworking firmware. 
Le windows descriptor error, and CDC parameters errors after flashing a firmware with BLE enable elad me to think the CDC state machine and/or the whole USB stack stops to be served after the BLE is initialized, one way or another.

I tested on two board to make sure the problem was not hardware related, also i used a fresh download from master branch of few hours ago, so the CDC fix for wifi is implemented here for example. 

### Sketch

```cpp
#include "BLEDevice.h"

uint32_t chipId = 0;

void setup() {

  //BLEDevice::init("");// tried to init here, same issue
  
  Serial.begin(115200);
  while(!Serial)delay(10);
  Serial.println("\nSystem ready\n");
  for(int i=0;i<17;i=i+8){chipId|=((ESP.getEfuseMac()>>(40-i))&0xff)<<i;}
   
  BLEDevice::init("");
}

void loop() {
  
  Serial.printf("id:%d\n",chipId);
  delay(3000);
}
```

### Debug Message

```plain
when we try to openserial console after programming we got nothing, if we unplug and replu USB then try to open console we get "port parameters definition error" from the CDC, since it doesnt reply. We also get error message from windows most of time since the USB is no longer recognized.
```

### Other Steps to Reproduce

just flash the firmware as is, it shoudl reproduce the issue right away, after testing comment BLEDevice::init(""); power cycle with GPIO9 to GND and flash, it will then work normally and can be reflashed without unpluging USB.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-05:
known issue in ESP-IDF

0x0fe, 2022-03-08:
I noticed that on the old release 2.0 the BLE works along with the CDC.

me-no-dev, 2022-03-08:
this is because things in IDF changed that caused this situation. Hopefully will be "fixed" when 2.0.3 is ready

0x0fe, 2022-03-08:
N.B when i run on 2.0 the BLE works, the CDC works too (with unstable boot), however, everything freezes after few minutes and the watchdog does not reset, i still need to confirm wether this issue is actually related to the CDC or BLE.

0x0fe, 2022-03-16:
i ran on 2.0 for a while now, it is stable, wifi, BLE and CDC, ran for few days

VojtechBartoska, 2022-03-24:
it was already fixed in ESP-IDF, take a look on https://github.com/espressif/esp-idf/issues/8046. Still issue is supposed to be solved in Arduino ESP32 core 2.0.3 release, where ESP-IDF v4.4.1 will be used.

0x0fe, 2022-03-28:
Oh ok, thank you, i will take a look, is the change in the current mastr branch?

VojtechBartoska, 2022-04-06:
@0x0fe You can test it with development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1). Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

P-R-O-C-H-Y, 2022-04-14:
@0x0fe I have tested you sketch with version 2.0.3-RC1 and I have no problems :)

VojtechBartoska, 2022-04-20:
Closing as solved, @0x0fe if needed, you can reopen the issue. Thanks for your contribution!

closed 2022-04-20 by VojtechBartoska

NicoBocki, 2022-12-29:
Hello I have the same problem and im using 2.05 with an ESP32-S3 WROOM, pleas can somone help me.