espressif/arduino-esp32 #6375: ESP32-S2 Guru Meditation Error: Core 0 panic'ed

MattPChoy, 2022-03-05:
### Board

ESP32-S2-WR00M

### Device Description

https://www.dfrobot.com/product-2233.html

### Hardware Configuration

Digital pin 17 is connected to GND / +3.3v

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

ESP32 kernel panic occurs when moving wire in between pins

### Sketch

```cpp
#define INPUT_PIN 17

void IRAM_ATTR isr_falling() {
    Serial.println("0");
}

void IRAM_ATTR isr_rising() {
    Serial.println("1");
}

void setup() {
  Serial.begin(115200);
  pinMode(INPUT_PIN, INPUT);
  attachInterrupt(INPUT_PIN, isr_falling, FALLING);
  attachInterrupt(INPUT_PIN, isr_rising, RISING);
  // Mode: {LOW, CHANGE, RISING, FALLING}
}

void loop() {
    Serial.print("");
}
```

### Debug Message

```plain
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026305
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-05:
where is the backtrace from the error? and it should be decoded too.

felmue, 2022-03-07:
Hello @MattPChoy 

the issue most likely are the `Serial.println()` statements in the interrupt functions. AFAIK everything in interrupt functions needs to be in RAM which is not true for the built-in println(). Try setting a flag in the interrupt functions instead and then test the flag in the `loop()`.

Thanks
Felix

MattPChoy, 2022-03-07:
HI @me-no-dev, I believe I've found the error with the code earlier - I believe that the Serial.println implementation cannot be used with interrupts. For future reference (i.e. people struggling with this issue), this is what I've done.

Additionally, you'll notice I've removed the isr_falling() code as well as I think there's a (potentially hardware limitation) issue with attaching two interrupts to the same digital pin. There may be a way to connect both digital pins to the same physical input and add the rising-edge interrupt on one and falling-edge interrupt on the other, but I've ommitted the code for that as it isn't required for my use case.

```
#define INPUT_PIN 17
volatile bool detect_rising = false;

void IRAM_ATTR isr_rising() {
    detect_rising = true;
}

void setup() {
  Serial.begin(115200);
  pinMode(INPUT_PIN, INPUT);
  attachInterrupt(INPUT_PIN, isr_rising, RISING);
}

void loop() {
    if (detect_rising) {
        System.println("1");
        detect_rising=false;
    }
}
```

MattPChoy, 2022-03-07:
> the issue most likely are the Serial.println() statements in the interrupt functions
Yep, funny you mention this. I was just about to post the above comment :)

closed 2022-03-07 by MattPChoy