espressif/arduino-esp32 #6376: ESP32-C3 I2C does not work except GPIO 8 and 9

h1romas4, 2022-03-05:
### Board

M5Stamp C3

### Device Description

M5Stamp C3

### Hardware Configuration

GPIO 0 and 1 are connected to I2C sensor.

### Version

latest master

### IDE Name

VS Code C++ extention

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

ESP32-C3 I2C does not work except GPIO 8 and 9.

I have confirmed that the this issue occurs with the following versions.

arduino-esp32 master '4da1051'
eps-idf `v4.4`

### Sketch

```cpp
#include "Arduino.h"
#include "esp_log.h"

static const char *TAG = "test_i2c_gpio.cpp";

QMP6988 qmp6988;

#define I2C_SDA 0
#define I2C_SCL 1

void init_i2c_gpio(void)
{
    Wire.begin(I2C_SDA, I2C_SCL);
    if(!qmp6988.init()) {
        ESP_LOGE(TAG, "Could not find a valid qmp6988 sensor");
    }
}
```

### Debug Message

```plain
E test_i2c_gpio1819.cpp Could not find a valid qmp6988 sensor
```

### Other Steps to Reproduce

It works with the combination of GPIO 8 and 9.

Calling the esp-idf `i2c_set_pin()` initialization as follows worked for the GPIO 0 and 1 combination.

```c
#include "Arduino.h"
#include "driver/i2c.h"
#include "esp_log.h"

static const char *TAG = "test_i2c_gpio.cpp";

QMP6988 qmp6988;

#define I2C_SDA 0
#define I2C_SCL 1

void init_i2c_gpio(void)
{
    // initialize by esp-idf
    i2c_set_pin(0, I2C_SDA, I2C_SCL, true, true, I2C_MODE_MASTER);

    Wire.begin(I2C_SDA, I2C_SCL);
    if(!qmp6988.init()) {
        ESP_LOGE(TAG, "Could not find a valid qmp6988 sensor");
    }
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

HYXianghua, 2022-03-05:
Is it possible that your SDA and SCL pins are reversed?  Or there is a short circuit in the pin

me-no-dev, 2022-03-05:
Try this:
```cpp
Wire.setPins(I2C_SDA, I2C_SCL);
Wire.begin();
```

h1romas4, 2022-03-05:
Thank you for your response.

I tried modifying it to `Wire.setPins(I2C_SDA, I2C_SCL)` and `Wire.begin();` but got the following error.

```
[   287][E][Wire.cpp:126] setPins(): bus already initialized. change pins only when not.
E (331) test_i2c_gpio.cpp: Could not find a valid qmp6988 sensor
```

Modified source code:

```c
#include "Arduino.h"
#include "driver/i2c.h"
#include "esp_log.h"
#include "hal/gpio_hal.h"
#include "UNIT_ENV.h"

#include "test_i2c_gpio.h"

static const char *TAG = "test_i2c_gpio.cpp";

SHT3X sht30;
QMP6988 qmp6988;

#define I2C_SDA 1
#define I2C_SCL 0

void init_i2c_gpio(void)
{
    // i2c_set_pin(0, I2C_SDA, I2C_SCL, true, true, I2C_MODE_MASTER);

    Wire.setPins(I2C_SDA, I2C_SCL);
    // Wire.begin(I2C_SDA, I2C_SCL);
    Wire.begin();
    if(!qmp6988.init()) {
        ESP_LOGE(TAG, "Could not find a valid qmp6988 sensor");
    }
}
```

me-no-dev, 2022-03-05:
That message would suggest that something already started Wire before you called this function (or you called it more then once)

h1romas4, 2022-03-06:
Thank you for youe advice.

As you pointed out, when I set the program to the minimum configuration, both `Wire.setPins(I2C_SDA, I2C_SCL);` ` `Wire.begin(I2C_SDA, I2C_SCL);` now work!

The process I deleted did not use Wire, but SPI/SD/SPIFFS. I still don't know how these are related.

The issue seems to be the same as the question in the following link.

https://community.m5stack.com/topic/3715/m5stamp-c3-i2c-interface

I pushed all the source code to GitHub for this ticket.

https://github.com/h1romas4/m5stamp-c3dev/tree/i2c-on-gpio1819

I would like to close this ticket as there seems to be another cause.

Thank you very much.

closed 2022-03-06 by h1romas4

udayalawa, 2023-05-05:
Any updates on this?

orkungedik, 2023-09-03:
I am running on running on ESP-C3-13 kit and used SDA: GPIO4 and SCL: GPIO5. Only works with this configuration by my side.

brightproject, 2025-12-12:
I confirm that pins 8 and 9 don't work for the `ESP32-C3 Super Mini`, and the I2C bus is unavailable.
I was able to connect to this board 

[image]

using these pins:
```c++
// These pins work
// #define SDA_I2C (GPIO_NUM_10)
// #define SCL_I2C (GPIO_NUM_20)
// These pins also work
#define SDA_I2C (GPIO_NUM_6)
#define SCL_I2C (GPIO_NUM_7)
``` 

The pinout is as follows:

[image]