espressif/arduino-esp32 #6377: PlatformIO package if outdated.

nicosandller, 2022-03-05:
### Board

ESP32 WROOM 32

### Device Description

Its a development board with the ESP32 WROOM 32 shield in it. Pretty standard.

### Hardware Configuration

Using default I2C interface to talk to a BME 280 bosch sensor.

### Version

v1.0.6

### IDE Name

PlatformIO on VS code

### Operating System

Mac OS 12.2.1

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

The PlatformIO framework repository: 

https://registry.platformio.org/tools/platformio/framework-arduinoespressif32/versions

Is out of date. The latest version runs an old (1.0.6) version of this repository. And this is creating problems. The microcontroller crashes with bluetooth communication.

This issue is discussed here: https://github.com/HomeSpan/HomeSpan/issues/241

Can you update it on PlatformIO?

### Sketch

```cpp
Can't share the details of the sketch. But I'm using this Arduino package: 

https://github.com/HomeSpan/HomeSpan
```

### Debug Message

```plain
- Can't read the debug message. Will try to save the Serial output to share it here later.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-05:
It's up to PlatformIO to update the package on their end. You can find workarounds in our documentation and searching the issues here.

closed 2022-03-05 by me-no-dev

nicosandller, 2022-03-05:
@me-no-dev  can you point me to some links? I've been trying to find a workaround but haven't been able so far.

me-no-dev, 2022-03-07:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-platformio