espressif/arduino-esp32 #6379: ESP32C3 stopped connecting to home wifi after commit 5940d89

Sys64736, 2022-03-06:
### Board

esp32-c3-devkitc-02

### Device Description

Only the dev board

### Hardware Configuration

Nothing connect to GPIO's

### Version

v2.0.2

### IDE Name

Platformio

### Operating System

Win10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Was connecting to home router (Tplink AC1750) in version 2.0.0,  2.0.1 but stopped in v2.02

Last commit that works for me is https://github.com/espressif/arduino-esp32/commit/bb096153914ddadefc4e1a8b72ef61d0bef41921

Commit that stops my C3's, I have a couple, from connecting is:
https://github.com/espressif/arduino-esp32/commit/5940d89e671fa5ee5b065be46f5d959515feca6f

They do connect to cell phone hotspot (which I believe only uses WPA2_PSK)

but router has multiple authorization modes available and they usually connect with AUTH: PSK but after update to commit 5940d8 they will not connect.

Further info:

STA Connected: SSID: Router1, BSSID: d8:0d:17:48:77:34, Channel: 5, Auth: PSK <--- C3's will connect in v2.0.2 (bb09615)
STA Disconnected: SSID: Router1, BSSID: 00:00:00:00:00:00, Reason: 201 (NO_AP_FOUND) <--- C3's won't connect in v2.0.2 (5940d89)
STA Connected: SSID: HotSpot1, BSSID: b6:58:fa:2b:33:b3, Channel: 11, Auth: WPA2_PSK <--- C3's will connect in v2.0.2 (5940d89)

### Sketch

```cpp
const char* ssid     = "ssidname";        
const char* password = "password";    

void setup() {

  Serial.begin(115200);      
  delay(100);
  WiFi.begin(ssid, password); 
  delay(50);            
  Serial.print("Connecting..."); 
  while (WiFi.status() != WL_CONNECTED) { Serial.print("."); delay(500);}    
  Serial.println('\n');
  Serial.println("Connection established!");  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());     

}
```

### Debug Message

```plain
Unfortunately I didn't copy the commit 5940d89's output except
STA Disconnected: SSID: Router1, BSSID: 00:00:00:00:00:00, Reason: 201 (NO_AP_FOUND)
but the rest is the usual when it can't find an AP then disconnects and keeps repeating.

Up to and including commit bb09615 works as it should for me:
STA Connected: SSID: Router1, BSSID: d8:0d:17:48:77:34, Channel: 5, Auth: PSK 

Is there something that's preventing it from falling back to PSK authentication?
```

### Other Steps to Reproduce

No other steps, except a router with multiple authentication types?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Sys64736, 2022-03-06:
More info:

I've narrowed it down to the router having TKIP checked as an available algorithm.   Btw, router has DD-WRT installed.

If I uncheck TKIP but leave all others checked,  ESP32 and ESP32C3 connect with latest build.

These settings in the router cause the ESP's (32) and (C3) to not find the AP following https://github.com/espressif/arduino-esp32/commit/5940d89e671fa5ee5b065be46f5d959515feca6f

**Authentications Checked**
WPA Personal
WPA2 Personal
WPA2 Personal with SHA256

**WPA Algorithms**
CCMP-128 (AES)
TKIP

**Using latest release.**
Keeping TKIP checked:
NO_AP_FOUND

**If I remove TKIP, connections are made:**
[  2366][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: Router1, BSSID: ac:84:c6:0c:47:93, Channel: 1, Auth: WPA_WPA2_PSK

**If I remove TKIP and WPA Personal:**
[  2805][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: Router1, BSSID: ac:84:c6:0c:47:93, Channel: 1, Auth: WPA2_PSK

I believe this isssue, https://github.com/espressif/arduino-esp32/issues/6348, may be related?

Shouldn't it connect anyway to the strongest available mode, even if TKIP is checked?

Sys64736, 2022-03-06:
WiFiSTA.cpp

Changing the threshhold mode in:
101     		wifi_config->sta.threshold.authmode = **WIFI_AUTH_WPA_PSK**;  // Changed from **WIFI_AUTH_WPA2_PSK**

Fixed my connection issues.  But I noticed it's connecting as AUTH: PSK , as it was in previous releases.   Shouldn't it be connecting as WPA2_PSK because it's available?   My phone connects to this router on 2.4ghz as WPA/WPA2-Personal so I think the router software is working right.
I realize dropping the threshhold value down by 1 might take an extra few seconds on a scan but I'm sure there's other routers out there with this config.  Dropping to WIFI_AUTH_WEP might be an option also.... just in case?

Sys64736, 2022-03-06:
Doing a scan, encryption type is coming up zero
 Router1 (-52) Encryption Type: 0  (WIFI_AUTH_OPEN)

That's with these on:
WPA2 Personal
WPA2 Personal with SHA256
and
CCMP-128 (AES)

If I turn off **WPA2 Personal** and only leave **WPA2 Personal with SHA256** checked
 Router1 (-52) Encryption Type: 3 (WIFI_AUTH_WPA2_PSK)

That's with router firmware DD-WRT v3.0 (04/06/2020)

All other clients, laptop's, phones, etc see the band as encrypted and connect as WPA2-Personal

Jason2866, 2022-03-07:
You should (for security reasons) always disable TKIP. It is cracked.

Sys64736, 2022-03-08:
Putting the stock firmware on the C7, the scan results are correct with every security & encryption change.  It doesn't give the options like dd-wrt or possibly openwrt as it's mostly drop down or one or the other so it's hard to select certain conditions.  

Options are  [Auto]  [WPA]  [WPA2]   for dropdown WPA/WPA2 Personal

The only thing is the WiFiSTA.cpp change in threshold, if I keep it at WIFI_AUTH_WPA2_PSK it will not connect to the stock tp-link with WPA selected as the security mode. 

 Result is:

[ 33578][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 35636][V][WiFiGeneric.cpp:289] _arduino_event_cb(): STA Disconnected: SSID: 1TP-Link_4794, BSSID: 00:00:00:00:00:00, Reason: 201
[ 35637][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 35646][W][WiFiGeneric.cpp:852] _eventCallback(): Reason: 201 - NO_AP_FOUND

But running a scan shows this:
1TP-Link_4794 (-39)*Encryption Type: 2 WIFI_AUTH_WPA_PSK

Which may be confusing to some who attempt a WiFi manager type of configuration only to reboot and never connect.

WiFiSTA.cpp - changing the default to
101 wifi_config->sta.threshold.authmode = WIFI_AUTH_WPA_PSK; 

Result
[  3022][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: 1TP-Link_4794, BSSID: ac:84:c6:0c:47:94, Channel: 7, Auth: PSK
[  3024][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
.[  3768][V][WiFiGeneric.cpp:294] _arduino_event_cb(): STA Got New IP:192.168.0.156
[  3769][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  3773][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.0.156, MASK: 255.255.255.0, GW: 192.168.0.1

SuGlider, 2022-04-01:
@Sys64736 - Thanks for the detailed report and investigation.

I have tested WiFi connection with different Encryption modes using the ESP32-C3-DevkitM-1 v1.0 board and Arduino Core v2.0.3-RC1. Unfortunately, I don't have any C3-devkitc-02 to test it.

It seems that some C3 issues have been solved with latest IDF version used in the new Arduino Core RC version we released.
Could you please test it again using this new Arduino Core with your board and let me know.

Thanks!

SuGlider, 2022-04-05:
Thanks @Sys64736 for the detailed information!

I'll check it and propose a PR or something else to fix it.

Sys64736, 2022-04-05:
Thanks!  Btw, this happens with ESP32's also along with C3's and I would imagine any other model.

VojtechBartoska, 2022-05-04:
Hello @Sys64736, are you please able to retest this with 2.0.3 stable? Thanks for your help.

VojtechBartoska, 2022-05-04:
seems similar to #6430 which was solved on IDF side.

Sys64736, 2022-05-08:
Thanks.  Tested and working great.   Fxed with https://github.com/espressif/arduino-esp32/pull/6651

closed 2022-05-08 by Sys64736