espressif/arduino-esp32 #6381: Specified folder/zip file does not contain a valid library

ramankanythia, 2022-03-06:
### Board

UNO

### Device Description

Arduino: 1.8.19 (Windows 7), Board: "Arduino Uno"

ESP32_Bluetooth_Serial:1:10: fatal error: BluetoothSerial.h: No such file or directory

 #include "BluetoothSerial.h"

          ^~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

BluetoothSerial.h: No such file or directory

Specified folder/zip file does not contain a valid library

Specified folder/zip file does not contain a valid library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Hardware Configuration

Arduino: 1.8.19 (Windows 7), Board: "Arduino Uno"

ESP32_Bluetooth_Serial:1:10: fatal error: BluetoothSerial.h: No such file or directory

 #include "BluetoothSerial.h"

          ^~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

BluetoothSerial.h: No such file or directory

Specified folder/zip file does not contain a valid library

Specified folder/zip file does not contain a valid library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Version

v1.0.6

### IDE Name

ardiuno IDE

### Operating System

Windows 7

### Flash frequency

8

### PSRAM enabled

no

### Upload speed

9600

### Description

Arduino: 1.8.19 (Windows 7), Board: "Arduino Uno"

ESP32_Bluetooth_Serial:1:10: fatal error: BluetoothSerial.h: No such file or directory

 #include "BluetoothSerial.h"

          ^~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

BluetoothSerial.h: No such file or directory

Specified folder/zip file does not contain a valid library

Specified folder/zip file does not contain a valid library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Sketch

```cpp
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

### Debug Message

```plain
Arduino: 1.8.19 (Windows 7), Board: "Arduino Uno"

ESP32_Bluetooth_Serial:1:10: fatal error: BluetoothSerial.h: No such file or directory

 #include "BluetoothSerial.h"

          ^~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

BluetoothSerial.h: No such file or directory

Specified folder/zip file does not contain a valid library

Specified folder/zip file does not contain a valid library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-06:
Version 1.0.6 is afaik not supported anymore. Have you tried v2.0.2 ?

VojtechBartoska, 2022-03-29:
Hello I'm closing this issue due to no response. If needed you can reopen it.