espressif/arduino-esp32 #6382: Error trying to compile example code CameraWebServer for ESP32 Cam using Arduino IDE

Allasso, 2022-03-06:
### Board

ESP32 Cam

### Device Description

ESP32 Cam

### Hardware Configuration

FT232RL USB to TTL Serial Converter

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

MacOS 10.13.6

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Computer recognizes hardware.
GPIO0 pin held low via jumper to ground.
Unable to compile due to error.  Expect compile to be successful.

### Sketch

```cpp
Examples > ESP32 > Camera > CameraWebServer
```

### Debug Message

```plain
Arduino: 1.8.19 (Mac OS X), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

[75236] Error loading Python lib '/var/folders/xg/k651jnd11fsc5hljfrqcpfrc0000gn/T/_MEITQF9fF/libpython3.8.dylib': dlopen: dlopen(/var/folders/xg/k651jnd11fsc5hljfrqcpfrc0000gn/T/_MEITQF9fF/libpython3.8.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/xg/k651jnd11fsc5hljfrqcpfrc0000gn/T/_MEITQF9fF/libintl.8.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/xg/k651jnd11fsc5hljfrqcpfrc0000gn/T/_MEITQF9fF/libintl.8.dylib
Multiple libraries were found for "WiFi.h"
 Used: /Users/kevin/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board AI Thinker ESP32-CAM.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-03-06:
Likely a dupe of https://github.com/espressif/arduino-esp32/issues/6269

VojtechBartoska, 2022-03-07:
@lbernstone Thanks for pointing this out. Closing as duplicate. Please @Allasso, follow mentioned issue as there is more contributors and some solutions. Thanks!

closed 2022-03-07 by VojtechBartoska

Allasso, 2022-03-07:
Unfortunately the solutions mentioned are quite outside my competence range :-)