espressif/arduino-esp32 #6384: ESP32S2 Dev Module detected ad Deneyap Mini

colde, 2022-03-07:
### Board

ESP32S2 Dev Module

### Device Description

Some hardware presents itself as an ESP32S2 Dev Module at times when they are being set into flashing mode and uses the built-in USB driver, i've seen this with Wemos S2 Pico, and i believe i've also encountered it with Unexpected Maker's FeatherS2.

### Hardware Configuration

Not relevant

### Version

latest master

### IDE Name

Arduino IDE 2.0.0-rc3

### Operating System

MacOS

### Flash frequency

Not relevant

### PSRAM enabled

yes

### Upload speed

Not relevant

### Description

When ever hardware that presents itself as ESP32S2 Dev Module, Arduino will detect it as a Deneyap Mini, as Deneyap Mini re-uses the same PID and VID.

I propose that the Deneyap Mini PID and VID get's less priority, so that board are primarily detected as ESP32S2 dev modules by default instead.

### Sketch

```cpp
Not relevant
```

### Debug Message

```plain
Not relevant
```

### Other Steps to Reproduce

Attach a ESP32S2 Dev Module to the Arduino 2.0.0-rc3 IDE and it will detect it as a Deneyap Mini

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

colde, 2022-03-08:
I don't think there is a 2.0.2 version of the Arduino IDE.
For the additional boards manager, I'm using the URL https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json which does seem to point to 2.0.2.
The problem to me looks like the deneyap mini is squatting on the official USB pid.

me-no-dev, 2022-03-08:
the very top version of the link you posted does include 2.0.2 :)

many boards will end up with the same pid/vid BTW... we will see how we will deal with this in near future.

VojtechBartoska, 2022-04-11:
Hello, sorry for late reply. Are you able to retest this with Arduino core ESP32 version 2.0.3-rc1? Thanks!

VojtechBartoska, 2022-05-04:
I'm closing the issue as expired due to no answer.

If needed, please reopen it.

Thanks for understanding.

OmpaOmpa, 2023-03-09:
This is still true for version 2.0.7
Seems like all unbranded ESP32S2 are wrongly identified as Deneyap Mini who uses the default ESP32S2 pid. I to suggest prioritizing the standard Dev Module available world wide above a product available only in turkey.

Serendo, 2023-07-10:
I find that the problem is with this api that arduino-cli uses, it gives the wrong result. 
```
curl -s https://builder.arduino.cc/v3/boards/byVidPid/0x303A/0x0002 
```
```
{"architecture":"esp32","fqbn":"esp32:esp32:deneyapmini","href":"/v3/boards/esp32:esp32:deneyapmini","id":"deneyapmini","menus":[{"id":"UploadSpeed","name":"Upload Speed","variants":[{"id":"921600","name":"921600"},{"id":"115200","name":"115200"},{"id":"230400","name":"230400"},{"id":"460800","name":"460800"}]},{"id":"CDCOnBoot","name":"USB CDC On Boot","variants":[{"id":"default","name":"Enabled"},{"id":"cdc","name":"Disabled"}]},{"id":"MSCOnBoot","name":"USB Firmware MSC On Boot","variants":[{"id":"default","name":"Disabled"},{"id":"msc","name":"Enabled"}]},{"id":"DFUOnBoot","name":"USB DFU On Boot","variants":[{"id":"default","name":"Disabled"},{"id":"dfu","name":"Enabled"}]},{"id":"UploadMode","name":"Upload Mode","variants":[{"id":"default","name":"Internal USB"},{"id":"cdc","name":"UART0"}]},{"id":"CPUFreq","name":"CPU Frequency","variants":[{"id":"240","name":"240MHz (WiFi)"},{"id":"160","name":"160MHz (WiFi)"},{"id":"80","name":"80MHz (WiFi)"},{"id":"40","name":"40MHz"},{"id":"20","name":"20MHz"},{"id":"10","name":"10MHz"}]},{"id":"FlashFreq","name":"Flash Frequency","variants":[{"id":"80","name":"80MHz"},{"id":"40","name":"40MHz"}]},{"id":"FlashMode","name":"Flash Mode","variants":[{"id":"qio","name":"QIO"},{"id":"dio","name":"DIO"},{"id":"qout","name":"QOUT"},{"id":"dout","name":"DOUT"}]},{"id":"FlashSize","name":"Flash Size","variants":[{"id":"4M","name":"4MB (32Mb)"},{"id":"8M","name":"8MB (64Mb)"},{"id":"2M","name":"2MB (16Mb)"},{"id":"16M","name":"16MB (128Mb)"}]},{"id":"PartitionScheme","name":"Partition Scheme","variants":[{"id":"default","name":"Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)"},{"id":"defaultffat","name":"Default 4MB with ffat (1.2MB APP/1.5MB FATFS)"},{"id":"default_8MB","name":"8M with spiffs (3MB APP/1.5MB SPIFFS)"},{"id":"minimal","name":"Minimal (1.3MB APP/700KB SPIFFS)"},{"id":"no_ota","name":"No OTA (2MB APP/2MB SPIFFS)"},{"id":"noota_3g","name":"No OTA (1MB APP/3MB SPIFFS)"},{"id":"noota_ffat","name":"No OTA (2MB APP/2MB FATFS)"},{"id":"noota_3gffat","name":"No OTA (1MB APP/3MB FATFS)"},{"id":"huge_app","name":"Huge APP (3MB No OTA/1MB SPIFFS)"},{"id":"min_spiffs","name":"Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)"},{"id":"fatflash","name":"16M Flash (2MB APP/12.5MB FATFS)"},{"id":"app3M_fat9M_16MB","name":"16M Flash (3MB APP/9.9MB FATFS)"}]},{"id":"DebugLevel","name":"Core Debug Level","variants":[{"id":"none","name":"None"},{"id":"error","name":"Error"},{"id":"warn","name":"Warn"},{"id":"info","name":"Info"},{"id":"debug","name":"Debug"},{"id":"verbose","name":"Verbose"}]},{"id":"PSRAM","name":"PSRAM","variants":[{"id":"disabled","name":"Disabled"},{"id":"enabled","name":"Enabled"}]},{"id":"EraseFlash","name":"Erase All Flash Before Sketch Upload","variants":[{"id":"none","name":"Disabled"},{"id":"all","name":"Enabled"}]}],"name":"Deneyap Mini","package":"esp32","plan":"create-free"}
```

Lehna291, 2023-08-18:
Hi there,

I´m encountering the same issue: 

--> Dev Board is detected as Deneyap Mini.
I´m using the following hardware: Waveshare ESP32-S2 Dev. Board with the Arduino IDE 2.1.1 and the ESP32 Core 2.0.11. and as board in the board manager i choose ESP32S2 Dev Module.

A workaround could be, that i exchange the pins_arduino.h file in the variants-deneyap folder, so i can use it as a "normal dev board". 
But maybe every time there is an update on the ESP Core the files in the variant folder will be overwritten.

IanEatsBabies, 2023-12-08:
2.2.1, still the same problem. If the dev's aren't interested in fixing this is there some way we can delete the Deneyap boards ourselves?

VojtechBartoska, 2023-12-08:
Hi @IanEatsBabies, I would not call it as we are not interested in solving. Feel free to propose the fix, this is an open public project. Actually this is a bigger problem that some boards match PID/VID, there is no not straight-forward solution how to deal with this.

Anyway, I am reopening the issue.

Any ideas @P-R-O-C-H-Y and @me-no-dev?

IanEatsBabies, 2023-12-08:
> Hi @IanEatsBabies, I would not call it as we are not interested in solving. Feel free to propose the fix, this is an open public project. Actually this is a bigger problem that some boards match [...]

Sorry man, came across as such. I'll give it a go tonight at work when I've got some time.

me-no-dev, 2023-12-11:
@IanEatsBabies this is both unslovable and partially solvable by the IDE folks.

- All ESP32-S2 devices have the same VID/PID in download mode (and by default while booted). Some boards use custom VID/PID and that helps, but causes port to be switched when in Download Mode.
- All integrated CDC/JTAG boards (S3, C3, C6, H2) also use the same VID/PID for that interface. It is hardcoded into the chip and can not be changed
- The IDE could somewhat help by not picking random board that matches VID/PID every time and show you our main dev module instead (first in the list)

IanEatsBabies, 2023-12-12:
I've been looking into fingerprinting until I came upon a far more powerfull solution: Why not build in a simple filter people can use?

OmpaOmpa, 2023-12-12:
_"The IDE could somewhat help by not picking random board that matches VID/PID every time and show you our main dev module instead (first in the list)"_
This realy should be the default, everything else is illogical. 

_"All integrated CDC/JTAG boards (S3, C3, C6, H2) also use the same VID/PID for that interface. It is hardcoded into the chip and can not be changed"_
This should make it very easy to do.

per1234, 2024-05-02:
> Feel free to propose the fix

The fix is to remove all the non-unique VID/PID associations from `boards.txt`. Associations should only ever be made for VID/PID pairs that are unique to a single specific board model.

The incorrect port identifications caused by the inappropriate associations make things confusing for Arduino IDE users, and make it impossible to use any of these boards in the "new" Arduino Cloud editor:

https://forum.arduino.cc/t/web-editor-misidentifies-connected-board/1193052/9

It is currently possible to workaround that incompatibility by using the "old" Cloud Editor, but Arduino is about to remove access to the "old" editor.

per1234, 2024-06-28:
This was resolved by https://github.com/espressif/arduino-esp32/pull/9717.

The innovative solution that was implemented will cause the port to be identified as "ESP32 Family Device" instead of as a random board.

Something to note is that this solution will result in the dummy "ESP32 Family Device" board to be selected when the user selects the board from the board selector menu in the Arduino IDE 2.x toolbar (which automatically sets the board when the port has a positive identification). This causes compilation to fail:

```text
FQBN: esp32:esp32:esp32_family
Using board 'esp32_family' from platform in folder: C:\Users\per\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2
Using core 'arduino' from platform in folder: C:\Users\per\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2

cmd /c if exist "C:\\Users\\per\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024527-21456-2j5kew.lyhnh\\sketch_jun27a\\partitions.csv" COPY /y "C:\\Users\\per\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024527-21456-2j5kew.lyhnh\\sketch_jun27a\\partitions.csv" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino\\sketches\\14EF117D0F1995B5877EA5151E37BDCB\\partitions.csv"
cmd /c if not exist "C:\\Users\\per\\AppData\\Local\\Temp\\arduino\\sketches\\14EF117D0F1995B5877EA5151E37BDCB\\partitions.csv" if exist "\\partitions.csv" COPY "\\partitions.csv" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino\\sketches\\14EF117D0F1995B5877EA5151E37BDCB\\partitions.csv"
cmd /c if not exist "C:\\Users\\per\\AppData\\Local\\Temp\\arduino\\sketches\\14EF117D0F1995B5877EA5151E37BDCB\\partitions.csv" COPY "C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.2\\tools\\partitions\\{build.partitions}.csv" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino\\sketches\\14EF117D0F1995B5877EA5151E37BDCB\\partitions.csv"
The system cannot find the file specified.
exit status 1

Compilation error: exit status 1
```

So the user must make sure to select the appropriate board from the **Tools > Board** menu.

However, the new system has been in production use for a few weeks now and I don't find any reports of users of this happening so it seems that the benefits of the port having a helpful label will outweigh the potential harm of this causing misconfiguration of the IDE.

me-no-dev, 2024-06-28:
@per1234 I'm glad that you find this helpful, as it was the only "solution" I could come up with.

closed 2025-01-16 by Parsaabasi