espressif/arduino-esp32 #6385: UART Callback on TX FIFO empty.

gonzabrusco, 2022-03-07:
### Related area

UART

### Hardware specification

ESP32

### Is your feature request related to a problem?

Nowadays, most modbus libraries based on Arduino, when writing to UART have to wait for flush() to be able to toggle the 485 transceiver pin. That means waiting a lot of time that could be used for something else. 

I don't know if it's technically possible, but it would be very useful to have a callback on TX complete (TX FIFO is empty).

Is this possible?

Thanks
Gonzalo

### Describe the solution you'd like

A callback on TX complete

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-03-25:
@gonzabrusco 

IDF UART driver controls all the Interrupt events from the ESP32 for UART.

There is an interrupt available, called UART_TX_DONE_INT, but **IDF UART driver doesn't use it to send any event to the user application**.

### Sorry, but we won't ESP32 be able to fulfill your request.

You can try to do something around IDF call [```esp_err_t uart_wait_tx_done(uart_port_t, uart_num, TickType_t ticks_to_wait)```](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv417uart_wait_tx_done11uart_port_t10TickType_t), like a task that keeps checking it or something similar.

closed 2022-03-25 by SuGlider

gonzabrusco, 2022-03-25:
Thanks @SuGlider for taking your time checking this.