espressif/arduino-esp32 #6386: Error compiling for board DOIT ESP32 DEVKIT V1

CrazyLionQC, 2022-03-07:
### Board

Esp32 Dev kit V1

### Device Description

Common Esp32 Dev kit V1

### Hardware Configuration

Noting is attach and no code is being used.

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

80HZ

### PSRAM enabled

no

### Upload speed

115200

### Description

Got a new laptop at the office, wich is a powerfull gaming laptop. Installed Arduino IDE and the ESP32 add-on boards. But I always have this "Error compiling for board DOIT ESP32 DEVKIT V1." error ( full error below ).

I've been looking for a fix for the last 4 hours now and tried everything I've found including:
-deleting Cache (appData/arduino15)
-Reinstalling Arduino IDE several time
-Copy past the cache from my old laptop, where everything work just fine.

I also have a the same error if no board is plugged at all.... like it's not even trying, but it work on my old laptop....

Is there any windows configuration, firewall configuration, authorizations to do ?

### Sketch

```cpp
Empty sketch...
```

### Debug Message

```plain
Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\R&D\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\R&D\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\R&D\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\R&D\AppData\Local\Temp\arduino_build_711433 -warnings=none -build-cache C:\Users\R&D\AppData\Local\Temp\arduino_cache_633889 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose C:\Users\R&D\AppData\Local\Temp\untitled1973103754.tmp\sketch_mar07a\sketch_mar07a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\R&D\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\R&D\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\R&D\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\R&D\AppData\Local\Temp\arduino_build_711433 -warnings=none -build-cache C:\Users\R&D\AppData\Local\Temp\arduino_cache_633889 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose C:\Users\R&D\AppData\Local\Temp\untitled1973103754.tmp\sketch_mar07a\sketch_mar07a.ino

Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Using core 'esp32' from platform in folder: C:\Users\R&D\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

cmd /c if exist "C:\\Users\\R&D\\AppData\\Local\\Temp\\untitled1973103754.tmp\\sketch_mar07a\\partitions.csv" copy /y "C:\\Users\\R&D\\AppData\\Local\\Temp\\untitled1973103754.tmp\\sketch_mar07a\\partitions.csv" "C:\\Users\\R&D\\AppData\\Local\\Temp\\arduino_build_711433\\partitions.csv"

& was unexpected at this time.

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.
```

### Other Steps to Reproduce

I think, someone screw up with the hidden file on windows, because the appData is not hidden like it supposed to be.

I don't think there is a link between the ESP32 and hidden file, but I mention it because it looks like the Arduino is not allowed to go find the ESP32 core...

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-03-08:
Try putting it in a directory without a ampersand (&) in the name.

VojtechBartoska, 2022-04-11:
good catch @lbernstone.

Related issue: https://github.com/espressif/arduino-esp32/issues/5694

VojtechBartoska, 2022-04-11:
Can I close this @CrazyLionQC?

CrazyLionQC, 2022-04-11:
Yess totally. 

Unfortunately the windows user name is the issue with the special character "&" .

I just installed arduino in the guest user and use it from my main user. Everything works now, thanks you !
( and sorry for the late reply )

closed 2022-04-12 by VojtechBartoska

DimitriiVM, 2024-03-21:
FQBN: esp32:esp32:esp32doit-devkit-v1
Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11
Using core 'esp32' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11

cmd /c if exist "D:\\Project\\ARDUINO ESP32\\BluetoothSerialTEST\\BLE_EddystoneURL_Beacon\\partitions.csv" COPY /y "D:\\Project\\ARDUINO ESP32\\BluetoothSerialTEST\\BLE_EddystoneURL_Beacon\\partitions.csv" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\partitions.csv"
cmd /c if not exist "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\partitions.csv" if exist "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\variants\\doitESP32devkitV1\\partitions.csv" COPY "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\variants\\doitESP32devkitV1\\partitions.csv" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\partitions.csv"
cmd /c if not exist "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\partitions.csv" COPY "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\tools\\partitions\\default.csv" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\partitions.csv"
�����஢��� 䠩���:         1.
cmd /c IF EXIST "D:\\Project\\ARDUINO ESP32\\BluetoothSerialTEST\\BLE_EddystoneURL_Beacon\\bootloader.bin" ( COPY /y "D:\\Project\\ARDUINO ESP32\\BluetoothSerialTEST\\BLE_EddystoneURL_Beacon\\bootloader.bin" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\BLE_EddystoneURL_Beacon.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\variants\\doitESP32devkitV1\\bootloader.bin" ( COPY "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\variants\\doitESP32devkitV1\\bootloader.bin" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\BLE_EddystoneURL_Beacon.ino.bootloader.bin" ) ELSE ( "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.5.1/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\BLE_EddystoneURL_Beacon.ino.bootloader.bin" "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\tools\\sdk\\esp32\\bin\\bootloader_dio_80m.elf" ) )
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
cmd /c if exist "D:\\Project\\ARDUINO ESP32\\BluetoothSerialTEST\\BLE_EddystoneURL_Beacon\\build_opt.h" COPY /y "D:\\Project\\ARDUINO ESP32\\BluetoothSerialTEST\\BLE_EddystoneURL_Beacon\\build_opt.h" "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\build_opt.h"
cmd /c if not exist "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\build_opt.h" type nul > "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D\\build_opt.h"
cmd /c type nul > "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino\\sketches\\1ED5F1FDFC1EF1A703C9126E50902B1D/file_opts"
Detecting libraries used...
/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.5" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/newlib/platform_include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions/freertos -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/port/xtensa/include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32 -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/port/esp32 -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/heap/include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/log/include -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/lwip/include/apps -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1 [... cut ...] 1D/file_opts -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\cores\esp32 -IC:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\variants\doitESP32devkitV1 C:\Users\Admin\AppData\Local\Temp\arduino\sketches\1ED5F1FDFC1EF1A703C9126E50902B1D\sketch\BLE_EddystoneURL_Beacon.ino.cpp -o nul
exec: "/bin/xtensa-esp32-elf-g++": file does not exist

Compilation error: exec: "/bin/xtensa-esp32-elf-g++": file does not exist

Что делать? ччто значит ошибка?