espressif/arduino-esp32 #6388: BLE Advertising type are not defined in v4.4

tsctrl, 2022-03-08:
### Board

ESP32 Dev Module

### Device Description

esp32 devkit

### Hardware Configuration

esp32 devkit

### Version

latest master

### IDE Name

IDF Component

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am stuck and with this issue where i am not able to find the root cause, i have been enabling and disabling various of BLE option in menuconfig but still build failed with the error type are still undefined as below:

In file included from ../components/arduino/libraries/BLE/src/BLEServer.h:20,
`BLEAdvertising.h:72:2: error: 'esp_ble_adv_data_t' does not name a type; did you mean 'esp_ble_adv_data_type'`
`/components/arduino/libraries/BLE/src/BLEAdvertising.h:73:2: error: 'esp_ble_adv_params_t' does not name a type; did you mean 'esp_ble_sm_param_t'?`
`/components/arduino/libraries/BLE/src/BLEScan.h:34:2: error: 'esp_ble_gap_phy_t' does not name a type; did you mean 'esp_ble_key_t'?`
`../components/arduino/libraries/BLE/src/BLEUtils.h:59:21: error: expected ';' at end of member declaration`
`../components/arduino/libraries/BLE/src/BLEUtils.h:59:65: error: expected ')' before 'searchEvt'`
`../components/arduino/libraries/BLE/src/BLEScan.h:118:24: error: 'esp_ble_gap_periodic_adv_report_t' has not been declared`

is this related to v4.4 are not supporting the ble advertising modules or is there additional combination or pair to be enabled in the menuconfig to load this type?
i have check the CMakeList to define all the required library properly. but yet not sure is there is anything i miss.

Anyone have the same issue on 4.4 or any suggestion for me to investigate futher?

Thank you

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-09:
what is esp32s1? Single Core ESP32?

tsctrl, 2022-03-11:
the exact config is BLE_42_FEATURE_SUPPORT which only enabled for c3.

VojtechBartoska, 2022-04-11:
Hello, are you able to test this on Arduino core esp32 v2.0.3-rc1? Thanks!

alwint3r, 2022-04-25:
Hi @VojtechBartoska,

I had a similar issue and I have tried using V2.0.3-RC1 tag but the issue remains.
I tested using PlatformIO and here is my platformio.ini configuration

```
[env:ttgo-t-watch]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3-RC1
board = ttgo-t-watch
framework = arduino
```

alwint3r, 2022-04-25:
I managed to remove the compilation errors by adding the following into my `build_flags` configuration in platformio.ini

```
-DBLE_42_FEATURE_SUPPORT=TRUE -DBLE_50_FEATURE_SUPPORT=TRUE
```

Here's my complete platformio.ini

```
[env:ttgo-t-watch]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = ttgo-t-watch
framework = arduino
platform_packages =
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3-RC1
build_flags = -DBLE_42_FEATURE_SUPPORT=TRUE -DBLE_50_FEATURE_SUPPORT=TRUE
```

VojtechBartoska, 2022-04-26:
@alwint3r Thanks for testing, I'm adding this to our Roadmap and we will take a look on it.

spaceAngel, 2022-06-03:
Hey there,

 have same issue with TTGO V3... 
When (approx) will you fix this? 

Thanks :-)

closed 2025-01-16 by Parsaabasi