espressif/arduino-esp32 #6389: Reboots when using external ram within a task.

zekageri, 2022-03-08:
### Board

esp-wrover-kit

### Device Description

This is a custom PCB, but the issue presist with dev modul with the exact same chip.

### Hardware Configuration

There is an ethernet board, an i2c RTC module, an SPI tft dispaly, Modbus on hardwareSerial1 and couple of pins used for reset and hardware reset.

### Version

latest master

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When printing ESP.getPsramSize() from setup() it is working fine, however when creating a task ( pinned to core or not ) it is crashing immidiately. PlatformIO.ini file has psram flag and i got psram enabled message on serial.

**PIO Ini:**

```C
[env:esp32dev]
;platform    = espressif32
platform    = https://github.com/platformio/platform-espressif32.git
board       = esp-wrover-kit
framework   = arduino

board_build.f_cpu   = 240000000L

upload_port     = COM6
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = ./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags = -DBOARD_HAS_PSRAM
              -mfix-esp32-psram-cache-issue
              -DCORE_DEBUG_LEVEL=5
              -D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
              -D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
              
extra_scripts = mklittlefs/replace_fs.py
```

### Sketch

```cpp
void setup() {
    Serial.begin(115200,SERIAL_8N1,DEBUG_RX,DEBUG_TX);
   
    log_d("Total heap: %d", ESP.getHeapSize());
    log_d("Free heap: %d", ESP.getFreeHeap());
    log_d("Total PSRAM: %d", ESP.getPsramSize());
    log_d("Free PSRAM: %d", ESP.getFreePsram());

    // Working fine.

    xTaskCreate(
        psLoopTask,     /* Task function. */
        "HSH_PS_TASK",  /* name of task. */
        15000,          /* Stack size of task */
        NULL,           /* parameter of the task */
        2,              /* priority of the task */
        NULL
    );
}

void loop() {
    vTaskDelete(NULL);
}

void psLoopTask(void* parameter) {
    log_d("Total heap: %d", ESP.getHeapSize());
    log_d("Free heap: %d", ESP.getFreeHeap());
    log_d("Total PSRAM: %d", ESP.getPsramSize());
    log_d("Free PSRAM: %d", ESP.getFreePsram());

    // Crash in here.
}
```

### Debug Message

```plain
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ea0c:0x3ffe7880 0x4008ec85:0x3ffe78a0 0x40089a85:0x3ffe78c0 0x40089bb1:0x3ffe78f0 0x40118fcf:0x3ffe7910 0x40115021:0x3ffe7bd0 0x40114f91:0x3ffe7c20 0x40090cbd:0x3ffe7c50 0x40093252:0x3ffe7c80 0x400935c9:0x3ffe7ca0 0x4013c8f2:0x3ffe7cc0 0x400f5ca9:0x3ffe7d00 0x400d8228:0x3ffe7d40 0x400d876f:0x3ffe7d60 0x4008fc8a:0x3ffe7d80
  #0  0x4008ea0c:0x3ffe7880 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008ec85:0x3ffe78a0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40089a85:0x3ffe78c0 in lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
  #3  0x40089bb1:0x3ffe78f0 in _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
  #4  0x40118fcf:0x3ffe7910 in _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
  #5  0x40115021:0x3ffe7bd0 in fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
  #6  0x40114f91:0x3ffe7c20 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
  #7  0x40090cbd:0x3ffe7c50 in vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #8  0x40093252:0x3ffe7c80 in multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) multi_heap_get_info_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:726
  #9  0x400935c9:0x3ffe7ca0 in multi_heap_get_info at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:327      
  #10 0x4013c8f2:0x3ffe7cc0 in heap_caps_get_info at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:371
  #11 0x400f5ca9:0x3ffe7d00 in EspClass::getPsramSize() at C:\Users\Pc\.platformio\packages\framework-arduinoespressif32\cores\esp32/Esp.cpp:203
  #12 0x400d8228:0x3ffe7d40 in pSystem::refreshStaticData() at src/utilities/performanceSystem.cpp:103
  #13 0x400d876f:0x3ffe7d60 in psLoopTask(void*) at src/utilities/performanceSystem.cpp:180
  #14 0x4008fc8a:0x3ffe7d80 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)   

Rebooting...
```

### Other Steps to Reproduce

No.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zekageri, 2022-03-08:
ps_malloc is working outside of a task but not inside it. Test:

```C
void checkPS_RAM(){
    char * psRam = (char *) ps_malloc (500 * sizeof (char));
    if (psRam == NULL){
        Serial.println ("PS_RAM: ERROR");
    }else{
        Serial.println ("PS_RAM: OK");
        free(psRam);
    }
}

void pSystem::init(byte Core, byte Priority, int StackSize){
    BaseType_t result = xTaskCreatePinnedToCore(psLoopTask, "HSH_PS_TASK", StackSize, NULL, Priority, NULL, Core);
    if(result == pdPASS){
        Serial.printf("HSH_PS_TASK created on core: %d with priority: %d and stack space: %d\n",Core,Priority,StackSize);
    }else{
        Serial.println("HSH_PS_TASK failed to create.");
    }
    /*
    xTaskCreate(
        psLoopTask,
        "HSH_PS_TASK",
        15000,
        NULL,
        1,
        NULL    
    );*/
}

void setup(){
    Serial.begin(115200);
    delay(500);
    checkPS_RAM(); // <-- Working. "PS_RAM: OK"
    delay(500);
    hsh_Performance.init(1,2,15000); // tryed on both cores with various priority and stack depth.
}

void loop(){
    vTaskDelete(NULL);
}

void psLoopTask(void* parameter) {
    //esp_spiram_init();
    checkPS_RAM(); // <-- Rebooting before printing anything.
}
```

**Crash:** 

```C
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

NS - Connecting to Ethernet
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40090cdd  PS      : 0x00060b34  A0      : 0x80092d7e  A1      : 0x3ffe7c90  
A2      : 0x37757434  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060b20  A7      : 0x0000cdcd  A8      : 0x40114d98  A9      : 0x3ffe7d80  
A10     : 0x3f423984  A11     : 0x00000069  A12     : 0x3f423cb8  A13     : 0x3f423a10  
[... 39 lines cut ...]
;platform    = espressif32
platform    = https://github.com/platformio/platform-espressif32.git
board       = esp-wrover-kit
framework   = arduino

board_build.f_cpu   = 240000000L

upload_port     = COM6
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = ./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags = -DBOARD_HAS_PSRAM
              -mfix-esp32-psram-cache-issue
              -DCORE_DEBUG_LEVEL=5
              ;-D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
              ;-D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
              
extra_scripts = mklittlefs/replace_fs.py
```

me-no-dev, 2022-03-08:
`setup()` and `loop()` also run in a task. Maybe try to not execute one line in the task?
```cpp
void testTask(void * arg){
  checkPS_RAM();
  vTaskDelete(NULL);
}
```

Also to note that the version of esp32-arduino that you are using is quite old now. PIO has not officially updated it's support since 1.0.6. There is a workaround on how to get it going with something recent: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-platformio

zekageri, 2022-03-08:
Thank you for your response.
The task actually looks like this:

```C
void psLoopTask(void* parameter) {
    //esp_spiram_init();
    hsh_Performance.pinSetup();
    checkPS_RAM(); // <-- never gets past this line...
    //hsh_Performance.refreshStaticData();
    for (;;) {
        hsh_Performance.sendAdminData();
        hsh_Performance.restart();
        hsh_Performance.hwReset();
        //hsh_Performance.factoryPinHandler();
        vTaskDelay(10);
    }
}
```

I will upgrade esp32-arduino and see if that solves the problem.

zekageri, 2022-03-08:
I upgraded but this is far worse.

```C
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[     2][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (428) gpio: gpio_set_level(2rx(I:A%=�output gpio_num error
[     6][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   862][D][esp32-hal-psram.c:92] psramInit(): PSRAM enabled
[  1902][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
[  1902][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (1898) gpio: gpio_set_level(226): GPIO output gpio_num error

NS - ConnHSH_PS_TASK created on core: 1 with priority: 2 and stack space: 15000
FS - Filesystem mounted
FS - Config file read successfully
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40091ff8  PS      : 0x00060d33  A0      : 0x80093e34  A1      : 0x3ffaf9b0  
A2      : 0xdf010000  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060d23
A6      : 0x00060d20  A7      : 0x0000cdcd  A8      : 0x0000cdcd  A9      : 0xffffffff  
A10     : 0x00060d23  A11     : 0x3f8008d4  A12     : 0x3ffb584c  A13     : 0x00000000
A14     : 0x9f810000  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xdf010000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x40091ff5:0x3ffaf9b00x40093e31:0x3ffaf9f0 0x400940d3:0x3ffafa10 0x40084301:0x3ffafa30 0x400944d5:0x3ffafa50 0x401145ff:0x3ffafa70 0x4011465b:0x3ffafa90 0x4011468f:0x3ffafab0 0x4011ad89:0x3ffafad0 0x40110ced:0x3ffafaf0
  #0  0x40091ff5:0x3ffaf9b00 in compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:288

ELF file SHA256: 0000000000000000

Rebooting...
```

I was able to use all my GPIO pins before. There was no Invalid pin selected message at all.

zekageri, 2022-03-08:
This restart was because the `WiFiEvent_t` type in the ethernet connection callback.

I used `system_event_id_t` before, replaced with `arduino_event_id_t` and crashing begins.
But i use `WiFiEvent_t` i get compiler warning: 

```C
src/utilities/networkSystem.cpp: In function 'void ethEvent(arduino_event_id_t)':
src/utilities/networkSystem.cpp:115:18: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     if( event == SYSTEM_EVENT_ETH_START ){
                  ^~~~~~~~~~~~~~~~~~~~~~
src/utilities/networkSystem.cpp:117:24: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     }else if( event == SYSTEM_EVENT_ETH_GOT_IP ){
                        ^~~~~~~~~~~~~~~~~~~~~~~
src/utilities/networkSystem.cpp:137:24: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     }else if( event == SYSTEM_EVENT_ETH_DISCONNECTED ){
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

and also a crash:
```C

CORRUPT HEAP: Bad head at 0x3f8008cc. Expected 0xabba1234 got 0xffffffff

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)       

Backtrace:0x40083e1d:0x3ffaf8a00x4008ee6d:0x3ffaf8c0 0x400944a5:0x3ffaf8e0 0x400940eb:0x3ffafa10 0x40084301:0x3ffafa30 0x400944d5:0x3ffafa50 0x401145ff:0x3ffafa70 0x4011465b:0x3ffafa90 0x4011468f:0x3ffafab0 0x4011ad89:0x3ffafad0 0x40110ced:0x3ffafaf0
  #0  0x40083e1d:0x3ffaf8a00 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000
```

ETH callback:

```C
void ethEvent(WiFiEvent_t event){
    if( event == SYSTEM_EVENT_ETH_START ){
        ETH.setHostname(hsh_fileSystem.config.hostname);
    }else if( event == SYSTEM_EVENT_ETH_GOT_IP ){
        if( !hsh_networkSystem.ethConnected ){
            hsh_networkSystem.stopAP();
            hsh_networkSystem.ethConnected      = true;
            hsh_networkSystem.connectionMode    = 0;
            hsh_networkSystem.connected         = true;
            hsh_networkSystem.shouldCheckETH    = false;
            hsh_fileSystem.config.gateway       = ETH.gatewayIP();
            hsh_fileSystem.config.currentIP     = ETH.localIP();
            hshDisplay.firstDisplay             = true;
            hshDisplay.addNoty("Ethernet connected.");

            #if NETWORK_DEBUG_MODE
                Serial.printf("\nConnected to Ethernet\n");
                Serial.printf("NS - IP:\t%s\n",          ETH.localIP().toString().c_str());
                Serial.printf("NS - MAC:\t%s\n",         ETH.macAddress().c_str());
                Serial.printf("NS - Host:\t%s\n",        ETH.getHostname());
            #endif
            hsh_networkSystem.setupMDNS();
        }
    }else if( event == SYSTEM_EVENT_ETH_DISCONNECTED ){
        hsh_networkSystem.ethConnected      = false;
        hshDisplay.addNoty("Ethernet disconnected.");
        hsh_networkSystem.connectionMode    = -1;
        hsh_networkSystem.connected         = false;
        hsh_networkSystem.ethStartTime      = millis();
        hsh_networkSystem.shouldCheckETH    = true;
        hshDisplay.firstDisplay             = true;

        #if NETWORK_DEBUG_MODE
            Serial.println("\nNS - Ethernet disconnected");
        #endif
    }
}
```

zekageri, 2022-03-08:
Basically everything falls apart with the new library. :(
Suddenly stack overflow on asyncTCP task, ethernet stopped working, had to update LittleFS library.
But the spiram seems to work at least.

zekageri, 2022-03-08:
So the spi ram seems to work.
Increased the asyncTCP task stack, it seems to work.
LittleFS library seems to work except now `file.getLastWrite();` and `file.size();` returns nothing if i did not open the file, just with `file.openNextFile();`. I will investigate it further.

**But the big problem is that Ethernet is crashing my program now. Here is my ethernet approach ( which is worked fine before the upgrade )** 

```C

void ethEvent(WiFiEvent_t event){
    if( event == ARDUINO_EVENT_ETH_START ){
        ETH.setHostname(hsh_fileSystem.config.hostname);
    }else if( event == ARDUINO_EVENT_ETH_GOT_IP ){
        if( !hsh_networkSystem.ethConnected ){
            hsh_networkSystem.stopAP();
            hsh_networkSystem.ethConnected      = true;
            hsh_networkSystem.connectionMode    = 0;
            hsh_networkSystem.connected         = true;
            hsh_networkSystem.shouldCheckETH    = false;
            hsh_fileSystem.config.gateway       = ETH.gatewayIP();
            hsh_fileSystem.config.currentIP     = ETH.localIP();
            hshDisplay.firstDisplay             = true;
            hshDisplay.addNoty("Ethernet connected.");

            #if NETWORK_DEBUG_MODE
                Serial.printf("\nConnected to Ethernet\n");
                Serial.printf("NS - IP:\t%s\n",          ETH.localIP().toString().c_str());
                Serial.printf("NS - MAC:\t%s\n",         ETH.macAddress().c_str());
                Serial.printf("NS - Host:\t%s\n",        ETH.getHostname());
            #endif
            hsh_networkSystem.setupMDNS();
        }
    }else if( event == ARDUINO_EVENT_ETH_DISCONNECTED ){
        hsh_networkSystem.ethConnected      = false;
        hshDisplay.addNoty("Ethernet disconnected.");
        hsh_networkSystem.connectionMode    = -1;
        hsh_networkSystem.connected         = false;
        hsh_networkSystem.ethStartTime      = millis();
        hsh_networkSystem.shouldCheckETH    = true;
        hshDisplay.firstDisplay             = true;

        #if NETWORK_DEBUG_MODE
            Serial.println("\nNS - Ethernet disconnected");
        #endif
    }
}

void networkSystem::connectETH(){
    stopAP();
    #if NETWORK_DEBUG_MODE
        Serial.println("\nNS - Connecting to Ethernet");
    #endif
    shouldCheckETH = true;
    if( !hsh_fileSystem.config.dhcp )
        {ETH.config(hsh_fileSystem.config.ip, hsh_fileSystem.config.gateway, hsh_fileSystem.config.subnet,
            hsh_fileSystem.config.dns, hsh_fileSystem.config.dns2);}
    WiFi.onEvent(ethEvent);
    ETH.begin(
        ETH_PHY_ADDR,
        PIN_PHY_POWER,
        PIN_SMI_MDC,
        PIN_SMI_MDIO,
        ETH_PHY_LAN8720,
        ETH_CLK_MODE
    );
    ethStartTime = millis();
}
```

Crash report:

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00        
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[     2][E][esp32-hal-gpio.c:95] __pinModeȆF'J�م����pin selected
E (422) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1899][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
[  1899][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (1895) gpio: gpio_set_level(226): GPIO output gpio_num error

NS - Connecting to Ethernet
FS - Filesystem mounted

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == 
SPINLOCK_FREE)

Backtrace:0x40083e1d:0x3ffdc5500x4008ee6d:0x3ffdc570 0x400944a5:0x3ffdc590 0x4009209f:0x3ffdc6c0 0x40093e31:0x3ffdc700 0x400940ad:0x3ffdc720 0x4008423a:0x3ffdc740 0x400d3872:0x3ffdc770 0x400d5f21:0x3ffdc790 0x400d642c:0x3ffdc830 0x400d6f2b:0x3ffdc860
  #0  0x40083e1d:0x3ffdc5500 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

Rebooting...
```

closed 2022-03-08 by zekageri