espressif/arduino-esp32 #6390: Transmission delay when Client.Write data received through UART

CheolsoonKwon, 2022-03-08:
### Board

ESP32-WROOM-32D

### Device Description

ESP32-WROOM-32D

### Hardware Configuration

Pins 16 and 17 are connected to UART RX, TX

### Version

v2.0.2

### IDE Name

Arduino

### Operating System

Windows10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

There is a transmission delay when Client.Write data received through UART.

It's around 140ms to 150ms (check Wireshark and oscilloscope)

I set client.setNoDelay(true) and server.setNoDelay(true) in the setup function.

Is there anything more I need to check?

Previously, 1.0.6 Lib version was used, but today it has been changed to 2.0.2 Lib version. Delay increased further from 140ms -> 180ms.

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
#include <HardwareSerial.h>
#include <EEPROM.h>

void setup() 
{
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  EEPROM.begin(64);
  dfSerial.begin(892857, SERIAL_8N1, 16, 17); 
  dfSerial.setRxBufferSize(1024);
  client.setNoDelay(true);
}

void GP_AP_START_Process(void)
{

  if(UART_Received_Data[6] == 9)
  {
    //Server Configuration 
    Serial.println("Configuring access point...");
    // You can remove the password parameter if you want the AP to be open.
    WiFi.softAPConfig(local_IP, gateway, subnet);

    WiFi.softAP(Read_SSID, Read_PW);
    IPAddress myIP = WiFi.softAPIP();
    Serial.print("AP IP address: ");
    Serial.println(myIP);
    server.begin();
    server.setNoDelay(true);
    Serial.println("Server started");
    SSID_PW_Change_Flag = 2;
    SSID_PW_Diff_Flag = 2;
    Serial.println("AP START");
  }
}

void GA_Send_Res_WiFi()
{
  int i = 0;
  Serial.println("Send WiFi Message"); 

  for(i=7;i<=(WiFi_Payload + 7);i++)
  {
    UART_Received_Data[i-7] = UART_Received_Data[i];
  }
  client.write(UART_Received_Data,WiFi_Payload);

  client.flush();
  WiFi_Payload = 0;
  for(i=0;i<=1024;i++)
  {
    UART_Received_Data[i] = 0;
  }
}
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-08:
most of your example sketch is missing. We need something that we can compile and test.

CheolsoonKwon, 2022-03-10:
> most of your example sketch is missing. We need something that we can compile and test.

Source Code Update completed. Please review.

bertmelis, 2022-03-11:
If you want, I can also send you my hourly rate for reviewing.

https://en.wikipedia.org/wiki/Minimal_reproducible_example

VojtechBartoska, 2022-04-11:
Hello, are you able to test this on Arduino core esp32 v2.0.3-rc1? Thanks!

CheolsoonKwon, 2022-04-11:
> Hello, are you able to test this on Arduino core esp32 v2.0.3-rc1? Thanks!

[image]

In my arduino program, the latest version of esp32 in the board manager is 2.0.2?

VojtechBartoska, 2022-04-11:
You need to change to Development releases link, take a look here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

CheolsoonKwon, 2022-04-13:
> You need to change to Development releases link, take a look here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

[image]

I tested it with the version you gave me (2.0.3).

Take a look at the Wireshark screenshot above.

192.168.10.1 - ESP32 Module
192.168.10.2 - PC Wi-Fi Module

Number 913: At 29.08 seconds, the PC sent a JSON message to the ESP32.
Number 916: At 29.23 seconds, the ESP32 sends an ACK to the PC.

There is a delay of about 140~150ms between 913 and 916.

ESP32 seems to be late with ACK. Any other solutions?

CheolsoonKwon, 2022-04-13:
> You need to change to Development releases link, take a look here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

Version 2.0.3 seems to have a problem with the target board and UART communication.
In version 2.0.2, ACK delay occurs, but UART communicates is working.

me-no-dev, 2022-04-13:
@SuGlider PTAL

@CheolsoonKwon that is a bit too much of code to go through :)

CheolsoonKwon, 2022-04-13:
> @SuGlider PTAL
> 
> @CheolsoonKwon that is a bit too much of code to go through :)

you mean,
Does it mean that the content of the source code I uploaded is long? T.T.

SuGlider, 2022-04-18:
@CheolsoonKwon 
The source code you sent seems to be part of a bigger system. At this point it is impossible to run any test case with it.
I'm not sure if the "transmission delay" is due to UART processing... but looking into your code I can see a few changes you can try to apply:

#### UART call for ```setRxBufferSize()``` must come before ```begin()``` -- please turn on ```Core Debug Level: "Verbose"``` in order to see warning messages from Arduino APIs.

``` cpp
  // right order is:
  dfSerial.setRxBufferSize(1024);
  dfSerial.begin(892857, SERIAL_8N1, 16, 17); 
```

#### Instead of running the loop wating for incoming data from UART, use ```onReceive()``` to react fast, like an UART ISR, to any UART packet that arrives to the board. 

This is a new API that can be used to run a callback function as soon as UART data arrives.
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L176-L229

``` cpp
  // on setup() it can add a callback for whenever UART data comes in:
  dfSerial.onReceive(Received_UART_Data);
  // no more need to run "Received_UART_Data()" in loop().
```

SuGlider, 2022-04-18:
@CheolsoonKwon 
My suggestion is to break the sketch in a few pieces, use some [stub/mock](https://en.wikipedia.org/wiki/Method_stub) code in order to simplify it and find out exactly what is root cause of the delay you are seeing.

@VojtechBartoska 
There is no much I can do with the current code used to demonstrate the issue.

VojtechBartoska, 2022-05-04:
Closing the issue as we are not able to reproduce it.

closed 2022-05-04 by VojtechBartoska

ballet-baxter, 2025-10-17:
@me-no-dev https://vaultedproduction43.github.io/dfUhZIwH4LjuW031nhCB1WQ7qDdSGizOBCuxkbfXfQhi4jE4iaY/