espressif/arduino-esp32 #6393: Ethernet event handler will crash the board on callback.

zekageri, 2022-03-08:
### Board

esp-wrover-kit

### Device Description

It is a custom PCB but it can be reproduced with any other.

### Hardware Configuration

Ethernet

### Version

latest master

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Upgraded to the latest arduino-esp32 and suddenly the ethernet callback is crashing my board.

### Sketch

```cpp
#define ETH_PHY_ADDR  1
#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

void ethEvent(WiFiEvent_t event){
    if( event == ARDUINO_EVENT_ETH_START ){
        ETH.setHostname(hsh_fileSystem.config.hostname);
    }else if( event == ARDUINO_EVENT_ETH_GOT_IP ){
                Serial.printf("\nConnected to Ethernet\n");
                Serial.printf("NS - IP:\t%s\n",          ETH.localIP().toString().c_str());
                Serial.printf("NS - MAC:\t%s\n",         ETH.macAddress().c_str());
                Serial.printf("NS - Host:\t%s\n",        ETH.getHostname());
    }else if( event == ARDUINO_EVENT_ETH_DISCONNECTED ){
       Serial.println("\nNS - Ethernet disconnected");
    }
}

void setup(){
ETH.begin(
        ETH_PHY_ADDR,
        PIN_PHY_POWER,
        PIN_SMI_MDC,
        PIN_SMI_MDIO,
        ETH_PHY_LAN8720,
        ETH_CLK_MODE
    );
    WiFi.onEvent(ethEvent);
}
```

### Debug Message

```plain
Core  0 register dump:
PC      : 0x40091ff8  PS      : 0x00060333  A0      : 0x80093e34  A1      : 0x3ffdc7e0
A2      : 0xdf010000  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060323  
A6      : 0x00060320  A7      : 0x0000cdcd  A8      : 0x0000cdcd  A9      : 0xffffffff
A10     : 0x3ffe1270  A11     : 0x00000004  A12     : 0x00000004  A13     : 0x3ffb9270  
A14     : 0x9f810000  A15     : 0x003fffff  SAR     : 0x00000015  EXCCAUSE: 0x0000001d
EXCVADDR: 0xdf010000  LBEG    : 0x40104868  LEND    : 0x401048ac  LCOUNT  : 0x0000003c  

Backtrace:0x40091ff5:0x3ffdc7e00x40093e31:0x3ffdc820 0x400940ad:0x3ffdc840 0x4008423a:0x3ffdc860 0x400d38f6:0x3ffdc890 0x400d5fa5:0x3ffdc8b0 0x400d64b0:0x3ffdc950 0x400d6faf:0x3ffdc980 
  #0  0x40091ff5:0x3ffdc7e00 in compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:288

ELF file SHA256: 0000000000000000

Rebooting...
```

### Other Steps to Reproduce

Don't know if there is any other way to setup the ethernet.
The version before this, i used other event types like this:

```C
void ethEvent(WiFiEvent_t event){
    if( event == SYSTEM_EVENT_ETH_START ){
        ETH.setHostname(hsh_fileSystem.config.hostname);
    }else if( event == SYSTEM_EVENT_ETH_GOT_IP ){
        Serial.printf("\nConnected to Ethernet\n");
        Serial.printf("NS - IP:\t%s\n",          ETH.localIP().toString().c_str());
        Serial.printf("NS - MAC:\t%s\n",         ETH.macAddress().c_str());
        Serial.printf("NS - Host:\t%s\n",        ETH.getHostname());
    }else if( event == SYSTEM_EVENT_ETH_DISCONNECTED ){
         Serial.println("\nNS - Ethernet disconnected");
    }
}
```

Worked fine. But these now throws compiler warnings:

```C
src/utilities/networkSystem.cpp: In function 'void ethEvent(arduino_event_id_t)':
src/utilities/networkSystem.cpp:115:18: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     if( event == SYSTEM_EVENT_ETH_START ){
                  ^~~~~~~~~~~~~~~~~~~~~~
src/utilities/networkSystem.cpp:117:24: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     }else if( event == SYSTEM_EVENT_ETH_GOT_IP ){
                        ^~~~~~~~~~~~~~~~~~~~~~~
src/utilities/networkSystem.cpp:137:24: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
     }else if( event == SYSTEM_EVENT_ETH_DISCONNECTED ){
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zekageri, 2022-03-09:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40093650  PS      : 0x00060133  A0      : 0x80093e65  A1      : 
0x3ffd50d0
A2      : 0x3f800014  A3      : 0xffffffff  A4      : 0x3ffc5cc4  A5      : 
0x00060123
A6      : 0x00060120  A7      : 0x00000001  A8      : 0x00000000  A9      : 
0x0000000f
A10     : 0x3f800794  A11     : 0x000003f4  A12     : 0x00000015  A13     : 
0x00060123
A14     : 0x00060123  A15     : 0xfffffffc  SAR     : 0x0000001b  EXCCAUSE: 
0x0000001c
EXCVADDR: 0x00000003  LBEG    : 0x401048cc  LEND    : 0x40104910  LCOUNT  : 
0x0000003c

Backtrace:0x4009364d:0x3ffd50d00x40093e62:0x3ffd50f0 0x400940b4:0x3ffd5110 0x4008423a:0x3ffd5130 0x400d3942:0x3ffd5160 0x400d5f95:0x3ffd5180 0x400d649c:0x3ffd5220 0x400d6f9b:0x3ffd5250 
  #0  0x4009364d:0x3ffd50d00 in block_locate_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:445
      (inlined by) tlsf_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:757

ELF file SHA256: 0000000000000000
```

**Right after:**

```C
#define ETH_PHY_ADDR  1
#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

ETH.begin(
        ETH_PHY_ADDR,
        PIN_PHY_POWER,
        PIN_SMI_MDC,
        PIN_SMI_MDIO,
        ETH_PHY_LAN8720,
        ETH_CLK_MODE
    );
```

mrengineer7777, 2022-03-09:
1) Crash issue appears to be on memory allocation (malloc).  No idea what changed in the core, so can't help here.

2) Compile warnings on your WiFi event handler are due to a recent type change in Arduino core library.

WiFiType.h:
`#define WiFiEvent_t  arduino_event_id_t`

WiFiGeneric.h:
```
typedef enum {
	ARDUINO_EVENT_WIFI_READY = 0,
	ARDUINO_EVENT_WIFI_SCAN_DONE,
	ARDUINO_EVENT_WIFI_STA_START,
	ARDUINO_EVENT_WIFI_STA_STOP,
	ARDUINO_EVENT_WIFI_STA_CONNECTED,
	ARDUINO_EVENT_WIFI_STA_DISCONNECTED,
	ARDUINO_EVENT_WIFI_STA_AUTHMODE_CHANGE,
	ARDUINO_EVENT_WIFI_STA_GOT_IP,
	ARDUINO_EVENT_WIFI_STA_GOT_IP6,
	ARDUINO_EVENT_WIFI_STA_LOST_IP,
	ARDUINO_EVENT_WIFI_AP_START,
	ARDUINO_EVENT_WIFI_AP_STOP,
	ARDUINO_EVENT_WIFI_AP_STACONNECTED,
	ARDUINO_EVENT_WIFI_AP_STADISCONNECTED,
	ARDUINO_EVENT_WIFI_AP_STAIPASSIGNED,
	ARDUINO_EVENT_WIFI_AP_PROBEREQRECVED,
	ARDUINO_EVENT_WIFI_AP_GOT_IP6,
	ARDUINO_EVENT_WIFI_FTM_REPORT,
	ARDUINO_EVENT_ETH_START,
	ARDUINO_EVENT_ETH_STOP,
	ARDUINO_EVENT_ETH_CONNECTED,
	ARDUINO_EVENT_ETH_DISCONNECTED,
	ARDUINO_EVENT_ETH_GOT_IP,
	ARDUINO_EVENT_ETH_GOT_IP6,
	ARDUINO_EVENT_WPS_ER_SUCCESS,
	ARDUINO_EVENT_WPS_ER_FAILED,
	ARDUINO_EVENT_WPS_ER_TIMEOUT,
	ARDUINO_EVENT_WPS_ER_PIN,
	ARDUINO_EVENT_WPS_ER_PBC_OVERLAP,
	ARDUINO_EVENT_SC_SCAN_DONE,
	ARDUINO_EVENT_SC_FOUND_CHANNEL,
	ARDUINO_EVENT_SC_GOT_SSID_PSWD,
	ARDUINO_EVENT_SC_SEND_ACK_DONE,
	ARDUINO_EVENT_PROV_INIT,
	ARDUINO_EVENT_PROV_DEINIT,
	ARDUINO_EVENT_PROV_START,
	ARDUINO_EVENT_PROV_END,
	ARDUINO_EVENT_PROV_CRED_RECV,
	ARDUINO_EVENT_PROV_CRED_FAIL,
	ARDUINO_EVENT_PROV_CRED_SUCCESS,
	ARDUINO_EVENT_MAX
} arduino_event_id_t;
```

A couple weeks ago I had to update my code to handle the new datatype for WiFi.onEvent().

zekageri, 2022-03-10:
Thank you for your help. I have seen the new event types and changed them. The thing is that if I go with wifi everything works. My sketch runs without problem. If I replace wifi connection with ethernet connection the esp reboots right on ETH.begin () with these error messages.

😤

mrengineer7777, 2022-03-10:
> ```c
> #define ETH_PHY_ADDR  1
> #define PIN_PHY_POWER 17
> ```

Your defines don't match the defaults in ETH.h.  Is it possible one of them is wrong?

zekageri, 2022-03-10:
> > ```c
> > #define ETH_PHY_ADDR  1
> > #define PIN_PHY_POWER 17
> > ```
> 
> Your defines don't match the defaults in ETH.h.  Is it possible one of them is wrong?

Yeah i know that. It's our hardware design choise. It was working before the update with the exact same defines. :/

Jason2866, 2022-03-10:
What Arduino version you have used? Release 2.0.2 or GIT version?
We had problems with ethernet too and finally found the issue. Provided this PR #6340

zekageri, 2022-03-10:
I am using the upstream version

```C
[env:arduino-esp32]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages =
>    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

Jason2866, 2022-03-10:
The ethernet PR is at the moment only in branch `esp32-s3-support`

zekageri, 2022-03-11:
I will try that. Thank you for your help, i appreciate it.

zekageri, 2022-03-11:
I copyed this build but i got a crash when i try to allocate space from PSRAM.

```C
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40091ff8  PS      : 0x00060933  A0      : 0x80093e34  A1      : 0x3ffcff10  
A2      : 0xdf010000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060923      
A6      : 0x00060920  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x3ffdb7b8  A11     : 0x00000004  A12     : 0x00000004  A13     : 0x3ffb9270      
A14     : 0x9f810000  A15     : 0x003fffff  SAR     : 0x00000015  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xdf010000  LBEG    : 0x40101804  LEND    : 0x40101848  LCOUNT  : 0x0000003c      

Backtrace:0x40091ff5:0x3ffcff100x40093e31:0x3ffcff50 0x400940ad:0x3ffcff70 0x4008423a:0x3ffcff90 0x400d3672:0x3ffcffc0 0x400d5d8f:0x3ffcffe0 0x400d6254:0x3ffd0080 0x400d6d1f:0x3ffd00b0
  #0  0x40091ff5:0x3ffcff100x40093e31:0x3ffcff50 in compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:288
  #1  0x400940ad:0x3ffcff70 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:229
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:219
  #2  0x4008423a:0x3ffcff90 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:145
  #3  0x400d3672:0x3ffcffc0 in SpiRamAllocator::allocate(unsigned int) at src/utilities/globals.h:8
      (inlined by) ArduinoJson6192_F1::AllocatorOwner<SpiRamAllocator>::allocate(unsigned int) at lib/ArduinoJson-6.x/src/ArduinoJson/Document/BasicJsonDocument.hpp:20
      (inlined by) ArduinoJson6192_F1::BasicJsonDocument<SpiRamAllocator>::allocPool(unsigned int) at lib/ArduinoJson-6.x/src/ArduinoJson/Document/BasicJsonDocument.hpp:137
      (inlined by) ArduinoJson6192_F1::BasicJsonDocument<SpiRamAllocator>::BasicJsonDocument(unsigned int, SpiRamAllocator) at lib/ArduinoJson-6.x/src/ArduinoJson/Document/BasicJsonDocument.hpp:44
  #4  0x400d5d8f:0x3ffcffe0 in fSystem::readConfig() at src/utilities/fileSystem.cpp:91     
  #5  0x400d6254:0x3ffd0080 in fSystem::mount() at src/utilities/fileSystem.cpp:36
      (inlined by) fSystem::mount() at src/utilities/fileSystem.cpp:23
  #6  0x400d6d1f:0x3ffd00b0 in fsLoopTask(void*) at src/utilities/fileSystem.cpp:694        

ELF file SHA256: 0000000000000000

Rebooting...
```

It didn't even get to the ethernet part.

I'm using ArduinoJSON and allocating memory from the external ram with this allocator template for the json objects:

```C
#include <Arduino.h>
#include <ArduinoJson.h>

struct SpiRamAllocator {
    void* allocate(size_t size) {return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);}
    void deallocate(void* pointer) {heap_caps_free(pointer);}
    void* reallocate(void* ptr, size_t new_size) {return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);}
};
using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
```

And with that build ( just like with the old PIO build ) it crashed at this part:

```C
SpiRamJsonDocument doc(CONFIG_SIZE); // <-- Like right here
DeserializationError error = deserializeJson(doc, configFile);
if (!error) {}
```

zekageri, 2022-03-11:
With this url: `platform    = https://github.com/platformio/platform-espressif32.git#master` my sketch can't even compile

```C
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe
Compiling .pio\build\arduino-esp32\src\main.cpp.o
Compiling .pio\build\arduino-esp32\src\utilities\fileSystem.cpp.o
Compiling .pio\build\arduino-esp32\src\utilities\geolocate.cpp.o
Compiling .pio\build\arduino-esp32\src\utilities\modBus.cpp.o
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\arduino-esp32\src\utilities\networkSystem.cpp.o
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\arduino-esp32\src\utilities\performanceSystem.cpp.o
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
Compiling .pio\build\arduino-esp32\src\utilities\serverSystem.cpp.o
Compiling .pio\build\arduino-esp32\src\utilities\tftDisplay.cpp.o
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
In file included from src/main.cpp:7:0:
src/utilities/networkSystem.h:12:0: warning: "ETH_PHY_ADDR" redefined
 #define ETH_PHY_ADDR  1
 ^
In file included from src/utilities/networkSystem.h:6:0,
                 from src/main.cpp:7:
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.h:29:0: note: this is the location of the previous definition
 #define ETH_PHY_ADDR 0
 ^
In file included from src/utilities/geolocate.cpp:2:0:
src/utilities/networkSystem.h:12:0: warning: "ETH_PHY_ADDR" redefined
 #define ETH_PHY_ADDR  1
 ^
In file included from src/utilities/networkSystem.h:6:0,
                 from src/utilities/geolocate.cpp:2:
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.h:29:0: note: this is the location of the previous definition
 #define ETH_PHY_ADDR 0
 ^
In file included from src/utilities/networkSystem.cpp:1:0:
src/utilities/networkSystem.h:12:0: warning: "ETH_PHY_ADDR" redefined
 #define ETH_PHY_ADDR  1
 ^
In file included from src/utilities/networkSystem.h:6:0,
                 from src/utilities/networkSystem.cpp:1:
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.h:29:0: note: this is the location of the previous definition
 #define ETH_PHY_ADDR 0
 ^
In file included from src/utilities/tftDisplay.cpp:3:0:
src/utilities/networkSystem.h:12:0: warning: "ETH_PHY_ADDR" redefined
 #define ETH_PHY_ADDR  1
 ^
In file included from src/utilities/networkSystem.h:6:0,
                 from src/utilities/tftDisplay.cpp:3:
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.h:29:0: note: this is the location of the previous definition
 #define ETH_PHY_ADDR 0
 ^
In file included from src/utilities/serverSystem.h:10:0,
                 from src/utilities/performanceSystem.cpp:3:
src/utilities/networkSystem.h:12:0: warning: "ETH_PHY_ADDR" redefined
 #define ETH_PHY_ADDR  1
 ^
In file included from src/utilities/networkSystem.h:6:0,
                 from src/utilities/serverSystem.h:10,
                 from src/utilities/performanceSystem.cpp:3:
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.h:29:0: note: this is the location of the previous definition
 #define ETH_PHY_ADDR 0
 ^
In file included from src/utilities/serverSystem.h:10:0,
                 from src/utilities/serverSystem.cpp:1:
[... 61 lines cut ...]
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
Compiling .pio\build\arduino-esp32\lib763\WiFiClientSecure\WiFiClientSecure.cpp.o
Compiling .pio\build\arduino-esp32\lib763\WiFiClientSecure\esp_crt_bundle.c.o
<command-line>:0:19: warning: ISO C++11 requires whitespace after the macro name
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: In static member function 'static bool WiFiGenericClass::setDualAntennaConfig(uint8_t, uint8_t, wifi_rx_ant_t, wifi_tx_ant_t)':
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
<command-line>:0:19: warning: ISO C99 requires whitespace after the macro name
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
*** [.pio\build\arduino-esp32\lib6a7\WiFi\WiFiGeneric.cpp.o] Error 1
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
=================================== [FAILED] Took 15.14 seconds ===================================The terminal process "C:\Users\Pc\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Jason2866, 2022-03-11:
You can NOT use PSRAM and Ethernet with GPIO17. Gpio17 is used for PSRAM.
So it can not be used for Ethernet too.

zekageri, 2022-03-11:
How was it worked with pio build before?

zekageri, 2022-03-11:
We have a complete design with working code.
That setup working like 1.5 year ago without problem.

I'm redesigning the software, changed to the new build and now i can't use that pin?

sauttefk, 2022-03-11:
> I'm redesigning the software, changed to the new build and now i can't use that pin?

If you had used PSRAM ever since and Ethernet worked, then obviously the CLK for Ethernet is either `GPIO0_IN` or `GPIO0_OUT` and **not** `GPIO17`

zekageri, 2022-03-11:
```C
#define ETH_PHY_ADDR  1
#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18
```

These are my defines for the ethernet ( which is currently working with the PIO build ).

So what is the correct define for these?

Still dont know how can it work with #define PIN_PHY_POWER 17

Thanks for all the help. I'm confused right now.

zekageri, 2022-03-11:
Okay so it turns out that the PIO version build does not check for the `PIN_PHY_POWER` and just assumes that i defined -1 for this pin. Now i changed my configuration to

```C
#define ETH_PHY_ADDR  1
#define PIN_PHY_POWER -1 //17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18
```

and it works.

Thank you guys, i really appreciate all of your help and sorry for the trouble!

closed 2022-03-11 by zekageri