espressif/arduino-esp32 #6403: error when update firmware over https on esp32 with w5500

mikeme2009, 2022-03-09:
### Board

custom board

### Device Description

The custom board include:
esp32, w5500, ST232

### Hardware Configuration

no

### Version

latest master

### IDE Name

Platformio

### Operating System

windows 11

### Flash frequency

no

### PSRAM enabled

no

### Upload speed

no

### Description

when I try to update firmware I receive an error.

### Sketch

```cpp
#include "HttpsOTAUpdate.h"
static HttpsOTAStatus_t otastatus;
static const char *url = "https://xxxxxxxxxxx/firmwares/yyyyyyyyyyy.bin"
///// update code
HttpsOTA.onHttpEvent(HttpEvent);
      Serial.println("Starting OTA");
      HttpsOTA.begin(url, certificate);
      Serial.println("Please Wait it takes some time ...");
      otastatus = HttpsOTA.status();
      if(otastatus == HTTPS_OTA_SUCCESS) { 
          Serial.println("Firmware written successfully. To reboot device, call API ESP.restart() or PUSH restart button on device");
      } else if(otastatus == HTTPS_OTA_FAIL) { 
          Serial.println("Firmware Upgrade Fail");
      }
```

### Debug Message

```plain
https://xxxxxxxxxxx/firmwares/yyyyyyyyyyy.bin
Starting OTA
Please Wait it takesassertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 267, function: tcpip_callback_with_block

abort() was called at PC 0x4011008b on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a348:0x3ffd5b90 0x4008a5c5:0x3ffd5bb0 0x4011008b:0x3ffd5bd0 0x4014aeb6:0x3ffd5c00 0x4014ad4d:0x3ffd5c20 0x4013eff1:0x3ffd5c60 0x401c25ea:0x3ffd5cb0 0x401c262f:0x3ffd5d00 0x401c2b45:0x3ffd5d40 0x401c2db4:0x3ffd5d70 0x401bfbf1:0x3ffd5da0 0x401c8d25:0x3ffd5dc0 0x40195260:0x3ffd5de0 0x40195494:0x3ffd5e00 0x40121d62:0x3ffd5e20 0x40121ef8:0x3ffd5e40 0x401221b5:0x3ffd5e60 0x400e2633:0x3ffd5e90 0x4008b5d6:0x3ffd5eb0

  #0  0x4008a348:0x3ffd5b90 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008a5c5:0x3ffd5bb0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4011008b:0x3ffd5bd0 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x4014aeb6:0x3ffd5c00 in tcpip_callback_with_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
  #4  0x4014ad4d:0x3ffd5c20 in netconn_gethostbyname_addrtype at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c:1057
  #5  0x4013eff1:0x3ffd5c60 in lwip_getaddrinfo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/netdb.c:342
  #6  0x401c25ea:0x3ffd5cb0 in resolve_host_name at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp-tls/esp_tls.c:545
  #7  0x401c262f:0x3ffd5d00 in esp_tcp_connect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp-tls/esp_tls.c:545
  #8  0x401c2b45:0x3ffd5d40 in esp_tls_low_level_conn at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp-tls/esp_tls.c:545
  #9  0x401c2db4:0x3ffd5d70 in esp_tls_conn_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp-tls/esp_tls.c:545
  #10 0x401bfbf1:0x3ffd5da0 in ssl_connect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/tcp_transport/transport_ssl.c:72
  #11 0x401c8d25:0x3ffd5dc0 in esp_transport_connect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/tcp_transport/transport.c:252
  #12 0x40195260:0x3ffd5de0 in esp_http_client_connect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/esp_http_client.c:1242
  #13 0x40195494:0x3ffd5e00 in esp_http_client_open at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/esp_http_client.c:1242
  #14 0x40121d62:0x3ffd5e20 in _http_connect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_https_ota/src/esp_https_ota.c:377
  #15 0x40121ef8:0x3ffd5e40 in esp_https_ota_begin at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_https_ota/src/esp_https_ota.c:377
  #16 0x401221b5:0x3ffd5e60 in esp_https_ota at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_https_ota/src/esp_https_ota.c:391
  #17 0x400e2633:0x3ffd5e90 in https_ota_task(void*) at C:\Users\mmmm\.platformio\packages\framework-arduinoespressif32\libraries\Update\src/HttpsOTAUpdate.cpp:47
  #18 0x4008b5d6:0x3ffd5eb0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

### Other Steps to Reproduce

no

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mikeme2009, 2022-04-11:
Hello. I made the test with Arduino core esp32 v2.0.3-rc1. The result:
Initialize Ethernet with DHCP:
20:25:35.079 ->   DHCP assigned IP 192.168.1.64
20:25:36.059 -> Starting OTA
20:25:36.059 -> Please Wait it takes some time ...
20:25:36.106 -> 
20:25:36.106 -> assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)
20:25:36.200 -> 
20:25:36.200 -> 
20:25:36.200 -> Backtrace:0x40083479:0x3ffb47f00x400882a9:0x3ffb4810 0x4008d1b1:0x3ffb4830 0x400e6ca6:0x3ffb4960 0x400f315d:0x3ffb4990 0x40112c96:0x3ffb49d0 0x40111a5d:0x3ffb4a20 0x40111f7d:0x3ffb4ab0 0x401121f7:0x3ffb4ae0 0x400f78dc:0x3ffb4b10 0x40123573:0x3ffb4b30 0x400e3360:0x3ffb4b50 0x400e35e3:0x3ffb4b70 0x400e180e:0x3ffb4b90 0x400e1a65:0x3ffb4bc0 0x400e1e50:0x3ffb4bf0 0x400d435f:0x3ffb4c30 
20:25:36.622 -> 
20:25:36.622 -> 
20:25:36.622 -> 
20:25:36.622 -> 
20:25:36.622 -> ELF file SHA256: 0000000000000000
20:25:36.669 -> 
20:25:36.669 -> Rebooting...

The problem is that the HttpsOTAUpdate.h library work only with wifi.h, and I use Ethernet.h, because I use w5500.

VojtechBartoska, 2022-04-12:
Thanks for testing, adding this to our Backlog. Https needs our check.

closed 2025-01-16 by Parsaabasi