espressif/arduino-esp32 #6405: Integrate the library OPEnSLab-OSU/SSLClient with esp-idf as component

roysG, 2022-03-09:
### Board

Esp32

### Device Description

fsd

### Hardware Configuration

fsf

### Version

latest master

### IDE Name

idf component

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am trying to connect this library:
This library is: https://github.com/OPEnSLab-OSU/SSLClient
with esp-idf as component.

But i get this errors:

[6/9] Linking CXX executable arduino-lib-builder.elf
FAILED: arduino-lib-builder.elf
: && /Users/roysabah/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address CMakeFiles/arduino-lib-builder.elf.dir/project_elf_src_esp32.c.obj -o arduino-lib-builder.elf esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/efuse/libefuse.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/driver/libdriver.a esp-idf/esp_pm/libesp_pm.a esp-idf/mbedtls/libmbedtls.a esp-idf/app_update/libapp_update.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp_phy/libesp_phy.a esp-idf/esp_system/libesp_system.a esp-idf/esp_rom/libesp_rom.a esp-idf/hal/libhal.a esp-idf/vfs/libvfs.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/console/libconsole.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/soc/libsoc.a esp-idf/esp_hw_support/libesp_hw_support.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_timer/libesp_timer.a esp-idf/freertos/libfreertos.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/unity/libunity.a esp-idf/cmock/libcmock.a esp-idf/coap/libcoap.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_hid/libesp_hid.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_lcd/libesp_lcd.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/perfmon/libperfmon.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a esp-idf/button/libbutton.a esp-idf/json_parser/libjson_parser.a esp-idf/json_generator/libjson_generator.a esp-idf/esp_schedule/libesp_schedule.a esp-idf/esp_rainmaker/libesp_rainmaker.a esp-idf/qrcode/libqrcode.a esp-idf/ws2812_led/libws2812_led.a esp-idf/arduino/libarduino.a esp-idf/esp-dsp/libesp-dsp.a esp-idf/esp32-camera/libesp32-camera.a esp-idf/esp_littlefs/libesp_littlefs.a esp-idf/fb_gfx/libfb_gfx.a -Wl,--cref -Wl,--Map="/Users/roysabah/esp/esp32-arduino-lib-builder/build/arduino-lib-builder.map" -Wl,--gc-sections -fno-rtti -fno-lto esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/cmock/libcmock.a esp-idf/unity/libunity.a esp-idf/coap/libcoap.a esp-idf/esp_hid/libesp_hid.a esp-idf/esp_lcd/libesp_lcd.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/libsodium/liblibsodium.a esp-idf/perfmon/libperfmon.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/fatfs/libfatfs.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/esp_rainmaker/libesp_rainmaker.a esp-idf/mqtt/libmqtt.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/json/libjson.a esp-idf/json_parser/libjson_parser.a esp-idf/json_generator/libjson_generator.a esp-idf/esp_schedule/libesp_schedule.a esp-idf/qrcode/libqrcode.a -u _Z5setupv -u _Z4loopv -L /Users/roysabah/esp/esp32-arduino-lib-builder/components/esp-face/lib/esp32 -lcat_face_detect -lhuman_face_detect -ldl esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/efuse/libefuse.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/driver/libdriver.a esp-idf/esp_pm/libesp_pm.a esp-idf/mbedtls/libmbedtls.a esp-idf/app_update/libapp_update.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp_phy/libesp_phy.a esp-idf/esp_system/libesp_system.a esp-idf/esp_rom/libesp_rom.a esp-idf/hal/libhal.a esp-idf/vfs/libvfs.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/console/libconsole.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/soc/libsoc.a esp-idf/esp_hw_support/libesp_hw_support.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_timer/libesp_timer.a esp-idf/freertos/libfreertos.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/ulp/libulp.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libcore.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libespnow.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libmesh.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libnet80211.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libsmartconfig.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libwapi.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/efuse/libefuse.a esp-idf/esp_ipc/libesp_ipc.a esp-idf/driver/libdriver.a esp-idf/esp_pm/libesp_pm.a esp-idf/mbedtls/libmbedtls.a esp-idf/app_update/libapp_update.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/pthread/libpthread.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp_phy/libesp_phy.a esp-idf/esp_system/libesp_system.a esp-idf/esp_rom/libesp_rom.a esp-idf/hal/libhal.a esp-idf/vfs/libvfs.a esp-idf/esp_eth/libesp_eth.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_netif/libesp_netif.a esp-idf/esp_event/libesp_event.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/console/libconsole.a esp-idf/lwip/liblwip.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/soc/libsoc.a esp-idf/esp_hw_support/libesp_hw_support.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_timer/libesp_timer.a esp-idf/freertos/libfreertos.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a esp-idf/ulp/libulp.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /Users/roysabah/esp/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a /Users [... cut ...]
/Users/roysabah/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino/libarduino.a(ec_prime_fast_256.c.obj):(.literal.api_generator+0x4): undefined reference to `br_ec_prime_i15'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

### Sketch

```cpp
....fsdfsf
```

### Debug Message

```plain
....
```

### Other Steps to Reproduce

....

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-03-09:
ESP IDF as Arduino component is problematic.  See #5819.  I strongly recommend using the Arduino framework.

roysG, 2022-03-10:
@mrengineer7777, in case you need to using in flash encryption you can׳t do it with Arduino framework

mrengineer7777, 2022-03-10:
> @mrengineer7777, in case you need to using in flash encryption you can׳t do it with Arduino framework

In that case develop on the IDF framework.  I've spent the last 2 months developing on it and modifying the Arduino libraries I needed to work in IDF. But it's super frustrating and I finally gave up as I couldn't get ESPAsyncWebserver to compile under IDF.  It has dependency to ArduinoJSON.hpp that I couldn't get to work, among other issues.

Jason2866, 2022-03-11:
It seems you are looking for a SSL Client solution. Maybe you can use our modified stripped down BearSSL version (working for esp8266 and all esp32 mcu).
See https://github.com/arendst/Tasmota/tree/development/lib/lib_ssl

VojtechBartoska, 2022-05-30:
Closing this issue as expired. If needed, you can reopen it.

closed 2022-05-30 by VojtechBartoska