espressif/arduino-esp32 #6406: Compile error with current arduino-upstream branch

HDPlayser, 2022-03-10:
### Board

esp-wrover-kit / esp32dev

### Device Description

Custom Board for IoT-Purposes

### Hardware Configuration

All GPIOs are used up. 
SPI-Devices are also connected

### Version

latest master

### IDE Name

PlatformIO

### Operating System

Windows 10 21H2

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I compile my software with the default esp32 version, which includes the ESP-IDF Version 3.3.5 everything works like expected. Now I wanted to update to the lastest branch, which includes the new ESP-IDF version 4.x. At the end of compilation when the compiler is linking. I get the following error: 

> _...
> Compiling .pio\build\esp-wrover-kit\FrameworkArduino\wiring_shift.c.o
> Archiving .pio\build\esp-wrover-kit\libFrameworkArduino.a
> Linking .pio\build\esp-wrover-kit\firmware.elf
> xtensa-esp32-elf-g++: error: Client.map: No such file or directory
> *** [.pio\build\esp-wrover-kit\firmware.elf] Error 1_

The Compiler gives me absolutely no hint where to trace the error.
When I start a fresh project, it compiles flawlessly. 

I want to know why the compiler does not generate the file client.map if I understand this correctly.
Also I could not find any information about the file "Client.map" online. 
If you want, I could share the verbose build file if it helps. 
Somewhere is code then, which the compile does not like, because my simpler project compiles without errors.
I did an verbose build and could not trace the error either.
I changed the Board to another, it did also not compile.

Here is my current Environment configuration:

_

> [env:esp-wrover-kit]
> platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
> platform_packages =
>    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
>    espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch2
> 
> board = esp-wrover-kit
> framework = arduino
[... 481 lines cut ...]
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp-wrover-kit\libFrameworkArduino.a
Linking .pio\build\esp-wrover-kit\firmware.elf
xtensa-esp32-elf-g++: error: Client.map: No such file or directory
*** [.pio\build\esp-wrover-kit\firmware.elf] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-03-10:
Hi @HDPlayser, it's probably problem in PlatformIO. It doesn't support Arduino core for ESP32 from version v2.0 and newest. Take a look on https://github.com/espressif/arduino-esp32/issues/6044 to find a workaround.

Jason2866, 2022-03-10:
@HDPlayser It is Platformio. The Platformio `platform` is too old.
You can try `platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip`
It is a custom setup which i did for project Tasmota

HDPlayser, 2022-03-10:
> @HDPlayser It is Platformio. The Platformio `platform` is too old. Probably the used esptool.py v3.1 is the reason You can try `platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip` It is a custom setup which i did for project Tasmota

Thank you, but unfortunately it also didn`t compile. Same error. 🙄
Currently i am trying to remove step by step parts of the application until it compiles to trace the faulty piece of code. 🤷‍♀️

HDPlayser, 2022-03-10:
> Hi @HDPlayser, it's probably problem in PlatformIO. It doesn't support Arduino core for ESP32 from version v2.0 and newest. Take a look on #6044 to find a workaround.

Thanks for your reply. 
As i mentioned, I can compile a freshly started project even with the same libarys with the newest ESP-IDF, so something is working I guess 😅
> Currently i am trying to remove step by step parts of the application until it compiles to trace the faulty piece of code. 🤷‍♀️

mrengineer7777, 2022-03-10:
@HDPlayser you specify the toolchain in your ini file here `espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch2`.  I believe if you leave that out the specified platform will set the toolchain.
Here's what I'm using.  Thanks to @Jason2866 for keeping TASMOTA updated with the latest fixes.

```
[env]
board = featheresp32
monitor_speed = 115200
build_flags = 
	-DCORE_DEBUG_LEVEL=4

[env:featheresp32ard]
framework = arduino
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
```

HDPlayser, 2022-03-10:
> @HDPlayser you specify the toolchain in your ini file here `espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch2`. I believe if you leave that out the specified platform will set the toolchain. [...]

This is my current build configuration:
It also doesn´t compile with the following error:
But thanks for every possible solution 👍 
```
Archiving .pio\build\esp-wrover-kit\libFrameworkArduino.a
Linking .pio\build\esp-wrover-kit\firmware.elf
xtensa-esp32-elf-g++: error: Client.map: No such file or directory
*** [.pio\build\esp-wrover-kit\firmware.elf] Error 1
```
```
[env:esp-wrover-kit]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip

board = esp-wrover-kit
framework = arduino
monitor_speed = 115200
upload_speed = 921600
build_flags= -O3 -DKMS_MB_DE12_STD=1 -DHWGEN=3 -DCORE_DEBUG_LEVEL=10 -fexceptions
board_build.partitions = partitions_wrover.csv
```

HDPlayser, 2022-03-10:
> I searched for "Client.map" in the .platformio folder, which includes all platforms, frameworks, libraries. There are no references. Is that somewhere in your code or libraries?
> 
> I also tried compiling my project with your settings.
> 
> ```
> [env:esp-wrover-kit]
> platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
> board = esp-wrover-kit
> framework = arduino
> monitor_speed = 115200
> upload_speed = 921600
> build_flags= -O3 -DKMS_MB_DE12_STD=1 -DHWGEN=3 -DCORE_DEBUG_LEVEL=10 -fexceptions
> lib_deps = esphome/ESPAsyncWebServer-esphome@^2.1.0
> ;board_build.partitions = partitions_wrover.csv
> ```
> 
> Note I had to skip the partition, and include a webserver library.
> 
> ```
> RAM:   [=         ]  11.1% (used 36368 bytes from 327680 bytes)
> Flash: [=====     ]  48.6% (used 636525 bytes from 1310720 bytes)
> Building .pio\build\esp-wrover-kit\firmware.bin
> esptool.py v3.2.1
> Creating esp32 image...
> Merged 25 ELF sections
> Successfully created esp32 image.
> =================================================================================================== [SUCCESS] Took 36.49 seconds ====
> ```
> 
> Minor note: I believe CORE_DEBUG_LEVEL maxes out at 5.

There is no such file called _Client.map_ on my whole C-Drive 😂 
I can`t imagine from where this file is referenced. I thought that the compiler is generating this file. 
 
_A: Minor Note: Yes the Field CORE_DEBUG_LEVEL goes normally up to 5 but within my own AsyncHTTPClient it goes up to 10  so it can display super verbose stuff. 😅
Setting it to 0 does absolute nothing, still does not compile 😭_

mrengineer7777, 2022-03-10:
Try skipping the build partition?

HDPlayser, 2022-03-10:
No, sorry. Still the absolute same error 😶
Currently i am trying to remove code step by step until it compiles to track the problem. 
But for this project size this may take a while.

Sys64736, 2022-03-10:
I believe the .map file is generated during build.  Should be in your [build]/[env:name] folder.

This is the first few lines of my [project_name].map file:
------------------------------------------------------------------------
Archive member included to satisfy reference by file (symbol)

C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFi.cpp.o)
                              C:/.pio/build/esp32c3/src/main.cpp.o (WiFi)
C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiAP.cpp.o)
                              C:/.pio/build/esp32c3/src/main.cpp.o (WiFiAPClass::softAP(char const*, char const*, int, int, int, bool))
C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiClient.cpp.o)
                              C:/.pio/build/esp32c3/src/main.cpp.o (WiFiClient::write(Stream&))
C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiGeneric.cpp.o)
                              C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiAP.cpp.o) (get_esp_interface_netif(esp_interface_t))
C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiSTA.cpp.o)
                              C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiGeneric.cpp.o) (WiFiSTAClass::_setStatus(wl_status_t))
C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiScan.cpp.o)
                              C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiGeneric.cpp.o) (WiFiScanClass::_scanDone())

.....goes on for about 70,000 + lines

I believe it's for the symbol and reference function in platformio.

I did have a .map error before with one project and I ended fixing it by removing spaces in my project path and name.

Hope this helps

mrengineer7777, 2022-03-10:
> I believe the .map file is generated during build. Should be in your [build]/[env:name] folder.

Good point.  It appears to be named after the project name.  For my D4500 project it's located at:
\.pio\build\featheresp32ard\D4500.map

HDPlayser, 2022-03-10:
> I believe the .map file is generated during build.  Should be in your [build]/[env:name] folder.
> 
> This is the first few lines of my [project_name].map file:
> ------------------------------------------------------------------------
> Archive member included to satisfy reference by file (symbol)
> 
> C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFi.cpp.o)
>                               C:/.pio/build/esp32c3/src/main.cpp.o (WiFi)
> C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiAP.cpp.o)
>                               C:/.pio/build/esp32c3/src/main.cpp.o (WiFiAPClass::softAP(char const*, char const*, int, int, int, bool))
> C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiClient.cpp.o)
>                               C:/.pio/build/esp32c3/src/main.cpp.o (WiFiClient::write(Stream&))
> C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiGeneric.cpp.o)
>                               C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiAP.cpp.o) (get_esp_interface_netif(esp_interface_t))
> C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiSTA.cpp.o)
>                               C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiGeneric.cpp.o) (WiFiSTAClass::_setStatus(wl_status_t))
> C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiScan.cpp.o)
>                               C:/.pio/build/esp32c3/libc0e/libWiFi.a(WiFiGeneric.cpp.o) (WiFiScanClass::_scanDone())
> 
> .....goes on for about 70,000 + lines
> 
> I believe it's for the symbol and reference function in platformio.  
> 
> I did have a .map error before with one project and I ended fixing it by removing spaces in my project path and name.
> 
> Hope this helps
> 
> [EDIT:  Taking a closer look at the .map file, I think it's generated and then needed by the compiler to finish the job.   I had this error also, drove me nuts for a couple days because I was working with a RFID reader that happen to have a map file, (unrelated), in it's library so I was looking in the wrong spot forever.   I can't remember for sure but I'm almost certain I removed any strange characters and spaces in the project path with underscores and it compiled.    Also another reason why a permanently moved my build path closer to the root.]

This is actually a very good point. My Project is indeed named _MeshSolutions Client_, with an space in between. I think we found the issue. I will try to rename the project tomorrow back at work.

Jason2866, 2022-03-10:
In general do not place project dirs in deeply directories when using Windows.
Windows has a max. path length limitation. If you encounter strange compile or linker errors check this. Avoid any! special character for directories and file names.
Use only characters and numbers.

HDPlayser, 2022-03-11:
**We found the Issue.** It was the space in the Project folder`s name. Just by removing it, it compiled without any problems. 
Thanks to everyone and especially @Sys64736 for help me tracing this error. You just saved me some days of horrible error tracing 😄

closed 2022-03-11 by HDPlayser