espressif/arduino-esp32 #6410: Problem with interrupt after switching from version 1.0.6 to 2.0.2

MiyukiDark, 2022-03-10:
### Board

Devkiv v1/WROM-32 + WROM-32E

### Device Description

Devkiv v1 + Devkit v4

### Hardware Configuration

most of them + SPI

### Version

v2.0.2

### IDE Name

Arduino IDE 2.0

### Operating System

Windows 10 

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have code that worked on old 1.0.6.
It is old bloated spaghetti. 
Using 2 hardware timers. One to update the DMD display. Using DMD32 library. 
The second timer is 10ms interval one doing long code and scanning inputs.
And it all runs with WiFi in AP mode doing captive portal 

Using following libraries 
WiFi.h
DNSServer.h ([zhouhan0126][DNSServer---esp32](https://github.com/zhouhan0126/DNSServer---esp32) 
AsyncTCP.h 
ESPAsyncWebServer.h
Preferences.h
DMD32.h

And without those prints it behaves little different 
And it even haveves little different on WROM-32 and WROM-32E

It crashes even when I disable all routines in interrupts except prints 

### Sketch

```cpp
void IRAM_ATTR triggerScan()
{
  Serial.print("d");
  dmd.scanDisplayBySPI();
  Serial.print("s");
}

void display_setup(void)
{
	// return the clock speed of the CPU
  uint8_t cpuClock = ESP.getCpuFreqMHz();
  
  // Use 1st timer of 4 
  // devide cpu clock speed on its speed value by MHz to get 1us for each signal  of the timer
  timer = timerBegin(0, cpuClock, true);
  // Attach triggerScan function to our timer 
  timerAttachInterrupt(timer, &triggerScan, false);
  // Set alarm to call triggerScan function  
  // Repeat the alarm (third parameter) 
  timerAlarmWrite(timer, 300, true);

  // Start an alarm 
  timerAlarmEnable(timer);

   //clear/init the DMD pixels held in RAM
   dmd.clearScreen( true );   //true is normal (all pixels off), false is negative (all pixels on)

   dmd.selectFont(FONT);

   Serial.println("Display done");
}

void IRAM_ATTR MainTimerISR()
{
  Serial.print("i");
  timer10ms();
}

void MainTimer_setup(void)
{
	// return the clock speed of the CPU
  uint8_t cpuClock = ESP.getCpuFreqMHz();
  
	Serial.println(cpuClock);

  // Use 2nd timer of 4 
  // devide cpu clock speed on its speed value by MHz to get 1us for each signal  of the timer
  timer_10_ms = timerBegin(1, cpuClock, true);
  // Attach triggerScan function to our timer 
  timerAttachInterrupt(timer_10_ms, &MainTimerISR, false);
  // Set alarm to call triggerScan function  
  // Repeat the alarm (third parameter) 
  timerAlarmWrite(timer_10_ms, 5000, true);

  // Start an alarm 
  timerAlarmEnable(timer_10_ms);
}

//Wifi Initialization
void setup_mainpage()
{
    Serial.println("WiFi Setup begin");
    WiFi.mode(WIFI_AP);
    Serial.println(ssid_used);
    //WiFi.softAP(ssid_used.c_str(), pswd_used.c_str());
    WiFi.softAP(ssid_used.c_str());
    WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

    dnsServer.start(DNS_PORT, "*", apIP);

  server.on("/action", HTTP_POST, [] (AsyncWebServerRequest * request) {
....
  }

  server.on("/readtimer", handleTimeVal);

  server.addHandler(new CaptiveMainPage());// Captive portal

  server.begin();
}
```

### Debug Message

```plain
When I try to run timerAttachInterrupt(timer, &triggerScan, true); as used to work and is in tutorials.

[ 240][W][esp32-hal-timer.c:225] timerAttachInterrupt(): EDGE timer interrupt is not supported! Setting to LEVEL...
[ 247][W][esp32-hal-timer.c:225] tGuru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

When timerAttachInterrupt(timer, &triggerScan, false);

Display doneGuru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.Memory dump at 0x400d58ac: 0020c009 00004892 82006136Core 1 register dump:PC : 0x400d58b3 PS : 0x00060430 A0 : 0x800d58ef A1 : 0x3ffb27d0A2 : 0x00000000 A3 : 0x00000000 A4 : 0x000000c0 A5 : 0x00000003A6 : 0x00000001 A7 : 0x00000001 A8 : 0x3ffc2ff8 A9 : 0x00000000A10 : 0x3ffc2ec4 A11 : 0x00000000 A12 : 0x000000c0 A13 : 0x3f400578A14 : 0x3ffb2760 A15 : 0x3ffb2760 SAR : 0x00000020 EXCCAUSE: 0x00000000EXCVADDR: 0x00000000 LBEG : 0x4008987d LEND : 0x4008988d LCOUNT : 0xffffffffBacktrace:0x400d58b0:0x3ffb27d00x400d58ec:0x3ffb27f0 0x400e2ec2:0x3ffb2820

and when I put some Serial.println in code it ends up different way 
assert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))
Backtrace:0x400835bd:0x3ffbefac |<-CORRUPTED

When I try to run at 240MHz it ends up with 
ddid[ 214][V][WiFiGeneric.cpp:317] _arduino_event_cb(): AP Stoppedd
[ 215][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Eveassert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))

And on WROM-32E it behaves like this
SGuru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1).

This is how it scrash when I comment out routines in both interrupts 
assert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))
```

### Other Steps to Reproduce

With the old version, it worked on both modules fine even at 160MHz
But it has bugged DHCP 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

MiyukiDark, 2022-03-10:
That is a good point 
I'll try to change it to TickTwo. It is a cleaner solution for a long routine on 10ms interrupt anyway. 
Thanks. 
I was not aware of this library.

VojtechBartoska, 2022-03-16:
@P-R-O-C-H-Y Probably you can help with answering this?

VojtechBartoska, 2022-04-11:
Hello, are you able to test this on Arduino core esp32 v2.0.3-rc1? Thanks!

P-R-O-C-H-Y, 2022-05-04:
Hello, on 2.0.3-RC1 the sketch you posted with some small edits to be able to run it, the timer interrupts work as expected :)
If you still have an issue, feel free to reopen this :)

closed 2022-05-04 by P-R-O-C-H-Y

ElArcanoCero, 2023-04-12:
timerAlarmWrite(timer, 300, true); change this for    timerAlarmWrite(timer, 600, true); or more is too faster to that model