espressif/arduino-esp32 #6414: ESP32_cam  error

ataris99, 2022-03-11:
### Board

ESP32_CAM, ESP32-S

### Device Description

FTDI FT232RL

### Hardware Configuration

Connect the 5V & GND Pin of ESP32 to5V & GND of FTDI Module. Similarly, connect the Rx to UOT and Tx to UOR Pin. And the most important thing, you need to short the IO0 and GND Pin together. .

### Version

v2.0.2

### IDE Name

Arduino-IDE

### Operating System

Windows10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am having this error 
Sketch uses 759489 bytes (57%) of program storage space. Maximum is 1310720 bytes.
Global variables use 42260 bytes (12%) of dynamic memory, leaving 285420 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.1
Serial port COM7
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:c2:07:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000cafff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

while if connected to the 3.3v it says connecting giving me dots and line but yet then says  afatal error occured

### Sketch

```cpp
#include <WebServer.h>
#include <WiFi.h>
#include <esp32cam.h>
 
const char* WIFI_SSID = "ASK4 Wireless (802.1x)";
const char* WIFI_PASS = "mimi12345";
 
WebServer server(80);
 
 
static auto loRes = esp32cam::Resolution::find(320, 240);
static auto midRes = esp32cam::Resolution::find(350, 530);
static auto hiRes = esp32cam::Resolution::find(800, 600);
void serveJpg()
{
  auto frame = esp32cam::capture();
  if (frame == nullptr) {
    Serial.println("CAPTURE FAIL");
    server.send(503, "", "");
    return;
  }
  Serial.printf("CAPTURE OK %dx%d %db\n", frame->getWidth(), frame->getHeight(),
                static_cast<int>(frame->size()));
 
  server.setContentLength(frame->size());
  server.send(200, "image/jpeg");
  WiFiClient client = server.client();
  frame->writeTo(client);
}
 
void handleJpgLo()
{
  if (!esp32cam::Camera.changeResolution(loRes)) {
    Serial.println("SET-LO-RES FAIL");
  }
  serveJpg();
}
 
void handleJpgHi()
{
  if (!esp32cam::Camera.changeResolution(hiRes)) {
    Serial.println("SET-HI-RES FAIL");
  }
  serveJpg();
}
 
void handleJpgMid()
{
  if (!esp32cam::Camera.changeResolution(midRes)) {
    Serial.println("SET-MID-RES FAIL");
  }
  serveJpg();
}
 
 
void  setup(){
  Serial.begin(115200);
  Serial.println();
  {
    using namespace esp32cam;
    Config cfg;
    cfg.setPins(pins::AiThinker);
    cfg.setResolution(hiRes);
    cfg.setBufferCount(2);
    cfg.setJpeg(80);
 
    bool ok = Camera.begin(cfg);
    Serial.println(ok ? "CAMERA OK" : "CAMERA FAIL");
  }
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.print("http://");
  Serial.println(WiFi.localIP());
  Serial.println("  /cam-lo.jpg");
  Serial.println("  /cam-hi.jpg");
  Serial.println("  /cam-mid.jpg");
 
  server.on("/cam-lo.jpg", handleJpgLo);
  server.on("/cam-hi.jpg", handleJpgHi);
  server.on("/cam-mid.jpg", handleJpgMid);
 
  server.begin();
}
 
void loop()
{
  server.handleClient();
}
```

### Debug Message

```plain
I am having this error 
Sketch uses 759489 bytes (57%) of program storage space. Maximum is 1310720 bytes.
Global variables use 42260 bytes (12%) of dynamic memory, leaving 285420 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.1
Serial port COM7
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:c2:07:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000cafff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

while if connected to the 3.3v it says connecting giving me dots and line but yet then says  afatal error occured
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-11:
You have a hardware problem. Not related to Arduino core.

ataris99, 2022-03-11:
so do i have to change my hardware?

Jason2866, 2022-03-11:
Your USB serial adapter is probably to blame. If it does not have a EXTRA 3.3V reg chip on board it can not reliable power the ESP32 (needed for flashing). I recommend https://tasmota.github.io/docs/Getting-Started/#serial-to-usb-adapter
Using the "golden" CH340G all such strange flashing where solved for our Tasmota users.

SDS712, 2022-03-25:
Pardon me if I say anything wrong here. I'm new to electronics. But we were facing this exact same error on a custom PCB with ESP-WROOM-32U (ESP32-D0WD chip)

After some searching and debugging our circuit, we found that GPIO12 was being pulled high by default. And Pulling it low solved the problem for us and we were able to upload the firmware. 

Check out [this](https://github.com/espressif/arduino-esp32/issues/333#issuecomment-324525755) & [this](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc#note-about-gpio12-esp32-only)

Hope this helps :-)

VojtechBartoska, 2022-05-10:
Hello, I'm closing this issue as we are not able to fix this due to HW issue.

If you face the same behavior on Espressif HW, please reopen it.

Thanks for understading

closed 2022-05-10 by VojtechBartoska

Kyrylo-Shyvam, 2022-10-16:
I faced a similar problem with ESP32_CAM on Linux. All problems were solved when I changed Upload speed from 921600 to 115200. 
Hope this helps people with similar problems.