espressif/arduino-esp32 #6417: RXonly / TXonly UART use; default pins of UART2 collides with PSRAM use of GPIO16 & GPIO17; PSRAM corruption

sauttefk, 2022-03-11:
### Board

ESP32-WROVER

### Device Description

There are use cases, when an UART only is needed for RX or TX to another device.
If `-1` is used as an argument for the RXD or TXD in `SerialX.begin` default pins are used, instead not using aGPIO for this transmit/receive direction (RXonly / TXonly UART).

The default GPIOs for Serial2 / UART2 are `GPIO16` and `GPIO17`. This collides with the use of those pins for the SPIRAM on ESP32-WROVER.
Therefore when `CONFIG_ESP32_SPIRAM_SUPPORT` is set, either `RX2` and `TXD2` should be mapped to other GPIOs or a runtime error should be printed to the serial console.

### Hardware Configuration

`GPIO16` and `GPIO17` are used for SPIRAM / PSRAM

### Version

v2.0.2

### IDE Name

IDF4.4

### Operating System

irrelevant

### Flash frequency

irrelevant

### PSRAM enabled

yes

### Upload speed

921600

### Description

`GPIO16` and `GPIO17` are used as default GPIOs for UART2 / Serial2

### Expected behaviour
`-1` as parameter on RXD or TXD should lead to an TXonly or RXonly serial connection.

Or at least a warning should be printed when default GPIOs are used together with `CONFIG_ESP32_SPIRAM_SUPPORT` enabled.

### Sketch

```cpp
Serial2.begin(115200, SERIAL_8N1, -1, 22);
```

### Debug Message

```plain
[ 11060][I][HeapStatisticsLogger.hpp:19] run():  Internal  |  267915 |   97867 |   65536 |       0 
Entered 'xt_unhandled_exception()'
Entered 'panic_handler()'
Entering 'esp_panic_handler'
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Core 0: Exception was unhandled.

Core  0 register dump:
PC        : 0x400971e1  PS(State) : 0x00060e30  A0(Return): 0x800980c7  A1(Stack) : 0x3ffd2b90  
0x400971e1: block_size at /esp-idf/components/heap/heap_tlsf_block_functions.h:75
 (inlined by) block_is_last at /esp-idf/components/heap/heap_tlsf_block_functions.h:86
 (inlined by) tlsf_walk_pool at /esp-idf/components/heap/heap_tlsf.c:581

A2        : 0x0000084f  A3        : 0x40098100  A4        : 0x3ffd2bd0  A5        : 0x3ffd2bec  
0x40098100: multi_heap_get_info_tlsf at /esp-idf/components/heap/multi_heap.c:345

A6        : 0x000000fe  A7        : 0x00000001  A8        : 0x8009735a  A9        : 0x3ffd2b70  
A10       : 0x00000854  A11       : 0x00060e23  A12       : 0x00060e20  A13       : 0x3ffd2bd0  
A14       : 0x00000070  A15       : 0x3ffd29e0  SAR       : 0x0000000a  EXCCAUSE  : 0x0000001c  
EXCVADDR  : 0x00000853  LBEG      : 0x4008d4fe  LEND      : 0x4008d509  LCOUNT    : 0x00000000  
0x4008d4fe: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x4008d509: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x400971de:0x3ffd2b900x400980c4:0x3ffd2bb0 0x400d5093:0x3ffd2bd0 0x400e981c:0x3ffd2c10 0x400dd618:0x3ffd2c60 0x400954b3:0x3ffd2c80 0x400954ed:0x3ffd2ca0 0x40095674:0x3ffd2cd0 0x400957d5:0x3ffd2d00 
0x400971de: offset_to_block at /esp-idf/components/heap/heap_tlsf_block_functions.h:134
 (inlined by) block_next at /esp-idf/components/heap/heap_tlsf_block_functions.h:146
 (inlined by) tlsf_walk_pool at /esp-idf/components/heap/heap_tlsf.c:588

0x400980c4: multi_heap_get_info_impl at /esp-idf/components/heap/multi_heap.c:370

0x400d5093: heap_caps_get_info at /esp-idf/components/heap/heap_caps.c:453

0x400e981c: HeapStatisticsLogger::run() at /main/personalities/../modules/HeapStatisticsLogger.hpp:25
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-11:
Should be solved with https://github.com/espressif/arduino-esp32/pull/6394

sauttefk, 2022-03-11:
OK, looks good.

closed 2022-03-11 by sauttefk