espressif/arduino-esp32 #6420: xtensa-esp32-elf-g++: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found

Speedy2k, 2022-03-12:
### Board

Adafruit ESP32 Feather

### Device Description

.

### Hardware Configuration

.

### Version

latest master

### IDE Name

Arduino IDE 1.8.19

### Operating System

macOS 12.2.1

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I can't compile anything with this error:

`xtensa-esp32-elf-g++: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found compilation terminated.`

### Sketch

```cpp
Anything just don't compile
```

### Debug Message

```plain
.
```

### Other Steps to Reproduce

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-12:
Prpbably duplicate of #6269

Speedy2k, 2022-03-12:
I don’t think it is the exact same error?

VojtechBartoska, 2022-03-16:
@Speedy2k did you try some of the proposals/updates which are proposed in linked issue above?