espressif/arduino-esp32 #6422: ESP-C3 Serial bluetooth won't build due to undefined references

AcuarioCat, 2022-03-13:
### Board

ESP-C3-32S

### Device Description

NodeMCU

### Hardware Configuration

Board not connected to anything.

### Version

v2.0.2

### IDE Name

Visual Micro

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

undefined references when building with ESP-C3

Works fine when building with ESP32

### Sketch

```cpp
//This example code is in the Public Domain (or CC0 licensed, at your option.)
//By Evandro Copercini - 2018
//
//This example creates a bridge between Serial and Classical Bluetooth (SPP)
//and also demonstrate that SerialBT have the same functionalities of a normal Serial

#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
    Serial.begin(115200);
    SerialBT.begin("ESP32test"); //Bluetooth device name
    Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
    if (Serial.available()) {
        SerialBT.write(Serial.read());
    }
    if (SerialBT.available()) {
        Serial.write(SerialBT.read());
    }
    delay(20);
}
```

### Debug Message

```plain
ld.exe: BluetoothSerial\BluetoothSerial.cpp.o: in function .L0
BluetoothSerial.cpp:651: undefined reference to esp_spp_disconnect
 
ld.exe: BluetoothSerial\BluetoothSerial.cpp.o: in function _stop_bt
BluetoothSerial.cpp:652: undefined reference to esp_spp_deinit
 
ld.exe: BluetoothSerial\BluetoothSerial.cpp.o: in function .LVL35
BluetoothSerial.cpp:253: undefined reference to esp_bt_gap_set_scan_mode
 
ld.exe: BluetoothSerial\BluetoothSerial.cpp.o: in function esp_spp_cb
BluetoothSerial.cpp:259: undefined reference to esp_spp_start_srv
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-03-13:
ESP32-C3 does not include a classic bluetooth device, it is BLE only.

AcuarioCat, 2022-03-13:
Just found that - a shame!

PR #6455 (2022-03-28): Warns about SPP only available for ESP32
## Summary
This PR helps users that try to use BT SPP (BluetoothSerial), warning that it only works for the ESP32. 

## Impact
None

## Related links
Fix #6308 
Fix #6422
closed 2022-03-13 by AcuarioCat