espressif/arduino-esp32 #6427: Support TTLS-PAP

hgshoggins, 2022-03-13:
### Related area

Wifi

### Hardware specification

ESP32

### Is your feature request related to a problem?

We're missing the EAP-TTLS + PAP, as it's clearly stated in `esp_wpa2.h` that
```
@attention 2. wpa2 enterprise authentication can only support TLS, PEAP-MSCHAPv2 and TTLS-MSCHAPv2 method.
```
A lot of corporate/student networks support/require PAP.

### Describe the solution you'd like

Support PAP as PHASE2 like it is offered in the `esp_eap_ttls_phase2_types` ENUM: 
```
typedef enum {
    ESP_EAP_TTLS_PHASE2_EAP,
    ESP_EAP_TTLS_PHASE2_MSCHAPV2,
    ESP_EAP_TTLS_PHASE2_MSCHAP,
    ESP_EAP_TTLS_PHASE2_PAP,
    ESP_EAP_TTLS_PHASE2_CHAP
} esp_eap_ttls_phase2_types ;

### Describe alternatives you've considered

None, we need PAP.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-03-14:
Will https://github.com/espressif/arduino-esp32/pull/6398 resolve the issue?

VojtechBartoska, 2022-03-29:
Hello @hgshoggins, any chance you take a look on mentioned PR above?

hgshoggins, 2022-03-29:
I'd love to, but actually I'm not sure how I can manage to do so. Ideally, I would work directly from published upstream master. Unless you guide me on how to work from that PR, but that's out of the scope of these comments, and I'm afraid I don't have time/skills for that. 

Clearly, here is what is at stake right now: I have a WiFi network to which I connect machines (Linux + Windows + Android) authenticating to a RADIUS server through EAP-TTLS (PAP for Phase2), and I can see it's _not working_ with my ESP32, but that's all I can debug (as well as the RADIUS server's debug log saying the ESP32 is refusing the EAP type). And noting that, the comment in the code saying that only `TLS`, `PEAP-MSCHAPv2` and `TTLS-MSCHAPv2` methods are supported pointed me here in the Issues section. But I might not be your best chance to debug further. 

I hope someone else can.

closed 2022-04-21 by VojtechBartoska