espressif/arduino-esp32 #6430: Unable to connect to WiFi network on ESP32-C3

brentru, 2022-03-14:
### Board

Adafruit QTPy ESP32-C3

### Device Description

Adafruit QTPy ESP32-C3

### Hardware Configuration

N/A

### Version

latest master

### IDE Name

Arduino IDE 1.8.20

### Operating System

macOS

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Unable to connect to both a secure WPA2 and an insecure (open) network. The ESP32-C3 seems to disconnect and the authorization expires. The logs show that the static IP is not assigned by the AP.

I've been able to connect to this network with ESP32, ESP8266, ESP32-S2.

Output from WiFiScan's listing, showing this network:
```
scan start
[ 34383][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 130, Status: 0, Results: 10
[ 34383][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
...
2: iotdev (-75)*
...
```

It may be similar to https://github.com/espressif/arduino-esp32/issues/6379, however, reverting to commit https://github.com/espressif/arduino-esp32/commit/bb096153914ddadefc4e1a8b72ef61d0bef41921 does not solve the issue.

### Sketch

```cpp
Using WiFiClientSecure library sketch, https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino
```

### Debug Message

```plain
[   290][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino EventAttempting to connect to SSID: iotdev
....[  3970][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Reason: 2
[  3970][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  3977][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  3984][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[  3994][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
...[  7048][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Reason: 2
[  7048][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  7055][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  7062][D][WiFiGeneric.cpp:898] _eventCallback(): WiFi AutoReconnect Running
[  7070][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brentru, 2022-03-14:
I've disabled the USB serial to follow per the issue @ladyada  was having in https://github.com/espressif/arduino-esp32/issues/6264#issuecomment-1041890192, still getting a crash when attempting to connect.

```
Attempting to connect to SSID: iotdev
[   418][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   535][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[   536][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
[   539]€ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x13 (GLITCH_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
```

ladyada, 2022-03-15:
my notes: also the same hardware works great with micropython wifi projects, with native USB, and micropy is using IDF 4.3.1 (or 4.4? hard to tell) https://github.com/micropython/micropython/tree/master/ports/esp32

me-no-dev, 2022-03-15:
there is a suspicion that this might be caused be recent change we did in WiFiSTA recently, but unfortunately it works here and I can not reproduce. @brentru could you try changing `WIFI_AUTH_WPA2_PSK` [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L101) to `WIFI_AUTH_WPA_PSK` and try again?

brentru, 2022-03-15:
@me-no-dev  Making that change does not work. If it helps, I am using a network with WPA2 security.

```
10:44:44.830 -> [ 16797][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: x, BSSID: b8:f8:53:35:4b:25, Reason: 2
10:44:44.830 -> [ 16797][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
10:44:44.830 -> [ 16804][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
10:44:44.864 -> [ 16811][D][WiFiGeneric.cpp:898] _eventCallback(): WiFi AutoReconnect Running
10:44:44.864 -> [ 16819][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
10:44:46.906 -> [ 18873][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: x, BSSID: 00:00:00:00:00:00, Reason: 201
10:44:46.906 -> [ 18874][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
10:44:46.906 -> [ 18881][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 201 - NO_AP_FOUND
10:44:46.939 -> [ 18888][D][WiFiGeneric.cpp:898] _eventCallback(): WiFi AutoReconnect Running
10:44:46.939 -> [ 18896][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
```

SuGlider, 2022-03-15:
@brentru 
Could you please try running ```WiFi.useStaticBuffers(true);``` before ```WiFi.begin(...)```
I think it may have something to do with C3 ```sdkconfig``` parameters related to WiFi dynamic/static and cache buffers.

Please let me know. Thanks.

brentru, 2022-03-15:
@SuGlider  It pushes it a bit further, but seems to be intermittently working.

**Successful Output from WiFiClientSecure**
```
14:07:02.967 -> [   296][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event[  2417][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: X, BSSID: b8:f8:53:35:4b:25, Reason: 2
14:07:04.941 -> [  2418][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
14:07:04.974 -> [  2425][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
14:07:04.974 -> [  2431][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
14:07:04.974 -> [  2439][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
14:07:08.110 -> [  5583][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: X, BSSID: b8:f8:53:35:4b:25, Channel: 6, Auth: WPA2_PSK
14:07:08.110 -> [  5584][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
14:07:10.340 -> [  7789][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.1.213
14:07:10.340 -> [  7790][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
14:07:10.340 -> [  7793][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.1.213, MASK: 255.255.255.0, GW: 192.168.1.1
14:07:10.902 -> [  8358][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 232332
14:07:10.902 -> [  8360][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
14:07:10.971 -> [  8445][V][ssl_client.cpp:144] start_ssl_client(): Seeding the random number generator
14:07:10.971 -> [  8447][V][ssl_client.cpp:153] start_ssl_client(): Setting up the SSL/TLS structure...
14:07:11.007 -> [  8451][V][ssl_client.cpp:176] start_ssl_client(): Loading CA cert
14:07:11.007 -> [  8459][V][ssl_client.cpp:252] start_ssl_client(): Setting hostname for TLS session...
14:07:11.007 -> [  8463][V][ssl_client.cpp:267] start_ssl_client(): Performing the SSL/TLS handshake...
14:07:12.276 -> [  9738][V][ssl_client.cpp:288] start_ssl_client(): Verifying peer X.509 certificate...
14:07:12.276 -> [  9739][V][ssl_client.cpp:296] start_ssl_client(): Certificate verified.
14:07:12.276 -> [  9741][V][ssl_client.cpp:311] start_ssl_client(): Free internal heap after TLS 200896
14:07:12.313 -> [  9759][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 46 bytes...
14:07:12.313 -> [  9761][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
14:07:12.313 -> [  9768][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 23 bytes...
14:07:12.313 -> [  9775][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
14:07:12.313 -> [  9783][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 17 bytes...
14:07:12.313 -> [  9791][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
14:07:12.350 -> [  9798][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
14:07:13.465 -> [ 10920][V][ssl_client.cpp:319] stop_ssl_socket(): Cleaning SSL connection.
14:07:13.465 -> [ 10921][V][ssl_client.cpp:319] stop_ssl_socket(): Cleaning SSL connection.
```

I reset the board and it connected, then immediately disconnected. Was unable to re-join.
```
14:08:19.995 -> [  2450][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
14:08:22.132 -> [  4567][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: x, BSSID: b8:f8:53:35:4b:25, Channel: 6, Auth: WPA2_PSK
14:08:22.132 -> [  4569][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
14:08:37.345 -> [ 19802][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: x, BSSID: b8:f8:53:35:4b:25, Reason: 7
14:08:37.345 -> [ 19803][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
14:08:37.383 -> [ 19809][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 7 - NOT_ASSOCED
```

SuGlider, 2022-03-15:
@brentru 

Well, that's an improvement from the previous situation. It seems that the issue is really related to static cache configuration from sdkconfig.

Are both sketches, used for testing it, in the link you sent here?
I'll test it more here and try to find out something else.

brentru, 2022-03-15:
@SuGlider  The first test is https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino

The second test is https://learn.adafruit.com/adafruit-qt-py-esp32-c3-wifi-dev-board/wifi-test#wifi-connection-test-3077319-8

Jason2866, 2022-03-16:
Out of interrest (since we don not have problems with the c3) can you try Tasmota?
You can easily install via Browser (Chrome or Edge) via this site
https://arendst.github.io/Tasmota-firmware/
Scroll down to Tasmota C3

me-no-dev, 2022-03-16:
@Jason2866 It's important things to be fixed here first. Running on your fork will not help in that case.

Jason2866, 2022-03-16:
@me-no-dev  I do understand, and i agree, bu we have not changed anything in code regarding wifi. If it works it is "just" a different setting in sdkconfig we have choosen. So this can help to fix this issue.

brentru, 2022-03-16:
@Jason2866  Thank you for the suggestion, I was able to connect to this network using the QTPY C3 and Tasmota firmware. How do your SDKConfig settings differ from what's in arduino-esp32?

Jason2866, 2022-03-17:
Tasmota lib builder https://github.com/Jason2866/esp32-arduino-lib-builder
C3 sdkconfig https://github.com/Jason2866/esp32-arduino-lib-builder/blob/main/configs/defconfig.esp32c3
vs. official c3 sdkconfig https://github.com/espressif/esp32-arduino-lib-builder/blob/master/sdkconfig.esp32
or the https://github.com/espressif/esp32-arduino-lib-builder/blob/esp32-s3-support/configs/defconfig.esp32c3

Our fork is based on esp32-s3 branch

me-no-dev, 2022-03-17:
the only notable difference  see is `CONFIG_ESP32C3_REV_MIN_0=y` in tasmota, but that would suggest that you are running some old C3 chip? In Arduino we set min to Rev3 which is the MP version of the chip.

Jason2866, 2022-03-17:
Yes, we have some c3 boards not v3 (side note; which have been sold as v3...)
Which do work fine (with two changes in [linker file](components/esp_rom/esp32c3/ld/esp32c3.rom.ld))

Jason2866, 2022-03-17:
We have enabled WPA3 too, which is not enabled in official build.
WPA3 works well, we had no user complaining about.

brentru, 2022-03-29:
@P-R-O-C-H-Y I tested this with the recent release of [ESP32 Arduino 2.0.3-RC1 based on ESP-IDF 4.4](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) and the QTPY ESP32-C3

The `WiFiClientSecure` sketch included with Arduino-ESP32 works intermittently,

On a successful run:
```
12:01:15.465 -> ...[  3162][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Channel: 6, Auth: WPA2_PSK
12:01:16.128 -> [  3163][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
12:01:16.442 -> ..[  5351][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.1.213
12:01:18.306 -> [  5352][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
12:01:18.340 -> [  5355][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.1.213, MASK: 255.255.255.0, GW: 192.168.1.1
12:01:18.443 -> Connected to iotdev
12:01:18.443 -> 
12:01:18.443 -> Starting connection to server...
```

On an unsuccessful run:
```
12:01:32.466 -> Attempting to connect to SSID: iotdev
12:01:32.466 -> ...[  3121][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Reason: 203
12:01:33.955 -> [  3121][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:01:33.955 -> [  3128][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 203 - ASSOC_FAIL
12:01:33.955 -> [  3135][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
12:01:33.955 -> [  3149][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:01:34.291 -> ...[  6206][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Reason: 2
12:01:37.043 -> [  6207][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:01:37.043 -> [  6214][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
12:01:37.043 -> [  6221][D][WiFiGeneric.cpp:898] _eventCallback(): WiFi AutoReconnect Running
...
12:01:43.236 -> [ 12421][D][WiFiGeneric.cpp:898] _eventCallback(): WiFi AutoReconnect Running
```

[The WiFiClient (insecure) sketch](https://learn.adafruit.com/adafruit-qt-py-esp32-c3-wifi-dev-board/wifi-test#wifi-connection-test-3077319-8) also works intermittently.

Unsuccessful run output:
```
12:05:16.595 -> Attempting to connect to SSID: iotdev
12:05:16.595 -> [  4108][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
12:05:16.701 -> [  4202][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
12:05:16.701 -> [  4203][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
12:05:16.701 -> [  4212][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:05:17.200 -> .....[  6884][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Reason: 203
12:05:19.384 -> [  6884][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:05:19.384 -> [  6891][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 203 - ASSOC_FAIL
12:05:19.384 -> [  6898][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
12:05:19.384 -> [  6912][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:05:19.844 -> ......[ 10055][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Channel: 6, Auth: WPA2_PSK
12:05:22.547 -> [ 10056][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
12:05:22.830 -> ..............................[ 25122][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: iotdev, BSSID: b8:f8:53:35:4b:25, Reason: 7
12:05:37.604 -> [ 25123][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:05:37.604 -> [ 25130][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 7 - NOT_ASSOCED
12:05:37.858 -> ......
```

Jason2866, 2022-03-29:
Which Brand is your Router / Accesspoint?

brentru, 2022-03-29:
@Jason2866  I'm using a Verizon FiOS G3100. Devices connect to the 2.4GHz "IoT Device Network".

Jason2866, 2022-03-29:
So a very new one. We have users which have problems with brand new feature rich routers when using [Tasmota](https://github.com/arendst/Tasmota) Brand Asus are most of the time problematic. But not only. Until now we did not find the reason for this issue. Other do just work fine. Maybe your issue is such a combination too.
Maybe you have the possibility to try a other AP? If this works a deep dive in the network traffic is needed what is going wrong.

brentru, 2022-03-29:
@Jason2866  I don't think it's the router's age - I've had this router a well over a year and have used it with a large number of ESP8266, ESP32, ESP32-S2 development boards for my work at Adafruit. Additionally, Tasmota does work on this board and router combination.

I would be open to trying another AP and/or diving into the network traffic w/ WireShark - I'll order one up.

Jason2866, 2022-03-29:
Okay, router issues hit all esp devices. If only the C3 behaves badly it is something else.
Tried a IDF44 wifi example sketch already?

SuGlider, 2022-03-29:
> @P-R-O-C-H-Y @SuGlider I tested this with the recent release of [ESP32 Arduino 2.0.3-RC1 based on ESP-IDF 4.4](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) and the QTPY ESP32-C3

@brentru I'll look into it.

SuGlider, 2022-03-30:
@brentru 
I tested both sketches and they worked fine with v2.0.3RC1.

Board: ESP32-C3-DevkitM1-v1.0 (C3 Mini1 module)
I reset the board several times and both sketches connected to AP, got IP and then connected to the remote server.
The AP was configured for ```WPA_PSK``` and then later ```WPA2_PSK``` - always with ```AES```.

### UART0 Debug output (CDC on -- which doens't work with WiFi):

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2370
SHA-256 comparison failed:
Calculated: aa1a3928be93313ac51da0f5a7b98a2f7e879a860d87d81c1ca48e1b45efd195
Expected: b6b3c68f4d9f1b7ef7af034358bf209b1e4d6e5cd0250d873938802b1ee08471
Attempting to boot anyway...
entry 0x403ce000
[   276][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   318][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[   319][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
[   321][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  2393][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: AP_ESP1, BSSID: 90:f6:52:6d:14:54, Channel: 13, Auth: WPA2_PSK
[  2394][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  3271][V][WiFiGeneric.cpp:294] _arduino_event_cb(): STA Got New IP:192.168.1.100
[  3271][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  3274][D][WiFiGeneric.cpp:885] _eventCallback(): STA IP: 192.168.1.100, MASK: 255.255.255.0, GW: 192.168.1.1
[  3334][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 242840
[  3334][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[  3553][V][ssl_client.cpp:144] start_ssl_client(): Seeding the random number generator
[  3555][V][ssl_client.cpp:153] start_ssl_client(): Setting up the SSL/TLS structure...
[  3559][V][ssl_client.cpp:176] start_ssl_client(): Loading CA cert
[  3567][V][ssl_client.cpp:252] start_ssl_client(): Setting hostname for TLS session...
[  3571][V][ssl_client.cpp:267] start_ssl_client(): Performing the SSL/TLS handshake...
[  4996][V][ssl_client.cpp:288] start_ssl_client(): Verifying peer X.509 certificate...
[  4996][V][ssl_client.cpp:296] start_ssl_client(): Certificate verified.
[  4999][V][ssl_client.cpp:311] start_ssl_client(): Free internal heap after TLS 211472
[  5007][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 46 bytes...
[  5016][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
[  5023][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 23 bytes...
[  5031][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
[  5038][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 17 bytes...
[  5046][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
[  5053][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
```

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2370
SHA-256 comparison failed:
Calculated: aa1a3928be93313ac51da0f5a7b98a2f7e879a860d87d81c1ca48e1b45efd195
Expected: b6b3c68f4d9f1b7ef7af034358bf209b1e4d6e5cd0250d873938802b1ee08471
Attempting to boot anyway...
entry 0x403ce000
[  1748][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  1785][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[  1786][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
[  1788][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  3859][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: AP_ESP1, BSSID: 90:f6:52:6d:14:54, Channel: 13, Auth: WPA2_PSK
[  3860][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  4742][V][WiFiGeneric.cpp:294] _arduino_event_cb(): STA Got New IP:192.168.1.100
[  4743][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  4746][D][WiFiGeneric.cpp:885] _eventCallback(): STA IP: 192.168.1.100, MASK: 255.255.255.0, GW: 192.168.1.1
```

SuGlider, 2022-04-01:
@brentru - Please let me know if you still have problems with WiFi using Arduino Core 2.0.3-RC1.

brentru, 2022-04-08:
@SuGlider @VojtechBartoska  Please do not automatically close this, just got back from travel. I will test on Monday.

VojtechBartoska, 2022-04-11:
Sure, thanks @brentru for your help.

brentru, 2022-04-11:
@SuGlider @VojtechBartoska I've re-tested both sketches with Arduino Core 2.0.3-RC1 and they're working.

OK to close this issue, thank you for the assistance!

VojtechBartoska, 2022-04-11:
Perfect, thank you. Closing.

nielsnl68, 2022-06-30:
Not sure why, but i'm using the wemos C3 mini with 2.0.3 version of the arduino lib, and i have still the same issue as reported above.
I found this issue while testing espHome on it while figuring out what is going on i switched to the Arduino IDE to see of i could reproduce the issue, i have been testing it with the `wifiClientEvents` sketch and i could reproduce the AUTH_EXPIRE disconnect reason, i saw earlier.

```
20:27:22.203 -> [WiFi-event] event: 5
20:27:22.203 -> Disconnected from WiFi access point
20:27:22.203 -> WiFi lost connection. Reason: 2
```

lbernstone, 2022-06-30:
It always gives me confidence in a design when the reference photo from the mfr looks like it has had a capacitor ripped off with pliers.

ljbotero, 2023-07-04:
The only way I was able to fix the AP not showing up was by setting the TX Power to a very low value, here's my full sample code:

```c
#include <WiFi.h>
#include <esp_wifi.h>  

void setup(){ 
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.setDebugOutput(true);
  delay(2000);
  Serial.println("Startup....");

  WiFi.hostname("esp32");
  WiFi.begin("BOTEROS-NET","qti6-9mi6-p5eu");
  WiFi.setTxPower(WIFI_POWER_8_5dBm);
  int txPower = WiFi.getTxPower();
  Serial.print("TX power: ");
  Serial.println(txPower);
  
  while (WiFi.status() != WL_CONNECTED && millis()<20000) {
    delay(500);
    Serial.print(".");
  }

  if(WiFi.status() == WL_CONNECTED){
    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("WiFi NOT CONNECTED, starting ap");
    delay(2000);    
    WiFi.mode(WIFI_AP);
    WiFi.softAP("espsoftap","12345678"); // no errors that I can see, but ap never shows up
  }
}

void loop(){
}
```

greycoding, 2023-07-14:
@ljbotero thank you very much for your comment!
I have been trying unsuccessfully for the past two days to establish a connection between my **Adafruit Qt Py ESP32 C3** and WiFi.
Using the Arduino Core 2.0.3-RC1 did not help either.

Only the line you provided
`"WiFi.setTxPower(WIFI_POWER_8_5dBm);`
allowed a connection to be established.

Does this mean that my board also has a faulty hf design like @Jason2866 mentioned?

Jason2866, 2023-07-15:
@greycoding Probably, not a good sign when reducing transmit power is needed.

ladyada, 2023-07-17:
we spent the last few days trying to find if something was changed - we had a good 'run' of the C3 working well after tweaking some component values. the latest boards were built with the same components that we can determine so not sure why they are not working nicely anymore.

@greycoding we can replace with S3 or S2 QT Py, which should do the trick. or just provide a refund, email support@adafruit with a link to this thread.

greycoding, 2023-07-29:
Thank you @ladyada, Adafruit support was quick to provide a replacement.