espressif/arduino-esp32 #6432: mDNS cannot register another

brownrb, 2022-03-15:
### Board

ESP32 Dev Module

### Device Description

Nothing

### Hardware Configuration

No other connected devices or sensors

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows qq

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have encountered an issue concerning mDNS. I have read the existing issues, I have been to the Wiki and Documentation. I a webserver running on Port80, and another Web Server on Port 6060.

I am unable to successfully add another service of the same type. For instance if I do this
      if( MDNS.addService("http", "tcp", 80) == false )
      {
        error_msg_println("Unable to add mdns service for web server");
      }
it succeeds. But if I follow that with this
      if( MDNS.addService("http", "tcp", 6060) == false )
      {
        error_msg_println("Unable to add mdns service for management server");
      }
it fails. 

If I reverse the order, then the other one fails. I can access the one that succeeds but not the one that fails. With mDNS disabled, everything runs fine.

In the documentation it talks about 
Note that MDNS_MULTIPLE_INSTANCE config option needs to be enabled for adding multiple instances

But I cannot find that anywhere in the files for ESP32 core - it only defines CONFIG_MDNS_MULTIPLE_INSTANCE
and there is no occurrence of MDNS_MULTIPLE_INSTANCE anywhere in the core files/libraries (did a global search) or in the issues etc.

Am I missing something?
regards
Robert

### Sketch

```cpp
#include <Arduino.h>
#include <WiFiServer.h>
#include <WiFi.h>

#include <ESPmDNS.h>

#include "bobwebserver.h"
#include "bobmanagementserver.h"

char mDNSNAME[] = "bob";
bool mdnsserverstate = false;

void setup()
{
  // setup webserver and management server
  init_webserver(80);
  init_managementserver(6060);

  if (!MDNS.begin(mDNSNAME))
  {
    Serial.println("err: MDNS: start");
    mdnsserverstate = false;
  }
  else
  {
    Serial.println("Ok: MDNS: Start");
    // Add service to MDNS-SD, MDNS.addService(service, proto, port)
    if ( MDNS.addService("http", "tcp", 80) == false )
    {
      Serial.println("Unable to add mdns service for web server");
    }
    if ( MDNS.addService("http", "tcp", 6060) == false )
    {
      Serial.println("Unable to add mdns service for management server");
    }
    mdnsserverstate = true;
    Serial.println("Ok: MDNS: Running");
  }
}

void loop()
{

}
```

### Debug Message

```plain
Core Debug Level Verbose
No compiler errors.

17:56:04.031 -> [  5626][E][ESPmDNS.cpp:148] addService(): Failed adding service http.tcp.
17:56:04.078 -> 
17:56:04.078 -> Unable to add mdns service for management server
```

### Other Steps to Reproduce

It happens all the time. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-15:
yours is not about multiple instances, but you are adding a service that already exists.

mrengineer7777, 2022-03-15:
MDNS_MULTIPLE_INSTANCE is defined as `#define CONFIG_MDNS_MULTIPLE_INSTANCE 1` in  "...\framework-arduinoespressif32\tools\sdk\esp32\include\config\sdkconfig.h".  Since Arduino libs are pre-compiled with specific sdkconfig.h settings, I can tell you it's already on.

Looking at ESPmDNS.cpp, here's your failing function:
```
bool MDNSResponder::addService(char *name, char *proto, uint16_t port){
    char _name[strlen(name)+2];
    char _proto[strlen(proto)+2];
	if (name[0] == '_') {
		sprintf(_name, "%s", name);
	} else {
		sprintf(_name, "_%s", name);
	}
	if (proto[0] == '_') {
		sprintf(_proto, "%s", proto);
	} else {
		sprintf(_proto, "_%s", proto);
	}

    if(mdns_service_add(NULL, _name, _proto, port, NULL, 0)) {
        log_e("Failed adding service %s.%s.\n", name, proto);
	return false;
    }
    return true;
}
```
Where `mdns_service_add` is returning the error.  That function is in the ESP IDF, specifically at "...\framework-idf\components\mdns\mdns.c".  I'll leave it to you to figure out what went wrong.  It may be your code.

brownrb, 2022-04-12:
Sorry, it can be closed.
As pointed out earlier, it was my undserstanding of the code that was the issue, user error

obviously the first parameter cannot be the same, so the code was changed to 

      if ( MDNS.addService("websrv", "tcp", (uint16_t) ControllerData->get_webserverport()) == false )
      {
        error_msg_println("Unable to add mdns service for web server");
      }
      else
      {
        Serial.println("mdns: added webserver");
      }
      if ( MDNS.addService("mansrv", "tcp", (uint16_t) ControllerData->get_managementserverport()) == false )
      {
        error_msg_println("Unable to add mdns service for management server");
      }
      else
      {
        Serial.println("mdns: added management server");
      }
      if ( MDNS.addService("tcpsrv", "tcp",  (uint16_t) ControllerData->get_tcpipport()) == false )
      {
        error_msg_println("Unable to add mdns service for tcpip server");
      }
      else
      {
        Serial.println("mdns: added tcpip server");
      }

where the first parameter was different for each, then it burst into life. Thanks to
[mrengineer7777](https://github.com/mrengineer7777) 
putting me onto the right path,

Regards
Robert

closed 2022-04-12 by brownrb