espressif/arduino-esp32 #6435: ESP32 Arduino Rainmaker Error

gcohen10, 2022-03-16:
### Board

ESP32-DevKitC-V4

### Device Description

DevKitC

### Hardware Configuration

N/A

### Version

latest master

### IDE Name

Arduino

### Operating System

Windows 10 

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

when I try to connect it to the Wi-Fi network using the Rainmaker app it gives me the following error message "Error, No Bluetooth device found with the given prefix"

I tried with both BLE and SoftAP methods and both don't work and when I go on the Wi-Fi and BLE scanner I dont even see it on the list of devices.

### Sketch

```cpp
//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

//GPIO for push button
static int gpio_0 = 0;
//GPIO for virtual device
static int gpio_switch = 16;
/* Variable for reading pin status*/
bool switch_state = true;

//The framework provides some standard device types like switch, lightbulb, fan, temperaturesensor.
static Switch my_switch("Switch", &gpio_switch);

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {      
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
        printQR(service_name, pop, "ble");
#else
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
#endif        
        break;
    }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();

    if(strcmp(param_name, "Power") == 0) {
        Serial.printf("Received value = %s for %s - %s\n", val.val.b? "true" : "false", device_name, param_name);
        switch_state = val.val.b;
        (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
        param->updateAndReport(val);
    }
}

void setup()
{
    Serial.begin(115200);
    pinMode(gpio_0, INPUT);
    pinMode(gpio_switch, OUTPUT);
    digitalWrite(gpio_switch, DEFAULT_POWER_MODE);

    Node my_node;    
    my_node = RMaker.initNode("ESP RainMaker Node");

    //Standard switch device
    my_switch.addCb(write_callback);
    
    //Add switch device to the node   
    my_node.addDevice(my_switch);

    //This is optional 
    RMaker.enableOTA(OTA_USING_PARAMS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone(). 
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();

    RMaker.enableSchedule();

    RMaker.start();

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{
    if(digitalRead(gpio_0) == LOW) { //Push button pressed

        // Key debounce handling
        delay(100);
        int startTime = millis();
        while(digitalRead(gpio_0) == LOW) delay(50);
        int endTime = millis();

        if ((endTime - startTime) > 10000) {
          // If key pressed for more than 10secs, reset all
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        } else {
          // Toggle device state
          switch_state = !switch_state;
          Serial.printf("Toggle State to %s.\n", switch_state ? "true" : "false");
          my_switch.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, switch_state);
          (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
      }
    }
    delay(100);
}
```

### Debug Message

```plain
None.
```

### Other Steps to Reproduce

I tried with 2 different ESP32's but I am getting the same result. I am not able to detect the ESP32 via BLE or SoftAP mode. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-16:
Since you are using Git, could I bother you to switch to the `esp32-s3-support` branch and give it another go? We have updated RainMaker in that branch

ncohen17, 2022-03-16:
I am not using ESP32-S3 and also not using git. I am using Arduino IDE. What does git have to do with it ?

gcohen10, 2022-03-16:
I am not using ESP32-S3 and also not using git. I am using Arduino IDE with a normal ESP32 WROOM module. What does git have to do with it ?

me-no-dev, 2022-03-16:
that branch support all other chips + ESP32-S3 and has updated code in many places. I asked you so because that is what you posted above:

### Version
latest master

gcohen10, 2022-03-16:
Sorry I meant to write master branch.

gcohen10, 2022-03-16:
Rainmaker partition is only available for ESP32 Dev Module on Arduino. I cannot see that option on any other module on the boards manager.

me-no-dev, 2022-03-16:
you should not need a special partition and since you are using ESP32-WROOM, you can safely select the "ESP32 Dev Module". It covers ALL ESP32 modules and variants (not S2, S3 C3, etc.). And is you are using "master branch" then you still say that you are using the git version of this repo and you could just go in there and `git checkout esp32-s3-support` to switch to the branch I asked :)

gcohen10, 2022-03-16:
I only tried the master branch

me-no-dev, 2022-03-16:
And are you willing to try the other branch? Your other option is to wait until we merge it into master and then you update your local copy.

gcohen10, 2022-03-16:
I will try the other branch

VojtechBartoska, 2022-03-30:
Hello @gcohen10, can you please give it a try on 2.0.3-Rc1? Thanks for your help!

VojtechBartoska, 2022-04-06:
Hello, any updates? Thanks!

VojtechBartoska, 2022-04-20:
@gcohen10 Any chance you have tested this?

ncohen17, 2022-04-20:
Hi @VojtechBartoska ,

I tested it and still has the same issue.

pedrominatel, 2022-04-21:
Hello @gcohen10 and @ncohen17,

I'm testing on ESP32-DevKitC and I'm getting this error:

```
If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_1234","pop":"abcd1234","transport":"ble"}
E (39642) wifi_prov_mgr: Failed to create prov_stop_task!

abort() was called at PC 0x4015feff on core 0

Backtrace:0x40083341:0x3ffdd8300x40094ce5:0x3ffdd850 0x4009a3c1:0x3ffdd870 0x4015feff:0x3ffdd8f0 0x401613d1:0x3ffdd920 0x40161470:0x3ffdd940 0x4016161b:0x3ffdd960 0x40161eb1:0x3ffdd9e0 0x4016206f:0x3ffdda50 0x401693db:0x3ffdda90 0x4016ab61:0x3ffddae0 0x4016b329:0x3ffddb20 0x4013134e:0x3ffddb60 0x40131cf6:0x3ffddb80 0x40125735:0x3ffddbd0 0x401275f3:0x3ffddbf0 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
```

Is this error similar to the error you are reporting?

pedrominatel, 2022-05-16:
Should we close this issue? I can't reproduce it and no more feedback about this issue on 2.0.3.

@VojtechBartoska @ncohen17 @gcohen10

VojtechBartoska, 2022-05-26:
Closing the issue, if needed you can reopen it.

closed 2022-05-26 by VojtechBartoska

zakirox123, 2022-10-10:
i am using ESP-WROOM-32 Device with Rainmaker App and having a problem tried with BLE and SoftAp, no luck BLE Error message "Communication Failed" sometime it says "Cannot find the device"
SoftAP Error message "Assisted Claiming not supported for SoftAP. Cannot Proceed" again tried to flashed with new firmware using espressif firmware tool still no luck. Tried to connect with nRF Connect App, got an error in log "Error (0x85): GATT ERROR" if i use auto connect it will get connect after long time to nRF Connect App
Did you find any solution if so Please help me?

`#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

const char *service_name = "PROV_12345";
const char *pop = "123456";

// define the Device Names
char deviceName_1[] = "Switch1";
char deviceName_2[] = "Switch2";
char deviceName_3[] = "Switch3";
char deviceName_4[] = "Switch4";

// define the GPIO connected with Relays and switches
static uint8_t RelayPin1 = 23;  //D23
static uint8_t RelayPin2 = 22;  //D22
static uint8_t RelayPin3 = 21;  //D21
static uint8_t RelayPin4 = 19;  //D19

static uint8_t SwitchPin1 = 13;  //D13
static uint8_t SwitchPin2 = 12;  //D12
static uint8_t SwitchPin3 = 14;  //D14
static uint8_t SwitchPin4 = 27;  //D27

static uint8_t wifiLed    = 2;   //D2
static uint8_t gpio_reset = 0;

/* Variable for reading pin status*/
// Relay State
bool toggleState_1 = LOW; //Define integer to remember the toggle state for relay 1
bool toggleState_2 = LOW; //Define integer to remember the toggle state for relay 2
bool toggleState_3 = LOW; //Define integer to remember the toggle state for relay 3
bool toggleState_4 = LOW; //Define integer to remember the toggle state for relay 4

// Switch State
bool SwitchState_1 = LOW;
bool SwitchState_2 = LOW;
bool SwitchState_3 = LOW;
bool SwitchState_4 = LOW;

//The framework provides some standard device types like switch, lightbulb, fan, temperature sensor.
static Switch my_switch1(deviceName_1, &RelayPin1);
static Switch my_switch2(deviceName_2, &RelayPin2);
static Switch my_switch3(deviceName_3, &RelayPin3);
static Switch my_switch4(deviceName_4, &RelayPin4);

void sysProvEvent(arduino_event_t *sys_event)
{

    switch (sys_event->event_id) {      
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32

 Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
       printQR(service_name, pop, "ble");
       
        // digitalWrite(wifiLed, true);
#else
       Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
         
#endif        
        break;
        case ARDUINO_EVENT_WIFI_STA_CONNECTED:
        Serial.printf("\nConnected to Wi-Fi!\n");
        digitalWrite(wifiLed, true);
[... 208 lines cut ...]
          RMakerWiFiReset(2);
        }
    }
    delay(100);

    if (WiFi.status() != WL_CONNECTED)
    {
      //Serial.println("WiFi Not Connected");
     // digitalWrite(wifiLed, false);
      ledBlink();
    }
    else
    {
      //Serial.println("WiFi Connected");
      digitalWrite(wifiLed, true);
    }

    manual_control();
}

void ledBlink(){
    digitalWrite(wifiLed, true);
    delay(300);
    digitalWrite(wifiLed, false);
}`