espressif/arduino-esp32 #6436: analogReadMilliVolts returns the wrong value in esp32-s2

yangminglong, 2022-03-16:
### Board

none

### Device Description

esp32-s2

### Hardware Configuration

none

### Version

latest master

### IDE Name

none

### Operating System

none

### Flash frequency

none

### PSRAM enabled

no

### Upload speed

none

### Description

esp32-s2-adc api-reference:
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html](url)

Attenuation                           Measurable input voltage range
ADC_ATTEN_DB_0                 0 mV ~ 750 mV
ADC_ATTEN_DB_2_5              0 mV ~ 1050 mV
ADC_ATTEN_DB_6                 0 mV ~ 1300 mV
ADC_ATTEN_DB_11               0 mV ~ 2500 mV

arduino-esp32-master code:
[https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-adc.c](arduino-esp32-master at line 251)

`
    uint16_t adc_reading = __analogRead(pin);
    uint16_t max_reading = 8191;
    uint16_t max_mv = 1100;
    switch(__analogAttenuation){
        case 3: max_mv = 3900; break;
        case 2: max_mv = 2200; break;
        case 1: max_mv = 1500; break;
        default: break;
    }
    return (adc_reading * max_mv) / max_reading;
`

[#4941](https://github.com/espressif/arduino-esp32/issues/4941#issue-831010258)

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-16:
#5502 is related. Hardware calibration is not used.

VojtechBartoska, 2022-03-16:
Thanks @Jason2866 for linking a relation. @yangminglong Is it okay to move to mentioned issue above. If yes, this issue will be closed as duplicate. Thank you for contributing and your answer.

VojtechBartoska, 2022-04-07:
Closing is duplicate.

closed 2022-04-07 by VojtechBartoska