espressif/arduino-esp32 #6438: WiFi.getNetworkinfo always returns channel 0

Crsarmv7l, 2022-03-17:
### Board

Lillygo T-watch 2020 v-1

### Device Description

No other hardware

### Hardware Configuration

Normal T-watch

### Version

other

### IDE Name

Platform IO

### Operating System

Win10

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Espressif32@3.3.0

Using WiFi.Scan, on ScanComplete I call WiFi.getNetworkinfo with a pointer. All information is correct (BSSID, RSSI, etc) except channel always returns 0.

Tired looking for an already open issue with no luck.

Any ideas?

### Sketch

```cpp
N/a
```

### Debug Message

```plain
N/a
```

### Other Steps to Reproduce

N/a

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, are you able to test this on Arduino core esp32 v2.0.3-rc1? Thanks!

VojtechBartoska, 2022-05-18:
Due to no response, I'm closing this issue as expired.

If needed, you can reopen it.

Thanks for understanding.

closed 2022-05-18 by VojtechBartoska

Crsarmv7l, 2022-05-18:
Sorry. I am not going to reopen the issue as it may be fixed in Arduino core esp32 v2.0.3-rc1, but I cannot change.

Instead I used a cumbersome workaround: Grab the BSSID from the scan, go into promiscuous mode with a beacon packet filter, if BSSID matches the sender of the beacon packet, pull the channel and stop promiscuous.