espressif/arduino-esp32 #6443: touchRead T2 conflict with GPIO 27 output high

htmltiger, 2022-03-17:
### Board

ESP32 Dev Module

### Device Description

Standard 38pin 

### Hardware Configuration

Jumper between 27 and 39 to read

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When pin 27 is set output and high it works normal but when T2 is read it floats

The same sketch works fine with December release

Output from old `ESP32 Arduino 2.0.2 based on ESP-IDF 4.4-beta1` works fine 
```
22:12:14.849 -> pin 27 is high
22:12:15.833 -> Jumper pin 27 to 39 = 4095
22:12:16.864 -> Jumper pin 27 to 39 = 4095
22:12:17.837 -> Jumper pin 27 to 39 = 4095
22:12:18.868 -> Jumper pin 27 to 39 = 4095
22:12:19.846 -> Jumper pin 27 to 39 = 4095
22:12:19.846 -> T2 = 47
22:12:20.830 -> Jumper pin 27 to 39 = 4095
22:12:21.861 -> Jumper pin 27 to 39 = 4095
22:12:22.837 -> Jumper pin 27 to 39 = 4095
22:12:23.868 -> Jumper pin 27 to 39 = 4095
22:12:24.852 -> Jumper pin 27 to 39 = 4095
22:12:24.852 -> T2 = 79
22:12:25.838 -> Jumper pin 27 to 39 = 4095
22:12:26.869 -> Jumper pin 27 to 39 = 4095
22:12:27.853 -> Jumper pin 27 to 39 = 4095
22:12:28.837 -> Jumper pin 27 to 39 = 4095
22:12:29.868 -> Jumper pin 27 to 39 = 4095
```

### Sketch

```cpp
void setup() {
	Serial.begin(115200);
	delay(1000);
	pinMode(27, OUTPUT);
	digitalWrite(27, HIGH);
	Serial.println("pin 27 is high");
}

void loop() {

	static unsigned long every1second = millis();
	if (millis() - every1second >= 1000) { every1second = millis();
		Serial.print("Jumper pin 27 to 39 = ");
		Serial.println(analogRead(39));
	}

	static unsigned long every5seconds = millis();
	if (millis() - every5seconds >= 5000) { every5seconds = millis();
		Serial.print("T2 = ");
		Serial.println(touchRead(T2));
	}
}
```

### Debug Message

```plain
22:18:26.952 -> pin 27 is high
22:18:27.981 -> Jumper pin 27 to 39 = 4095
22:18:28.977 -> Jumper pin 27 to 39 = 4095
22:18:29.962 -> Jumper pin 27 to 39 = 4095
22:18:30.982 -> Jumper pin 27 to 39 = 4095
22:18:31.966 -> T2 = 77
22:18:32.235 -> Jumper pin 27 to 39 = 0
22:18:33.219 -> Jumper pin 27 to 39 = 0
22:18:34.256 -> Jumper pin 27 to 39 = 0
22:18:35.230 -> Jumper pin 27 to 39 = 0
22:18:36.251 -> Jumper pin 27 to 39 = 0
22:18:36.954 -> T2 = 77
22:18:37.235 -> Jumper pin 27 to 39 = 0
22:18:38.258 -> Jumper pin 27 to 39 = 0
22:18:39.242 -> Jumper pin 27 to 39 = 0
22:18:40.235 -> Jumper pin 27 to 39 = 0
22:18:41.219 -> Jumper pin 27 to 39 = 0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-03-31:
Hello, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

htmltiger, 2022-04-01:
The bug is still present in latest master
```
12:24:31.300 -> entry 0x400805f0
12:24:32.471 -> pin 27 is high
12:24:33.459 -> Jumper pin 27 to 39 = 4095
12:24:34.495 -> Jumper pin 27 to 39 = 4095
12:24:35.481 -> Jumper pin 27 to 39 = 4095
12:24:36.469 -> Jumper pin 27 to 39 = 4095
12:24:37.502 -> Jumper pin 27 to 39 = 4095
12:24:37.502 -> T2 = 77
12:24:38.486 -> Jumper pin 27 to 39 = 44
12:24:39.470 -> Jumper pin 27 to 39 = 34
12:24:40.503 -> Jumper pin 27 to 39 = 70
12:24:41.489 -> Jumper pin 27 to 39 = 51
12:24:42.472 -> Jumper pin 27 to 39 = 32
12:24:42.472 -> T2 = 77
12:24:43.460 -> Jumper pin 27 to 39 = 80
12:24:44.496 -> Jumper pin 27 to 39 = 48
12:24:45.480 -> Jumper pin 27 to 39 = 36
12:24:46.466 -> Jumper pin 27 to 39 = 64
12:24:47.500 -> Jumper pin 27 to 39 = 37
12:24:47.500 -> T2 = 77

```

VojtechBartoska, 2022-04-01:
@htmltiger Thanks for your test, adding it to our Roadmap.

Adding you here @P-R-O-C-H-Y, it's a lower priority comparing to your other issues you are working on so to be picked up later :)

htmltiger, 2022-04-22:
I have tried with latest master and no changes, the bug is still present.

P-R-O-C-H-Y, 2022-04-22:
Its not merged yet. If you want to try, you need to make the changes manually according to the linked PR.

htmltiger, 2022-04-22:
The bug is fixed with manual changes,
Thanks.

PR #6609 (2022-04-26): Touch change to init only selected GPIO.
## Description of Change
After touch has been refactored to use ESP-IDF API, its initialising all channels for touch loosing previously setup pins for different purposes. 

The change is that the `touchRead()` and `touchAttachInterrupt()` will initialise touch sensor, but will configure only the pad selected. Other pins will remain as was.

## Tests scenarios
Sketch from issue #6443 tested on ESP32 and S2+S3 with different pins.

## Related links
Closes #6443
closed 2022-04-26 by me-no-dev