espressif/arduino-esp32 #6444: I2c frequency

fary99, 2022-03-18:
### Board

esp32

### Device Description

esp32

### Hardware Configuration

not have

### Version

v2.0.1

### IDE Name

Arduino IDE

### Operating System

windows10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I want to change the I2C clock, but I find that if the frequency exceeds 100K, the IIC frequency will not change to the size I set. For example, if I set 400K, it can only reach about 350K. Is there any way to optimize it?

### Sketch

```cpp
#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.begin(SDA,SCL,(uint32_t)409600);
}

void loop() {
  byte error, address;
  int nDevices = 0;

  Serial.println("Scanning for I2C devices ...");
  for(address = 0x01; address < 0x7f; address++){
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
  }
}
```

### Debug Message

```plain
not have
```

### Other Steps to Reproduce

not have

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-18:
Please try latest Arduino git version. Many changes since version 2.0.1
No support anymore for v2.0.1. Release of new version 2.0.3 will be in a few days

VojtechBartoska, 2022-03-24:
Hi @fary99, did you try it with v2.0.2?

fary99, 2022-03-29:
> 

Tried it. Same thing

VojtechBartoska, 2022-03-29:
Hello @fary99, thanks for your test. 

@PilnyTomas Can you please test this issue on the latest master? In the meantime.

PilnyTomas, 2022-03-30:
Hi, what exactly does it means that the frequency does not change - are you checking it on an oscilloscope, or how?
Right now I can check it only via software and it seems ok. The maximum frequency is 1 million - anything above that will set that 1 million.
Please try to call `Wire.getClock()` to see what the software has to say about the requested clock frequency.
It is also good practice to check the result of functions. `Wire.begin` returns `true` if started successfully.

PilnyTomas, 2022-03-31:
Hi again, we already checked the issue on the oscilloscope and indeed the real output CLK frequency is off by a lot. We will investigate...

fary99, 2022-04-01:
> Hi again, we already checked the issue on the oscilloscope and indeed the real output CLK frequency is off by a lot. We will investigate...
Yes, the IIC frequency can not exceed 100K set in the library file, but I have seen the introduction of ESP-IDF, the IIC frequency can exceed 100K. So I modified the source code so that it could be set to be greater than 100K. Then I used the logic analyzer to capture the frequency of IIC, but it was different from the actual. Looking forward to your solution. Here is the code I modified
esp_err_t i2cInit(uint8_t i2c_num, int8_t sda, int8_t scl, uint32_t frequency){
    if(i2c_num >= SOC_I2C_NUM){
        return ESP_ERR_INVALID_ARG;
    }
#if !CONFIG_DISABLE_HAL_LOCKS
    if(bus[i2c_num].lock == NULL){
        bus[i2c_num].lock = xSemaphoreCreateMutex();
        if(bus[i2c_num].lock == NULL){
            log_e("xSemaphoreCreateMutex failed");
            return ESP_ERR_NO_MEM;
        }
    }
    //acquire lock
    if(xSemaphoreTake(bus[i2c_num].lock, portMAX_DELAY) != pdTRUE){
        log_e("could not acquire lock");
        return ESP_FAIL;
    }
#endif
    if(bus[i2c_num].initialized){
        log_e("bus is already initialized");
        return ESP_FAIL;
    }
//-----------------------------This is modified-----------------------//
    if(!frequency){
        frequency = 100000UL;
    }
//--------------------------------------------------------------------//
    i2c_config_t conf = { };
    conf.mode = I2C_MODE_MASTER;
    conf.scl_io_num = (gpio_num_t)scl;
    conf.sda_io_num = (gpio_num_t)sda;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = frequency;
    conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL; //Any one clock source that is available for the specified frequency may be choosen

    esp_err_t ret = i2c_param_config((i2c_port_t)i2c_num, &conf);
    if (ret != ESP_OK) {
        log_e("i2c_param_config failed");
    } else {
        ret = i2c_driver_install((i2c_port_t)i2c_num, conf.mode, 0, 0, 0);
        if (ret != ESP_OK) {
            log_e("i2c_driver_install failed");
        } else {
            bus[i2c_num].initialized = true;
            bus[i2c_num].frequency = frequency;
        }
    }
#if !CONFIG_DISABLE_HAL_LOCKS
    //release lock
    xSemaphoreGive(bus[i2c_num].lock);
#endif
    return ret;
}

fary99, 2022-04-01:
> Hi again, we already checked the issue on the oscilloscope and indeed the real output CLK frequency is off by a lot. We will investigate...

Yes, the IIC frequency can not exceed 100K set in the library file, but I have seen the introduction of ESP-IDF, the IIC frequency can exceed 100K. So I modified the source code so that it could be set to be greater than 100K. Then I used the logic analyzer to capture the frequency of IIC, but it was different from the actual. Looking forward to your solution. Here is the code I modified
esp_err_t i2cInit(uint8_t i2c_num, int8_t sda, int8_t scl, uint32_t frequency){
if(i2c_num >= SOC_I2C_NUM){
return ESP_ERR_INVALID_ARG;
}
#if !CONFIG_DISABLE_HAL_LOCKS
if(bus[i2c_num].lock == NULL){
bus[i2c_num].lock = xSemaphoreCreateMutex();
if(bus[i2c_num].lock == NULL){
log_e("xSemaphoreCreateMutex failed");
return ESP_ERR_NO_MEM;
}
}
//acquire lock
if(xSemaphoreTake(bus[i2c_num].lock, portMAX_DELAY) != pdTRUE){
log_e("could not acquire lock");
return ESP_FAIL;
}
#endif
if(bus[i2c_num].initialized){
log_e("bus is already initialized");
return ESP_FAIL;
}
//-----------------------------This is modified-----------------------//
if(!frequency){
frequency = 100000UL;
}
//--------------------------------------------------------------------//

dalbert2, 2022-04-24:
Confirmed on an oscilloscope that the I2C bus frequency is significantly lower than configured.  For example, when Wire() is initialized at 400kbps, SCL is clocked around 360kHz.  When Wire.begin() is initialized at 450kbps, SCL is around 384kHz.

I'm using IO21, IO22 and current Arduino Core:
SDK:v4.4-dev-3569-g6a7d83af19-dirty
IDF:v4.4-dev-3569-g6a7d83af19-dirty

Perhaps a more serious problem is failure of the ESP32 to assert SDA fast enough in the cycle after each ACK, resulting in SDA glitching high after SCL goes low.  This does *not* happen at 100kbps and might contribute to ESP32 I2C instability at higher clock speeds.  Please see attached picture for an example.  At 400kbps, this happens after *every* ACK.
[image]

dalbert2, 2022-04-25:
I tried 4.4-beta1 and I2C reliability is dramatically better.
SDK:v4.4-beta1-189-ga79dc75f0a
IDF:v4.4-beta1-189-ga79dc75f0a

When I initialize wire at 400kbps: Wire.begin(IO_SDA, IO_SCL, (uint32_t)400000); 
SCL is clocked around 333kbps, but transactions are reliable (which they are not in 4.4-dev)

PilnyTomas, 2022-05-03:
Hi, we are investigating this issue with ESP-IDF team.
[IDF-4960](https://jira.espressif.com:8443/browse/IDF-4960)

VojtechBartoska, 2022-06-14:
Status update: Issue is solved in ESP-IDF, we will let you know when it's reflected to Arduino core.

dalbert2, 2022-06-14:
Thank you @VojtechBartoska, I'm actually working on switching to ESP-IDF, so I'll try it there.

VojtechBartoska, 2022-07-27:
@PilnyTomas Can you please just quickly retest this? It's supposed to be solved.

PilnyTomas, 2022-08-01:
The IDF fix has improved the situation, but it is still not exactly matching the requested frequency.
For example:
Requested 500 000Hz
Before the fix, with 10k Ohm external pullup: 430 000Hz
After the fix, with 10k Ohm external pullup: 460 000Hz
[image]

dpharris, 2022-08-01:
This appears to be pertinent:
https://github.com/espressif/esp-idf/issues/4300

David

PilnyTomas, 2022-08-05:
Hi everyone, 
The IDF fix is available for Arduino users on a branch [idf-release/v4.4](https://github.com/espressif/arduino-esp32/tree/idf-release/v4.4) `git checkout idf-release/v4.4` and it should be merged in next version (?) @VojtechBartoska 

I have retested the frequency response to the external pull-up value and it seems that lower values correspond better to the requester frequency - try using values less than 5kOhm - 1k should be pretty good.

[image]

Also according to [Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) the supported speed is up to 400kbit/s (see page 278)

[image]

VojtechBartoska, 2022-08-08:
Yes, this will be included for 2.0.5 milestone.

closed 2022-08-10 by VojtechBartoska